berlin-open-wireless-lab/DAWN #48: Data between APs not transferred reliable

Rising-Sun, 2018-11-25:
My devices needs a /etc/init.d/umdns restart to see the dawn service running. Using OpenWrt SNAPSHOT r8484.
After that dawn sees the other router but known APs are only exchanged randomly. Sometimes it works sometimes it does not.
Great program btw.

PolynomialDivision, 2018-11-26:
I know about this issue. I refactored the network connection stuff and changed everything to usock and bound socket to uloop. Now old connection will not bei reliable removed. The router already thinks that there is a connection to the some router and don't build a new one. Did have a Look at the Luci App? I think its good for debugging.

PolynomialDivision, 2018-11-26:
Related to #45

PolynomialDivision, 2018-11-26:
Oh you wrote that even umdns ist Not working. :/
Can u post your umdns config?

PolynomialDivision, 2018-11-26:
Maybe someone else wants to look at the [tcpsocket](https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/network/tcpsocket.c)?

Rising-Sun, 2018-11-26:
Hey,
thanks for your answer. I have two devices Openwrt-Xiaomi with mt7621a and Openwrt-Buffalo with ath9k. For both the configurations.

```
root@Openwrt-Xiaomi:~# cat /etc/config/umdns
config umdns
        option jail 1
        list network lan

```

```
root@Openwrt-Xiaomi:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd0f:61d1:4df2::/48'

config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.178.10'
        option netmask '255.255.255.0'
        option gateway '192.168.178.1'
        option broadcast '192.168.178.255'
        option dns '192.168.178.1'
        option ifname 'eth0.1 bat0.1'

config device 'lan_dev'
        option name 'eth0.1'
        option macaddr '34:ce:00:63:2b:f0'

config interface 'nwi_mesh0'
        option ifname 'mesh0'
        option proto 'batadv'
        option mesh 'bat0'
#       option routing_algo 'BATMAN_V'
        option mtu '2304'

config interface 'wan'
        option ifname 'eth0.2'
        option _orig_ifname 'eth0.2'
        option _orig_bridge 'false'
        option proto 'static'
        option auto '0'
        option type 'bridge'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '2 3 6t'

config switch_vlan
        option device 'switch0'
        option vlan '2'
        option ports '1 6t'

config interface 'lan6'
        option proto 'dhcpv6'
        option ifname '@lan'
        option reqaddress 'try'
        option reqprefix 'no'
```

```
root@Openwrt-Buffalo:~# cat /etc/config/umdns
config umdns
        option jail 1
        list network lan
```
```
root@Openwrt-Buffalo:~# cat /etc/config/network

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdf1:0b68:591c::/48'

config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.178.5'
        option netmask '255.255.255.0'
        option dns '192.168.178.1'
        option gateway '192.168.178.1'
        option broadcast '192.168.178.255'
        option ifname 'eth0.1 bat0.1'

config interface 'nwi_mesh0'
        option ifname 'mesh0'
        option proto 'batadv'
        option mesh 'bat0'
#       option routing_algo 'BATMAN_V'
        option mtu '2304'

config interface 'wan'
        option ifname 'eth1'
        option proto 'dhcp'

config interface 'wan6'
        option ifname 'eth1'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option ports '0 1 2 3 5t'

config interface 'LAN6'
        option proto 'dhcpv6'
        option ifname '@lan'
        option reqaddress 'try'
        option reqprefix 'no'
```

PolynomialDivision, 2018-11-28:
There was a time I Transmitted the data via Multicast or Broadcast. One Problem was the synchronisation of databases... Multicast and Broadcast use UDP.... If U are Not interested in clientsteering U can disable it and use Broadcast or Multicast

Rising-Sun, 2018-12-08:
The problem is gone but dont't know why. The hearing map is synched reliable after hours. I observed if the AP is connected through a 802.11s mesh umdns/dawn needs much longer to register the other Aps. If connected over a fixed cable its almost instantly. 
I testet Broadcast but than I got many wpa_supplicant errors in my syslog.

PolynomialDivision, 2018-12-09:
> The problem is gone but dont't know why. The hearing map is synched reliable after hours. I observed if the AP is connected through a 802.11s mesh umdns/dawn needs much longer to register the other [...]

Sry. I have not much time right now.
I looked again at your config and it's ``list network lan`` only. Maybe add ``nwi_mesh0``?

Rising-Sun, 2018-12-11:
I did add nwi_mesh0 to test that possibilty. It had no difference so removed it again. The same behavior, a fixed conencted device register other fixed aps instantly and needs much longer for the mesh connected device, also occured when I was not using batman-adv for routing. (option mesh_fwding '1' in config/wireless)

PolynomialDivision, 2018-12-12:
Can you give me the commit id of dawn you use?
I will update the commit id in the dawn feed... Maybe you use some old version?

Just building OpenWrt toolchain.

PolynomialDivision, 2018-12-12:
Can u please compile latest version and look at the output again?
I made some debug print for connections...

    Wed Dec 12 17:19:17 2018 daemon.info dawn[30914]: --------Connections------
    Wed Dec 12 17:19:17 2018 daemon.info dawn[30914]: Conenctin to Port: 1026, Connected: True
    Wed Dec 12 17:19:17 2018 daemon.info dawn[30914]: ------------------

Rising-Sun, 2018-12-16:
I have 
`tcp        0      0 0.0.0.0:1026            0.0.0.0:*               LISTEN      3550/dawn`
with a different PID directly after restarting the dawn service.

I tried the new commit of bowl-feed. The result and output in syslog is completly the same. 
The synching with the meshed router took around one hour. The fixed one synched directly.

closed 2020-04-15 by PolynomialDivision

quintani, 2025-05-12:
I also struggled with this issue, and found the solution to be to change the default transmission interval of umdns for ip addresses in DAWN.  See this post where the solution was found.  https://github.com/openwrt/mdnsd/issues/19

From @scapier.

I think, this is because umdns only queries PTR records on start and then gets IPs when they are announced from other hosts, which happens once in a 75 minutes by default.

[mdnsd/interface.c](https://github.com/openwrt/mdnsd/blob/c5ca22a71b2ecb83a2a28b46391dac443be315c2/interface.c#L433)

Line 433 in [c5ca22a](https://github.com/openwrt/mdnsd/commit/c5ca22a71b2ecb83a2a28b46391dac443be315c2)

 dns_send_question(iface, NULL, C_DNS_SD, TYPE_PTR, 0); 
Workaround is to reduce announcements ttl to something small, like 120 seconds. Then at least IPs are broadcasted frequently.
To do this edit /etc/init.d/umdns.
replace:

procd_set_param command "$PROG"
with:

procd_set_param command "$PROG" -t 120