berlin-open-wireless-lab/DAWN #49: Request: Per device metric

Rising-Sun, 2018-11-25:
Can dawn get an option to set one metric field per device. So one can un/prefer specific APs? Have one AP connected as 802.11s mesh and it would be nice if my clients would avoid this AP unless there is no other option for the client.

Is the file /etc/dawn/mac_list rewritten on to the flash often when dawn runs?

PolynomialDivision, 2018-11-26:
/etc/maclist ist used for recognizing devices that have Problems with Client steering.

I did not Chose one Leader in the network. It would be easy to Set some custom weight for routers. I will add this. :) 

Metrics have to be the Same. Orherwise Client steering will calculate different results leading to unwanted behavior.

PolynomialDivision, 2018-11-26:
#50

Rising-Sun, 2018-11-28:
I ask for myself if its bad for my flashrom when a package is constantly writing into the file (/etc/dawn/mac_list)?

Yes custom weight sounds great. I thought one field added in the metric which is not synched but announced to other routers. Or something else what you think is better.

PolynomialDivision, 2018-11-28:
Yeah. Flash has a limited Numbers. Ideal Case would be that maclist would stay empty. This ist mit Always the Case. Best solution probably would be to give the filepath AS congig Parameter.

PolynomialDivision, 2020-04-15:
I moved mac_list into tmp.

closed 2020-04-15 by PolynomialDivision