berlin-open-wireless-lab/DAWN #53: Add package to official OpenWrt repo

aparcar, 2019-04-10:
Any reason not to port DAWN? I'd like to use it but hesitate to compile stuff myself (if not needed for development).

aparcar, 2019-04-19:
ping

PolynomialDivision, 2019-04-21:
Maybe I should try. I consider it still as beta.
As you can see I tried a lot of different things. I tried to use Broadcast, Multicast (even Gnunet... xD) and finally I switched to TCP in a full mesh topology. I do not know if I handle all the tcp/ubus stuff correctly. I would love that somebody could look at this. I'm afraid that I could introduce some vulnerabilities.

Currently, I'm looking how to get further with this project. I

PolynomialDivision, 2019-08-19:
I think I'm okay with getting it upstream.
Next step is that I want to monitor passively client with `libuwifi`.

jneuhauser, 2020-03-02:
How about your plan to create a pr on openwrt/packages?

Probably your code will more attraction and your desired reviewers...

PolynomialDivision, 2020-03-09:
Okay. I will fix the luci-app and will make a pull request.
I think it will be rejected because of code quality... I do debugging via `printf`, and so on. Let's see.

PolynomialDivision, 2020-03-23:
@jneuhauser and @aparcar. Could you somehow say, if you experience any problems when you use umdns?

I always have to add

     list network wan
in

    /etc/config/umdns

Weird.

PolynomialDivision, 2020-03-23:
I made a PR.

PolynomialDivision, 2020-03-24:
Which target do you need? Just use this to get my feed running. :)

    echo -e "untrusted comment: Local build key\nRWSmTtwNnoE2BmVpJt6WB0sayQZtqNw/wVFkqgCVLyBWUv/ZFoRj7E6S" > /etc/opkg/keys/nick.pub

    source /etc/openwrt_release && echo "src/gz bowlfeed http://nhainke.inet.tu-berlin.de/packages/$DISTRIB_ARCH/bowlfeed" >> /etc/opkg/customfeeds.conf

I hosted it on my own...

PolynomialDivision, 2020-03-28:
DAWN is now in official package tree.
https://github.com/openwrt/packages/pull/11648#event-3174837544

I will close.

closed 2020-03-28 by PolynomialDivision

ymilord, 2020-04-30:
Has it been removed from the official repo?

PolynomialDivision, 2020-04-30:
@ymilord You can find it under: https://downloads.openwrt.org/snapshots/packages/

PolynomialDivision, 2020-04-30:
> So it had to be done manually? not via opkg? Only asking cause the verbiage made it seems like it was in the repo and could be install via the UI.

You can install it via opkg if you have a master branch openwrt build. :)
Otherwse, the package will be included in the next stable openwrt release.

ymilord, 2020-04-30:
Ah okay. That part was not clear to me. I just attempted and it complained about a missing dependency.

[image]

I'd love to try it, but I don't want to change my setup from it stock form to crowbar this in if possible. I have 6 WAPs and would love to have a means to manage them.

gstammw, 2020-07-09:
The repo at http://nhainke.inet.tu-berlin.de/packages seems gone. I'm looking to get DAWN and luci-app-dawn working on 19.07.3 as there is no snapshot-release available for my router yet (not ported to ath79 yet).
Any idea where to get it?

hoangminh88, 2020-11-09:
> Which target do you need? Just use this to get my feed running. :)
> Currently, I have `mips_24kc`- Since I made all this PR stuff, I have to tidy up my feeds and stuff.
> 
> Add singing key:
> 
> ```
> echo -e "untrusted comment: Local build key\nRWSmTtwNnoE2BmVpJt6WB0sayQZtqNw/wVFkqgCVLyBWUv/ZFoRj7E6S" > /etc/opkg/keys/nick.pub
> ```
> 
> Add feed:
> 
> ```
> source /etc/openwrt_release && echo "src/gz bowlfeed http://nhainke.inet.tu-berlin.de/packages/$DISTRIB_ARCH/bowlfeed" >> /etc/opkg/customfeeds.conf
> ```
> 
> I hosted it on my own...
> 
> ```
> opkg update
> opkg install dawn
> opkg install luci-app-dawn
> ```

I think link down. I want add repo of dawn and luci-app-dawn

PolynomialDivision, 2020-11-09:
@hoangminh88 That was a custom feed. The repro is no longer maintained since I switched to official openwrt repro.

hoangminh88, 2020-11-09:
> @hoangminh88 That was a custom feed. The repro is no longer maintained since I switched to official openwrt repro.

How add customfeeds of dawn and luci-app-dawn to my device (19.07.4)

teconmoon, 2020-11-12:
@PolynomialDivision Hey there sorry to bother you on a closed issue, but the dawn and luci-app-dawn packages appear to be missing from the openwrt repo with 19.07.4. Is this something that's easily fixed, or is this a repo maintenance issue that has nothing to do with you? Thanks for reading!

jabdoa2, 2020-12-12:
would be interested in trying this on 19.04 as well. can i install the master version on 19.04? or is there a way to build the package for 19.04 (without rebuilding the complete image)? What is the recommended way to contribute to this?

flo-m, 2021-07-28:
Is dawn not available any longer in OpenWRT OPK 19.07.7 ? Cant install it using 
opkg install dawn