berlin-open-wireless-lab/DAWN #56: Failed to look up test object for umdns

perry-amsterdam, 2020-04-08:
I installed dawn and it;s not working. I found the following error message.
Tue Apr  7 04:16:19 2020 daemon.notice procd: /etc/rc.d/S50dawn: Starting Service...
Tue Apr  7 04:16:19 2020 daemon.notice procd: /etc/rc.d/S50dawn: Dawn instance started!
Tue Apr  7 04:16:19 2020 daemon.notice procd: /etc/rc.d/S50dawn: UMDNS with port 1026
Tue Apr  7 04:16:22 2020 daemon.err dawn[1606]: Failed to look up test object for umdns
Tue Apr  7 04:16:22 2020 daemon.err dawn[1606]: Failed to look up test object for umdns
Tue Apr  7 04:19:52 2020 kern.info kernel: [  275.708560] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77d97004
Tue Apr  7 04:19:52 2020 kern.info kernel: [  275.720406] ra  = 0040569d in dawn[400000+b000]
Tue Apr  7 04:47:24 2020 daemon.err uhttpd[1695]: 	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Tue Apr  7 04:47:37 2020 daemon.err uhttpd[1695]: 	/usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:10: in function 'render'
Tue Apr  7 04:48:23 2020 daemon.err uhttpd[1695]: 	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'

PolynomialDivision, 2020-04-08:
Could u please use code formatting for debug messages? You can find that above the text input field.

Please provide the output

    ubus list | grep umdns

And maybe search for hosts

    ubus call umdns hosts

And provide opkg information

    opkg list-installed | grep umdns

Please provide some information about the target and the architecture.

perry-amsterdam, 2020-04-10:
umdns is installed, and i am using openwrt trunk latest version. 
I updated it to the latest sources and stil not working.
Openwrt is installed as a dumb wifi access point.

I found the following error message:
root@localhost:~# logread |grep umdns
Fri Apr 10 08:01:20 2020 daemon.err dawn[1740]: Failed to look up test object for umdns
Fri Apr 10 08:01:21 2020 daemon.err dawn[1740]: Failed to look up test object for umdns
Fri Apr 10 08:01:24 2020 daemon.warn procd: Seccomp support for umdns::instance1 not available
Fri Apr 10 08:01:25 2020 daemon.notice procd: /etc/rc.d/S80umdns: Command failed: Invalid argumen

What i found is that umdns is not listening on any interface:
 root@localhost:~# netstat -lap|grep umdns
unix  3      [ ]         STREAM     CONNECTED       3639 1983/umdns          
root@localhost:~#

root@localhost:~# ubus list | grep umdns
umdns
root@localhost:~# ubus call umdns hosts
{
	
}
root@localhost:~# opkg list-installed | grep umdns
umdns - 2020-04-05-ab7a39a5-3
root@localhost:~# cat /etc/board.json
{
	"model": {
		"id": "gl-inet",
		"name": "GL-iNet 6416A v1"
	},
	"led": {
		"lan": {
			"name": "LAN",
			"sysfs": "gl-inet:green:lan",
			"type": "netdev",
			"device": "eth1",
			"mode": "link tx rx"
		},
		"wlan": {
			"name": "WLAN",
			"sysfs": "gl-inet:red:wlan",
			"trigger": "phy0tpt"
		}
	},
	"network": {
		"lan": {
			"ifname": "eth1",
			"protocol": "static"
		},
		"wan": {
			"ifname": "eth0",
			"protocol": "dhcp"
		}
	}
}
root@localhost:~# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r12897-8b93a24208'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r12897-8b93a24208'
DISTRIB_TAINTS='no-all'
root@localhost:~#

PolynomialDivision, 2020-04-11:
> Fri Apr 10 08:01:20 2020 daemon.err dawn[1740]: Failed to look up test object for umdns
> Fri Apr 10 08:01:21 2020 daemon.err dawn[1740]: Failed to look up test object for umdns

Can u give me your umdns config? `etc/config/umdns`

perry-amsterdam, 2020-04-11:
root@localhost:~# cat /etc/config/umdns 
config umdns
	option jail 1
	list network lan

perry-amsterdam, 2020-04-11:
I installed openwrt 19.07, then umdns works with the same config as used with openwrt trunk.
root@Woonkamer:~# ubus call umdns browse
{
	"_ssh._tcp": {
		"OpenWrt": {
			"ipv4": "192.168.100.150",
			"ipv6": "fe80::6670:2ff:fe7e:24cc",
			"ipv4": "192.168.100.152",
			"ipv6": "fe80::6670:2ff:fef9:162",
			"ipv4": "192.168.100.153",
			"ipv6": "fe80::a2f3:c1ff:fed3:fc3c",
			"ipv4": "192.168.100.151",
			"ipv6": "fe80::6670:2ff:feb5:144a",
			"port": 22,
			"txt": "daemon=dropbear"
		},
		"Kast": {
			"ipv4": "192.168.100.101",
			"ipv6": "fe80::21b:c5ff:fe09:a6ef",
			"port": 22,
			"txt": "daemon=dropbear"
		},
		"Kantoor": {
			"ipv4": "192.168.100.102",
			"ipv6": "fe80::21b:c5ff:fe09:a664",
			"port": 22,
			"txt": "daemon=dropbear"
		},
		"Balkon": {
			"ipv4": "192.168.100.103",
			"ipv6": "fe80::e695:6eff:fe40:923",
			"port": 22,
			"txt": "daemon=dropbear"
		}
	}
}
root@Woonkamer:~#

perry-amsterdam, 2020-04-11:
Under openwrt 19.07 umdns is working, now dawn wil not startup.

PolynomialDivision, 2020-04-11:
> Under openwrt 19.07 umdns is working, now dawn wil not startup.

What do u mean with not starting up?

perry-amsterdam, 2020-04-11:
When i do the following
ps|grep dawn 
I dont see the process dawn on openwrt trunk it does.

perry-amsterdam, 2020-04-11:
I looked again at openwrt trunk:
When starting umdns on trunk i get invalid command.
I found the following line, on openwrt 19.07 it works and on trunk not. 
This maybe the problem with umdns ?

ubus call umdns set_config '{ "interfaces": [ "br-lan" ] }'
On openwrt 19.07 it works and on trunk not ?

perry-amsterdam, 2020-04-11:
When i try the following on openwrt trunk, it works:

 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "wan23,lan23" ] }'
 ubus call umdns set_config '{ "interfaces": [ "br-lan" ] }'
 
Woonkamer:~# netstat -lapn|grep umdns
udp        0      0 224.0.0.251:5353        0.0.0.0:*                           15847/umdns
udp        0      0 192.168.100.100:5353    0.0.0.0:*                           15847/umdns
udp        0      0 ff02::fb:5353           :::*                                15847/umdns
udp        0      0 fe80::e695:6eff:fe40:909:5353 :::*                                15847/umdns
unix  3      [ ]         STREAM     CONNECTED      16477 15847/umdns

mrcheap1984, 2020-04-12:
First thing, you have created an awesome program.

I can advise I am getting similar issues when using OpenWRT 19.0.2. I did not have issues on 19.0.0.
The odd things is when I install it, the program runs fine but as soon as I reboot dawn refuses to run.   
 If I remove and then install the program it's fine again.

I tried stopping and starting umdns and got a curios error in the log:
daemon.warn procd: Seccomp support for umdns::instance1 not available.
Interesting it has impacted other users for a different program transmission.
https://forum.openwrt.org/t/issues-with-transmission-after-upgrading/54626

Could OpenWRT have disabled seccomp support in their main releases and this is causing dawn issues?

Thanks.

PolynomialDivision, 2020-04-12:
>  daemon.warn procd: Seccomp support for umdns::instance1 not available.

I have the same error when I restart umdns. I think you can just ignore this warning. I even tried to use procd-ujail and procd-seccomp with seccmop supported kernel, and umdns failed completely. ^^

I know the restart issue: #52 
I think the tcp socket stuff blocks dawn. I have to fix this.

PolynomialDivision, 2020-04-13:
I can not reproduce the error with OpenWrt `19.07.02` on a WDR 4300 v1?

mrcheap1984, 2020-04-13:
I'm getting the issues on two architectures and variants for  OpenWrt 19.07.2, r10947-65030d81f3.
MVEBU - Linksys OpenWRT 1200AC, 1900AC and 1900ACS
ath79- cortexa9 - Netgear 3800
Though I used the same compile machine, I can try using a different machine to rule out a compiler issue, but it's odd it runs after install but not after reset.  Happy to provide further information.

Thanks again.

PolynomialDivision, 2020-04-13:
>  Though I used the same compile machine, I can try using a different machine to rule out a compiler issue, but it's odd it runs after install but not after reset. Happy to provide further information.

With reset you mean reboot?

mrcheap1984, 2020-04-13:
yes, reboot.

PolynomialDivision, 2020-04-14:
> yes, reboot.

Ah, that is important information. I found the bug. I have to increse the procd start parameter.
You can check by setting the `START=99` and switch the symlink

    mv /etc/rc.d/S50dawn /etc/rc.d/S99dawn 

Then it should work. I will make a PR.

mrcheap1984, 2020-04-15:
I rechecked my compiler, found I was pointed towards an old repository, changed to the github openwrt went from dawn_2020-03-23-1_arm_cortex-a9_vfpv3.ipk to dawn_2020-04-10-5709b96a-1_arm_cortex-a9_vfpv3.ipk and
luci-app-dawn_git-20.084.84976-f72fc8f-1_all.ipk to luci-app-dawn_git-20.106.26969-0e22d4d_all.ipk
Then applied your fix and it is now working after a reboot.  Thanks for your help.

PolynomialDivision, 2020-04-15:
>  Then applied your fix and it is now working after a reboot. Thanks for your help.

There is still some error! Sometimes network is not ready.
I changed my PR accordingly. That should fix everything.

PolynomialDivision, 2020-04-15:
@mrcheap1984 

I added several checks and changed my procd script that no error should appear anymore. Please upgrade the version, when the buildbot builded the new version...

tohid6, 2020-04-28:
hi is there can make in whatsminer m3 fake fan rpm instead 6000 in liquid cooling? like firmware or ?

edrikk, 2022-01-10:
Just adding a comment that this item should likely  be reopened.

As noted [here](https://forum.openwrt.org/t/dawn-a-decentralized-wireless-controller/108768/69?u=edrikk) [this ticket](https://bugs.openwrt.org/index.php?do=details&task_id=4138)  has found the root cause to be a missing dependency of DAWN on procd-seccomp.

I have confirmed this working / to resolve the issue on Xiaomi AC2100, TPLink C2600, and Negear R6700, using 21.02 master snapshot as of January 10th, 2022.

i.e. Build without procd-seccomp exhibits this behaviour, while building with procd-seccomp added resolves the issue.

artemklevtsov, 2022-11-15:
This issue appears every time after firmware upgrade.

shaarkys, 2023-02-19:
> This issue appears every time after firmware upgrade.

Check if installing procd-seccomp resolves your issue.

webknjaz, 2024-12-26:
I was just seeing this on a Turris Omnia (it's a router in the network with three TOTOLINK X5000R dumb APs and another old TP-Link TL-WDR4300 v1 dumb AP):
```log
Dec 26 17:14:38 omnia dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
```
Note that I'm not fully in control of the OpenWRT version or the packages there as they have their own repositories and some things are quite old there (for example, they'd package things from 2021 while all the fully controlled nodes let me have 2024 versions of the same software).
All the dumb AP nodes are running OpenWrt 23.05.5 r24106-10cc5fcd00 and are built pretty much the same. This is just some context that is probably irrelevant, though.

The ubus list output does not contain umdns. Here's some details:

```console
root@omnia:~# ubus list | grep umdns
root@omnia:~# /etc/init.d/umdns status
running
root@omnia:~# /etc/init.d/umdns restart
Command failed: Request timed out
root@omnia:~# ubus list | grep umdns
root@omnia:~# ubus call umdns hosts
Command failed: Not found
root@omnia:~# /etc/init.d/umdns status
running
root@omnia:~# opkg list-installed | grep umdns
umdns - 2021-05-13-b777a0b5-5
root@omnia:~# opkg list-installed | grep procd-seccomp
root@omnia:~# opkg list-installed | grep dawn
dawn - 2022-07-24-9e8060ea-1
luci-app-dawn - git-23.093.42704-9c92e46
prometheus-node-exporter-lua-dawn - 2022.06.12-1
root@omnia:~# cat /etc/board.json
{
	"model": {
		"id": "cznic,turris-omnia",
		"name": "Turris Omnia"
	},
	"system": {
		"hostname": "turris",
		"ntpserver": [
			"217.31.202.100",
			"195.113.144.201",
			"195.113.144.238",
			"2001:1488:ffff::100",
			"ntp.nic.cz",
			"0.openwrt.pool.ntp.org",
			"1.openwrt.pool.ntp.org",
			"2.openwrt.pool.ntp.org",
			"3.openwrt.pool.ntp.org"
		]
	},
	"led": {
		"wan": {
			"name": "wan",
			"sysfs": "omnia-led:wan",
			"type": "netdev",
			"device": "eth2",
			"mode": "link tx rx"
		},
		"pci1": {
			"name": "pci1",
			"sysfs": "omnia-led:pci1",
			"type": "netdev",
			"device": "wlan1",
			"mode": "link tx rx"
		},
		"pci2": {
			"name": "pci2",
			"sysfs": "omnia-led:pci2",
			"type": "netdev",
			"device": "wlan2",
			"mode": "link tx rx"
		},
		"pci3": {
			"name": "pci3",
			"sysfs": "omnia-led:pci3",
			"type": "netdev",
			"device": "wlan3",
			"mode": "link tx rx"
		}
	},
	"network": {
		"lan": {
			"ifname": "lan0 lan1 lan2 lan3 lan4",
			"protocol": "static"
		},
		"wan": {
			"ifname": "eth2",
			"protocol": "none"
		}
	}
}
root@omnia:~# cat /etc/openwrt_release
DISTRIB_ID='TurrisOS'
DISTRIB_RELEASE='7.1.3'
DISTRIB_REVISION='r20343+130-4e1d1b7df0'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='TurrisOS 7.1.3 4e1d1b7df0ce6fa96d7462dc883917682f428046'
DISTRIB_TAINTS='busybox'
root@omnia:~# cat /etc/os-release 
NAME="TurrisOS"
VERSION="7.1.3"
ID="turrisos"
ID_LIKE="lede openwrt"
PRETTY_NAME="TurrisOS 7.1.3"
VERSION_ID="7.1.3"
HOME_URL="https://www.turris.cz/"
BUG_URL="https://gitlab.nic.cz/groups/turris/-/issues/"
SUPPORT_URL="https://www.turris.cz/support/"
BUILD_ID="r20343+130-4e1d1b7df0"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3-d16"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="CZ.NIC"
OPENWRT_DEVICE_MANUFACTURER_URL="https://www.turris.cz/"
OPENWRT_DEVICE_PRODUCT="Turris Omnia"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="TurrisOS 7.1.3 4e1d1b7df0ce6fa96d7462dc883917682f428046"
```

I've found references to this and some upstream issue in my automation from the OpenWRT 19 times. But it's completely commented out:
```console
$ cat files/etc/uci-defaults/99-dawn-workaround 
# NOTE: Running umdns on OpenWRT 21 under seccomp causes weird issues that
# FIXME: Replace this with installing procd-seccomp per
#
# https://github.com/openwrt/openwrt/issues/9190#issuecomment-1040460902
# NOTE: cascade into DAWN not working. OpenWRT 19 and 22 don't have this issue.
# Refs:
# * https://github.com/berlin-open-wireless-lab/DAWN/issues/56#issuecomment-907329042
# * https://github.com/openwrt/openwrt/issues/9190#issuecomment-1040460883
#rm -fv /etc/seccomp/umdns.json
```
I seem to recall withdrawing the `procd-seccomp` package install from my post-OpenWRT 22 build scripts as it seems to be installed by default in later versions:
```console
$ rg -B4 -A4 seccomp *.sh
build-21.02.5.sh
1-#! /usr/bin/env bash
2-
3:# NOTE: An explicit procd-seccomp is needed under OpenWRT 21 to make umdns work
4-# NOTE: as it cascades into non-functional DAWN when broken. The previous
5:# NOTE: workaround was to remove the /etc/seccomp/umdns.json config file.
6-# Ref:
7-# * https://github.com/openwrt/openwrt/issues/9190#issuecomment-1040460902
8-# * https://github.com/openwrt/openwrt/issues/9190#issuecomment-1040460883
9-
--
19-  PACKAGES='\
20-    -dnsmasq -firewall4 -iptables -luci-app-firewall -odhcpd -odhcpd-ipv6only \
21-    -ppp -ppp-mod-pppoe -wpad-basic -wpad-basic-mbedtls -wpad-basic-wolfssl -wpad-mini \
22-    dawn ethtool iperf3 luci luci-app-dawn \
23:    procd-seccomp \
24-    prometheus-node-exporter-lua prometheus-node-exporter-lua-dawn prometheus-node-exporter-lua-nat_traffic prometheus-node-exporter-lua-netstat \
25-    prometheus-node-exporter-lua-openwrt prometheus-node-exporter-lua-wifi prometheus-node-exporter-lua-wifi_stations \
26-    tcpdump wpad-openssl'
```

webknjaz, 2024-12-26:
Alright, so I did that. And can confirm that after installing whatever version of `procd-seccomp` was available, it makes the ubus calls work and lets the network overview and the hearing map web pages actually display something beyond the local node:
```console
root@omnia:~# opkg install procd-seccomp
Installing procd-seccomp (2022-06-01-7a009685-4) to root...
Downloading https://repo.turris.cz/hbk/omnia/packages/base/procd-seccomp_2022-06-01-7a009685-4_arm_cortex-a9_vfpv3-d16.ipk
Configuring procd-seccomp.
root@omnia:~# ubus list | grep umdns
root@omnia:~# ubus call umdns hosts
Command failed: Not found
root@omnia:~# /etc/init.d/umdns restart
root@omnia:~# /etc/init.d/umdns status
running
root@omnia:~# ubus list | grep umdns
umdns
root@omnia:~# ubus call umdns hosts
{
	[...FINALLY SOME JSON OUTPUT HERE...]
}
```

There are no instances of this problem since.

@perry-amsterdam could you edit your original post and mention that the solution is `opkg install procd-seccomp` so that people landing here from Google could jump to applying it instead of having to read through the entire discussion?