berlin-open-wireless-lab/DAWN #60: configuration manual

finasfinas, 2020-04-17:
I have no idea on how to configure this.
I have a router with a 2.4 and 5GHZ radios.
I have an AP with a 2.4 and 5GHX radios.
All the radios are broadcasting the same SSID in different channels with the same authentication and password.
Both are running master OpenWRT with DAWN and Luci DAWN app.
When I open both DAWN pages in Luci they seem to not be aware of one another.
I haven't found any documentation on how to configure this.

PolynomialDivision, 2020-04-17:
Are they all in the same layer 2 network?

It looks like the umdns config is not correct. Often you need to add the wan port to umdns `/etc/config/umds`.
```
config umdns
	option jail 1
	list network lan
	list network wan
```

Which version do u use? I a huge amount of changes. Did u compile dawn by yourself?
This PR is very important because it has a lot of bugfixes included:

https://github.com/openwrt/packages/pull/11870

PolynomialDivision, 2020-04-17:
And if u changed the umdns config reload the stuff

    /etc/init.d/umdns restart

and dawn

    /etc/init.d/dawn restart

PolynomialDivision, 2020-04-17:
And u are right. I have to edit my README.md. It is outdated.

finasfinas, 2020-04-17:
Yes, they are on the same L2 segment. The issue was that I was only looking at the web page configuration and not the config file. 
The config file had broadcast ip as 10.0.0.255 ( or something ) and I changed it to the boadcast ip of my network and now it appears to be working. I changed the dawn config file in both ap's so that they are exactly the same, and what is happening now is that one of the ap's can see the other ap in the network overview page and the other one only sees itself. 

I had to turn off dawn on both ap's because logread is being spammed with daemon.notice hostapd: Beacon request: Unknown measurement mode 255, hundred of messages, and this is killing performance.

I am using the current master, as of 20 minutes ago on both ap's and the dawn/umdns that I am using is the one you get when making update -a 20 minutes ago.

PolynomialDivision, 2020-04-17:
> The config file had broadcast ip as 10.0.0.255 ( or something ) and I changed it to the boadcast ip of my network and now it appears to be working.

I'm not maintaining that broadcast option. I switched to use TCP connections to all APs. And for that u need a working umdns. Please use TCP connections.

> I had to turn off dawn on both ap's because logread is being spammed with daemon.notice hostapd: Beacon request: Unknown measurement mode 255, hundred of messages, and this is killing performance.

802.11k is new. To disable it just change the config parameter to:

    option update_beacon_reports '0' 
`

PolynomialDivision, 2020-04-17:
> Unknown measurement mode 255, hundred of messages, and this is killing performance.

Further, it seems that the config file is wrong. The mode can vary between 0 and 2 (?).
Can u please post your dawn config file?

PolynomialDivision, 2020-04-17:
A valid configuration looks like this:

```
config network
    option broadcast_ip         '10.0.0.255'
    option broadcast_port       '1025'
    option tcp_port             '1026'
    option network_option       '2' # 0 udp broadcast, 1 multicast, 2 tcp
    option shared_key           'Niiiiiiiiiiiiiik'
    option iv                   'Niiiiiiiiiiiiiik'
    option use_symm_enc         '1'
    option collision_domain     '-1'     # enter here aps which are in the same collision domain
    option bandwidth            '-1'     # enter network bandwidth

config ordering
    option sort_order           'cbfs'

config hostapd
    option hostapd_dir          '/var/run/hostapd'

config times
    option update_client         '10'
    option denied_req_threshold  '30'
    option remove_client         '15'
    option remove_probe          '30'
    option remove_ap             '460'
    option update_hostapd        '10'
    option update_tcp_con        '10'
    option update_chan_util      '5'
    option update_beacon_reports '20'

config metric
    option ap_weight            '0'
    option ht_support           '0'
    option vht_support          '0'
    option no_ht_support        '0'
    option no_vht_support       '0'
    option rssi                 '10'
    option low_rssi             '-500'
    option freq                 '100' # some kind of band-steering
    option chan_util            '0'
    option max_chan_util        '-500'
    option rssi_val             '-60'
    option low_rssi_val         '-80'
    option chan_util_val        '140'
    option max_chan_util_val    '170'
    option min_probe_count      '0'
    option bandwidth_threshold  '6'
    option use_station_count    '1'
    option max_station_diff     '1'
    option eval_probe_req       '0'
    option eval_auth_req        '0'     # no real reasoncode...
    option eval_assoc_req       '0'     # just deny assocs...
    option kicking              '0'
    option deny_auth_reason     '1'     # unspecified
    option deny_assoc_reason    '17'    # assoc rejected can't handle new station
    option use_driver_recog     '1'
    option min_number_to_kick   '3'
    option chan_util_avg_period '3'
    option set_hostapd_nr       '1'
    option op_class             '0' #802.11k stuff...
    option duration             '0'
    option mode                 '0'
    option scan_channel         '0'
```

PolynomialDivision, 2020-04-17:
Esepcially that parameter:
``` option set_hostapd_nr       '1'
    option op_class             '0' #802.11k stuff...
    option duration             '0'
    option mode                 '0'
    option scan_channel         '0'
```

And

```
    option update_beacon_reports '20'
```

seem wrong.

finasfinas, 2020-04-17:
the two configuration files that I am running in both ap's
the only thing I changed from default was the ip address

`config network                                                                                                                                                                 
        option broadcast_ip '192.168.88.255'                                                                                                                                   
        option broadcast_port '1025'                                                                                                                                           
        option tcp_port '1026'                                                                                                                                                 
        option network_option '2'                                                                                                                                              
        option shared_key 'Niiiiiiiiiiiiiik'                                                                                                                                   
        option iv 'Niiiiiiiiiiiiiik'                                                                                                                                           
        option use_symm_enc '1'                                                                                                                                                
        option collision_domain '-1'                                                                                                                                           
        option bandwidth '-1'                                                                                                                                                  
                                                                                                                                                                               
config ordering                                                                                                                                                                
        option sort_order 'cbfs'                                                                                                                                               
                                                                                                                                                                               
config hostapd                                                                                                                                                                 
        option hostapd_dir '/var/run/hostapd'                                                                                                                                  
                                                                                                                                                                               
config times                                                                                                                                                                   
        option update_client '10'                                                                                                                                              
        option denied_req_threshold '30'                                                                                                                                       
        option remove_client '15'                                                                                                                                              
        option remove_probe '30'                                                                                                                                               
        option remove_ap '460'                                                                                                                                                 
        option update_hostapd '10'                                                                                                                                             
        option update_tcp_con '10'                                                                                                                                             
        option update_chan_util '5'                                                                                                                                            
                                                                                                                                                                               
config metric                                                                                                                                                                  
        option ap_weight '0'                                                                                                                                                   
        option ht_support '0'                                                                                                                                                  
[... cut ...]
        option deny_assoc_reason '17'                                                                                                                                          
        option use_driver_recog '1'                                                                                                                                            
        option min_number_to_kick '3'                                                                                                                                          
        option chan_util_avg_period '3'    `

`config umdns
        option jail 1
        list network lan
`

finasfinas, 2020-04-17:
will try the config file you suggested on both ap's now.

PolynomialDivision, 2020-04-17:
Your config file does not include important 802.11k parameter.
If they are not specified, dawn will take arbitrary values.

I have to catch that...

Please, try to get umdns running. Maybe just use my umdns config.

PolynomialDivision, 2020-04-17:
> I am using the current master, as of 20 minutes ago on both ap's and the dawn/umdns that I am using is the one you get when making update -a 20 minutes ago.

How do u compile that master branch? I'm updating my installation instructions.

If 802.11k has still weird behavior, please disable it. I made important bug fixes in the last commit.
Clients respond sometimes with wrong beacon report responses and in the last version, this caused a null pointer exception.

finasfinas, 2020-04-17:
So, your configuration file ( dawn ) fixed it. 
 - luci pages on both ap's ( network overview and hearing map ) show the same information
 - no more logread 255 spam

the only thing I am seeing now, from time to time is:
`Fri Apr 17 14:52:26 2020 daemon.notice hostapd: Beacon request: e0:94:67:96:87:f0 does not support passive beacon report
Fri Apr 17 14:52:26 2020 daemon.notice hostapd: Beacon request: 8c:29:37:e8:1b:4c does not support passive beacon report`

So it seems that if you update the configuration file for a sane version everything will work out of the box.
The only thing that needs to be explained is what "working" is. I mean, I think I know that dawn is for, but I haven't seen it explained, with an example of its use. :)

For compilation of the latest master branch I did:
- git clone https://git.openwrt.org/openwrt/openwrt.git OpenWRT.Omnia
- scripts/feeds update -a
- scripts/feeds install -a
- make menuconfig and configure stuff as I like ( all built in, including luci and dawn, etc )
- make -j16
- flash the sysupgrade image

the same for the other device I am trying WPA8630.

PolynomialDivision, 2020-04-17:
I pinged someone in the matrix chat and the PR is now merged. :D

To compile the latest version you can do something like this:
https://github.com/berlin-open-wireless-lab/DAWN/blob/master/INSTALL.md#Compile-Latest-DAWN
BUT, this won't let you automatically get the latest config files.

If you now do

    scripts/feeds update -a

and 

    scripts/feeds install dawn

You will get the latest dawn with
- set nr reports in hostapd
- fix null pointer stuff with 802.11k

The warning `does not support passive beacon report` is that the client does not support 802.11k passive scanning mode. You can try changing the `option mode 0'` config parameter, to try active beacon report, or other modes. Everything is highly alpha stadium.

But please, recompile latest dawn version. It really has importaned bug fixes and improves the ap synchronisation.

PolynomialDivision, 2020-04-17:
Further, I will update the `luci-app-dawn` in the near future allowing to configure those parameters.

PolynomialDivision, 2020-04-17:
> The warning `does not support passive beacon report` is that the client does not support 802.11k passive scanning mode. You can try changing the `option mode '0'` config parameter, to try active beacon report, or other modes. Everything is highly alpha stadium.

In the future, I will also check what 802.11k parameter the client supports. Currently, I'm just try to send 802.1kk frames...

PolynomialDivision, 2020-04-17:
Another important point. You need ful wpad isntallation, since I switched to 802.11v.
Otherwise I can not send wmn_diassoc

finasfinas, 2020-04-17:
built new versions for both devices and flashed them ( before flash I deleted the dawn configuration file on the devices ). 
The new configuration file is applied. no more 255 spam, but each device, upon boot, only sees itself on the dawn luci page. If I restart dawn, they start seeing each other.

finasfinas, 2020-04-17:
I am using full wpad-openssl as default

PolynomialDivision, 2020-04-17:
> If I restart dawn, they start seeing each other.

Weird. That should not happen. How long do u wait?

Could u test steering? In the Luci-App you can activate `Activate Kicking`, and disable `Use Station Count`. And set `5GHz` to e.g. `-30` and everything else in the Metric section to 0. Afterwards, clients with an okay rssi and okay channe l utilization, should connect to 2.4 GHz. You can check the score in the hearing map.
Afterwards you can let clients connect in 5 GHz band again, changing the `5GHz`. If nothing works, please try a manual wmn_disassoc.

PolynomialDivision, 2020-04-17:
An wmn diassoc call looks like this:

    ubus call hostapd.wlan0 wnm_disassoc_imminent '{"addr":"xx:xx:xx:xx:xx:xx","duration":12,"abridged":true,"neighbors":["xxxxxxxxxxxxxxxxxxxxxxx"]}}'

finasfinas, 2020-04-17:
I waited maybe 5 minutes or so. Will wait more on next reboot.

I activated kicking, disabled use station count and set Metric section as you asked. After applying configuration I went to the other AP and luci showed the new configuration, so they are talking to each other.

After 10 minutes there are still clients connected to the 5ghz radio on both ap's.

` SSID is: HOMER

    Client MAC is: BC:41:01:29:E2:1C
        AP is: 98:DE:D0:68:6C:02
        Frequency is: 2.437 GHz (Channel: 6)
        HT support is: not available
        VHT support is: not available
        Signal is: -84
        Channel Utilization is: 66
        Station connected to AP is: 3
        Score is: 0
    Client MAC is: B8:86:87:FE:A4:6D
        AP is: 98:DE:D0:68:6C:02
        Frequency is: 2.437 GHz (Channel: 6)
        HT support is: not available
        VHT support is: not available
        Signal is: -88
        Channel Utilization is: 66
        Station connected to AP is: 3
        Score is: 0
        AP is: 04:F0:21:3C:D1:CB
        Frequency is: 2.462 GHz (Channel: 11)
        HT support is: available
        VHT support is: not available
        Signal is: -86
        Channel Utilization is: 5
        Station connected to AP is: 7
        Score is: 0
    Client MAC is: A0:CC:2B:4A:7F:7A
        AP is: 98:DE:D0:68:6C:02
        Frequency is: 2.437 GHz (Channel: 6)
        HT support is: not available
        VHT support is: not available
        Signal is: -90
        Channel Utilization is: 66
        Station connected to AP is: 3
        Score is: 0
        AP is: 04:F0:21:3C:D1:CB
        Frequency is: 2.462 GHz (Channel: 11)
        HT support is: available
        VHT support is: not available
        Signal is: -86
        Channel Utilization is: 5
        Station connected to AP is: 7
        Score is: 0
    Client MAC is: E0:94:67:96:87:F0
        AP is: 04:F0:21:49:44:1E
        Frequency is: 5.500 GHz (Channel: 100)
        HT support is: available
        VHT support is: available
        Signal is: -73
        Channel Utilization is: 0
        Station connected to AP is: 1
        Score is: -2
    Client MAC is: B0:19:C6:E5:CD:42
        AP is: 98:DE:D0:68:6C:02
        Frequency is: 2.437 GHz (Channel: 6)
        HT support is: not available
        VHT support is: not available
        Signal is: -72
        Channel Utilization is: 66
        Station connected to AP is: 3
        Score is: 0
    Client MAC is: E0:33:8E:18:0C:B2
[... 94 lines cut ...]
    Client MAC is: F4:5C:89:75:28:6A
        AP is: 98:DE:D0:68:6C:03
        Frequency is: 5.180 GHz (Channel: 36)
        HT support is: not available
        VHT support is: not available
        Signal is: -62
        Channel Utilization is: 42
        Station connected to AP is: 1
        Score is: -2
        AP is: 98:DE:D0:68:6C:02
        Frequency is: 2.437 GHz (Channel: 6)
        HT support is: not available
        VHT support is: not available
        Signal is: -48
        Channel Utilization is: 66
        Station connected to AP is: 3
        Score is: 0
        AP is: 04:F0:21:3C:D1:CB
        Frequency is: 2.462 GHz (Channel: 11)
        HT support is: available
        VHT support is: not available
        Signal is: -81
        Channel Utilization is: 5
        Station connected to AP is: 7
        Score is: 0`

finasfinas, 2020-04-17:
Do you want me to try a manual disassociation? what do I put in ["xxxxxxxxxxxxxxxxxxxxxxx"] ?

PolynomialDivision, 2020-04-17:
> I waited maybe 5 minutes or so.

That is to long. Something has to be wrong. :/ I will have a look.

I hope that only the Clients are in the 5 GHz network, who do not have a 2.4 GHz entry.
E.g. Your last client `F4:5C:89:75:28:6A` sees 3 different APs (at least he scaned and saw 3 APs).
The score for the only 5 GHz is `-2`. So actually after some time, it should switch to 2.4 GHz. You can increase the speed that happens if you set `Minimum Number After Kicking Client` to `0`. Then they get directly kicked.

You need the neighbor report of the radio, you want to steer the client. You can do this by calling

    ubus call dawn get_network

And that string under `"neighbor_report":`
If it is `"neighbor_report": "xxx90xx40706030xxxxxxx"` you have to call 

    ubus call hostapd.wlan0 wnm_disassoc_imminent '{"addr":"xx:xx:xx:xx:xx:xx","duration":12,"abridged":true,"neighbors":["xxx90xx40706030xxxxxxx"]}}'

> so they are talking to each other.

Yep. I even try to sync the config with each other.

PolynomialDivision, 2020-04-17:
I even found another bug in my code, that is not so problematic and should not cause any damage.

You can check if wmn disassocs are send with something like this:

    ubus monitor | grep neighbors

PolynomialDivision, 2020-04-17:
Further, if dawn is not showing the neighbor report string (because no station is connected). You can manually call

    ubus call hostapd.wlan0 rrm_nr_get_own

That shows you

```
{
	"value": [
		"xx:xx:xx:xx:xx:xx",
		"Your-SSID-Test",
		"xxx90xx40706030xxxxxxx"
	]
}
```
And u only need `xxx90xx40706030xxxxxxx`

PolynomialDivision, 2020-04-17:
Or you can call

    ubus call hostapd.wlan0 rrm_nr_list

There u should see all the neighboring entrys, since DAWN set them automatically, if you have the option enabled.

PolynomialDivision, 2020-04-17:
Further, it is possible that stations try to roam in the 5 GHz by their own choice. So a ping pong behvaior can happen.
You can forbid to connect to 5 GHz if you enable `Evaluate Authentication Requests`. BUT that can cause trouble! You can further try enable `Evaluate Probe Requests` cause this tries to hide probe responses to clients.

PolynomialDivision, 2020-04-17:
Further, if something with romaing is strange, I add client macs, that I do not want to steer to the `cat /tmp/dawn_mac_list` list. You can just empty the list again, if you want to try to steer the client again.

The mechanism behind that list is, if a client got denied to connect to an AP, and is still not connected after 10s in the network, I will add the client to the "not controllable" list, to avoid that the client can not connect to the network anymore.

But that only happens, if you have this brutal denying to connect to an AP activated and only happens, if you have a very bad driver implementation of roaming and scanning (e.g. old broadcom chips).

finasfinas, 2020-04-17:
ok. thanks!!

finasfinas, 2020-04-17:
So, something strange happened.
- I rebooted the WPA8630 ( without rebooting the Omnia ). After 30 minutes the WPA8630 could only see itself. When I went to check the Omnia, the dawn process had died. I had to start it again. After starting it again only on the Omnia, it saw both, and the WPA8630 also saw both. 

So, not sure why the process died and also why it has to be restarted after boot to work.

PolynomialDivision, 2020-04-17:
> So, not sure why the process died and also why it has to be restarted after boot to work.

Hmmm. Maybe some weird corner case. I will improve logging and then if such stuff happen it would be very nice if I get feedback. :)

finasfinas, 2020-04-17:
sure, I will help.
One thing I noticed is that the atk10 driver crashed. not sure if before or after, but I have a crash dump from the driver on dmesg and considering that the router uptime was like half an hour or so, it is possible that it was the atk driver crashing that triggered some fault on dawn.

PolynomialDivision, 2020-04-17:
Lol. I was able to kill ath9k, too... ^^ Especially, the 802.1k is implemented sometimes very buggy. Please open a bug report for ath10k.

I figured out, that the hearing map view caused a segmentation fault in dawn. So somehow the luci app triggers a bug. I will investigate that. :)

finasfinas, 2020-04-17:
ok, I usually check for master changes every day, so if I see that dawn or umdns is updated I will recompile and try it. 
I have a suggestion for you to consider: :) 
- open a thread in OpenWRT with a tittle like DAWN - the decentralized AP management system -  or similar and present the project to everyone, stating what it does, and how to get it running. I think you will have a huge amount of support for it.
I got notice of DAWN when I saw a new package on the luci apps feed and had to dig around a bit to figure out what's it's purpose and I think it is a really cool idea and clearly something that is missing in OpenWRT, a way to manage a wireless network spanning several nodes from a single location. I would say, something similar to ubiquiti, but much better. :)

PolynomialDivision, 2020-04-18:
Thanks I will do that. :) Currently, I'm afraid that I broke my hearing map. I have to reinvestigate. I always get that `16` error and I saw that I'm missing some parameters... I will have a look again. :S

Here is such a great PR by @JonnyTischbein :
https://github.com/openwrt/luci/pull/3927#event-3248306473

Looks amazing!!! :D

finasfinas, 2020-04-18:
It's quite more readable now!
A cool feature would be a way to manually steer a client from one AP to the other
Another one would be to visualize the network in a more graphical way but this is, I think, less important write now than to make it stable and to add features.

XiaoliChan, 2020-04-20:
HI, i setup a mesh wifi in my house by using openwrt 802.11s, also i open 802.11r in 2.4GHz and 5GHz wifi (My living room got one AP, my bedroom got one AP, and my mother's bedroom also got one AP). 

But i got a problem, how to setup RSSI parameter in dawn when my client signal is smaller than -55 dBm and then swith to each ap, i try to change Low RSSI Threshold in Thershold bar, but it did not show me any effect.

Another problem is the system log show me this report
```
Mon Apr 20 10:02:36 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:04:10 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:05:00 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:06:22 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:06:39 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:07:17 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:07:22 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:08:02 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:08:02 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:09:09 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:10:09 2020 daemon.err dawn[2841]: Neigbor-Report is null!
Mon Apr 20 10:11:01 2020 daemon.err dawn[2841]: Neigbor-Report is null!
```

PolynomialDivision, 2020-04-21:
> But i got a problem, how to setup RSSI parameter in dawn when my client signal is smaller than -55 dBm and then swith to each ap, i try to change Low RSSI Threshold in Thershold bar, but it did not show me any effect.

Here is what I wrote someone else:

> The _val is always the threshold value. rssi_val is the threshold for a good rssi. Maybe set it to -60. low_rssi_val is for a not acceptable rssi, maybe -80 or -90.
> 
> The score mechanism works that the rssi value is added to the score, if it is fullfilled. So if rssi_vaö=-60 and you rssi is -59 you will get score += rssi.
> low_rssi_val is the opposite. Sry, that is a bit complicated. If the rssi is below the value of rssi_low_val the score will be added in rssi_low, so score += low_rssi. Since this is bad, it has to be a negative value! I always set it to a high negative value, since it does not make any sense to steer a client to an ap with a low rssi value. The maximum negative score is -2. -1 is reseved for another corner case. 

So you can just set the rssi_val to `-55` and rssi to 100. Then APs should be preferred with a huge score that have an value above -55. :)

> Another problem is the system log show me this report

Could u try setting `update_beacon_reports` to something like 120 or even a higher value? Maybe the clients deny sending so often beacon reports.

XiaoliChan, 2020-05-01:
@PolynomialDivision Well, i change rssi_val to -55 and rssi to 100, it seen didn't get work.

PolynomialDivision, 2020-06-08:
> > Yes, they are on the same L2 segment. The issue was that I was only looking at the web page configuration and not the config file.
> > The config file had broadcast ip as 10.0.0.255 ( or something ) and I changed it to the boadcast ip of my network and now it appears to be working. I changed the dawn config file in both ap's so that they are exactly the same, and what is happening now is that one of the ap's can see the other ap in the network overview page and the other one only sees itself.
> > I had to turn off dawn on both ap's because logread is being spammed with daemon.notice hostapd: Beacon request: Unknown measurement mode 255, hundred of messages, and this is killing performance.
> > I am using the current master, as of 20 minutes ago on both ap's and the dawn/umdns that I am using is the one you get when making update -a 20 minutes ago.
> 
> I'm still unable to see the other AP's when calling `ubus call dawn get_network` each AP sees only itself. `ubus call umdns hosts` seems to be correctly configured, it shows the other AP's IP in the list (among others hosts I have in the lan). One AP has 2.4G and 5G enabled, the other only has 2.4G and they both have the same SSID...not sure if that matters.

You have IPv4 in your network, or?

jonozzz, 2020-06-08:
Both ipv4 and v6. And my logs were flooded by this `wrong type of rrm array` for which I had to disable stderr in the init file.

PolynomialDivision, 2020-06-08:
> Both ipv4 and v6. And my logs were flooded by this `wrong type of rrm array` for which I had to disable stderr in the init file.

The network stuff is now fixed with latest master.

jonozzz, 2020-06-08:
```
Mon Jun  8 15:12:37 2020 daemon.err dawn[5921]: wrong type of rrm array
Mon Jun  8 15:12:47 2020 daemon.err dawn[5921]: wrong type of rrm array
Mon Jun  8 15:12:57 2020 daemon.err dawn[5921]: wrong type of rrm array
```
I'm still seeing these with today's changes.
```
# opkg list-installed|grep dawn
dawn - 2020-06-08-050c1d7f-1
luci-app-dawn - git-20.141.74258-ffdc0ff
```

XiaoliChan, 2020-07-24:
@PolynomialDivision Hi, i think my client is too lazy to roam in my house, so how can i kick my client automatically? E.g. when my client is over -70dbm for my living room ap, then kick my client to another close-in ap.