berlin-open-wireless-lab/DAWN #68: Observability

mweinelt, 2020-04-30:
I'm currently not sure what DAWN does. I installed it to 5 APs in our hackerspace and I can confirm the APs are talking to each other. It's currently hard to tell what actually happens, partially because all I see logged is devices lacking passive beacon support (cf #67).

DAWN should actually log actions taken, so I can evaluate its usefulness and reason about its decisions. Even more interesting would be integration into prometheus-node-exporter-lua.

mweinelt, 2020-05-01:
In case of the dumb APs  the issue I currently have is that I don't have LuCi on those, since their configuration is not managed interactively anyway. 

Currently that makes me unable to check out the hearing map and the network overview.

PolynomialDivision, 2020-05-01:
DAWN has a ubus interface itself:

Network Overview:

    ubus call dawn get_network

Hearing Map:

    ubus call dawn get_hearing map

In the hearing map you can see some score value for each AP a client sees. DAWN tries to handover a client to an AP with the highest score, after evaluating that x-times.

More useful logging is definitely on my list, too. :)

PolynomialDivision, 2020-05-01:
> prometheus-node-exporter-lua

I have to inform myself what this exporter is doing. If it is for statitstics with Grafana, I think it would be great to have an alternative to luci. :)

PolynomialDivision, 2020-05-27:
Just changed network overview to always show all APs 3a6ef899983421ede69bd640741670328fc74cd0

PolynomialDivision, 2020-05-27:
@mweinelt I wrote some node exporter, I'm not sure if I may export the hearing map, too. Currently searching for a good way to do this. Any comments before I make PR official?

https://github.com/openwrt/packages/pull/12342

PolynomialDivision, 2020-05-27:
I thought maybe switching to Elastic, since it allows to gather logging messages?

mweinelt, 2020-05-27:
> @mweinelt I wrote some node exporter, I'm not sure if I may export the hearing map, too. Currently searching for a good way to do this. Any comments before I make PR official?
> 
> [openwrt/packages#12342](https://github.com/openwrt/packages/pull/12342)

Cool, will check this out later tonight or tomorrow.

> I thought maybe switching to Elastic, since it allows to gather logging messages?

Probably a bit too much for most users, why not rely on the remote syslog functionality of OpenWrt? I'm using this with Grafana Loki, which is quite more lightweight.

PolynomialDivision, 2020-05-27:
> Grafana Loki

Thank u! I was not aware of this programm.

PolynomialDivision, 2020-05-27:
I'm not sure if that is a good idea. Remember that u only need 1 AP as prometheus target. If u query more than 1 you will get redundant information.

[image]

I will add more information to export in future. Thanks for testing! :)

mweinelt, 2020-05-27:
> Remember that u only need 1 AP as prometheus target. If u query more than 1 you will get redundant information. (I'm not satisfied with all that redundant information)
> 

That is somewhat problematic, since the prometheus-node-exporter exports other metrics as well in a single HTTP request to the metrics page. I cannot not duplicate that then, expect by not installing the dawn exporter module on all but one node, which in turn would make image generation more tedious.

PolynomialDivision, 2020-05-27:
I'm very unhappy with that, too! :/

I could write exporter that only export the local AP information. Somehow my daemon has redundant information with other node exporter, e.g. the wifi node exporter...

I'm not sure what the best solution is. Or if I redesign my whole daemon to catch this information "on demand"... Maybe I could even query lua-node-exporter for that. Maybe more a micro service way of  doing stuff, so making an extra hearing map daemon that I can query via ubus...

Is there a good way to query that scripts using c programming?

PolynomialDivision, 2020-05-27:
Is there a small database that I can use on OpenWrt that queries the AP information? I tried installing Prometheus, but it is to big, That way I could just throw away a lot of code and all the network related stuf and just query the timeseries datanase. :)

mweinelt, 2020-05-27:
> Is there a small database that I can use on OpenWrt that queries the AP information? I tried installing Prometheus, but it is to big, That way I could just throw away a lot of code and all the network related stuf and just query the timeseries datanase. :)

No, not that I know of. You'd need something more powerful with a bit of storage to run Prometheus.

mweinelt, 2020-05-27:
> I could write exporter that only export the local AP information.

Probably the way to go.

> Somehow my daemon has redundant information with other node exporter, e.g. the wifi node exporter...

Yeah, I don't have enough overview to comment on that - yet.

PolynomialDivision, 2020-05-28:
I will merge the PR and make a seperate node exporter for people like u. :)

mweinelt, 2020-05-28:
Huh, is my use case so rare?

PolynomialDivision, 2020-05-28:
> Huh, is my use case so rare?

No, but I thought maybe it is a feature for the people who only want to query one node with prometheus. I'm not sure.

mweinelt, 2020-05-28:
If people are already using Prometheus they will likely want to monitor basic resource usage like CPU and memory usage, which requires one to query each node anyway.

PolynomialDivision, 2020-05-30:
I have an idea. :) I will make a commit.

PolynomialDivision, 2020-05-30:
The prometheus node exporter will only exort local node information. :)

jonozzz, 2020-06-09:
So I have one client that is connected to a far AP and it's only seeing a weak signal of -86:
```
                "9C:8E:CD:06:42:09": {
                        "14:91:82:84:E1:35": {
                                "signal": -86,
                                "rcpi": -1,
                                "rsni": -1,
                                "freq": 2432,
                                "ht_capabilities": false,
                                "vht_capabilities": false,
                                "channel_utilization": 0,
                                "num_sta": 10,
                                "ht_support": true,
                                "vht_support": false,
                                "score": -2
                        }
                },
```

My configuration relevant to RSSI is:
```
        option rssi '10'
        option low_rssi '-500'
        option chan_util '0'
        option rssi_val '-60'
        option low_rssi_val '-80'
```

Now, I'm confused on how that score of -2 is calculated. I'd assume that if the RSSI is below -80 (which it is) its score will be dropped by 500 points, so I expect to see a really low value in there for score, not -2. What am I missing?

PolynomialDivision, 2020-06-10:
> not -2. What am I missing?

-2 is a magic number for me that indicates the AP is very bad and should not be used. (-1 is another magic number for me). If u have only bad APs it does not make any sense trying to steer a client to a specific one, because this is highly error prone. So if all APs are bad and all have a score of -2, no AP is favored.

jonozzz, 2020-06-12:
Looks like the ustream PR has some issues still...
```
Fri Jun 12 00:30:49 2020 daemon.err dawn[9032]: Neigbor-Report is null!
Fri Jun 12 00:30:52 2020 daemon.err dawn[9032]: Neigbor-Report is null!
Fri Jun 12 00:31:20 2020 daemon.err dawn[9032]: Neigbor-Report is null!
Fri Jun 12 00:31:20 2020 daemon.err dawn[9032]: Neigbor-Report is null!
Fri Jun 12 00:31:36 2020 daemon.err dawn[9032]: not complete msg, len:2892, expected len:3760
Fri Jun 12 00:31:36 2020 kern.info kernel: [133612.625207] do_page_fault(): sending SIGSEGV to dawn for invalid write access to 00000000
Fri Jun 12 00:31:36 2020 kern.info kernel: [133612.634019] epc = 77ef4b60 in libc.so[77e82000+97000]
Fri Jun 12 00:31:36 2020 kern.info kernel: [133612.639838] ra  = 77e6cf4d in libubox.so[77e68000+17000]
```

PolynomialDivision, 2020-06-12:
Hmmm, could be realloc is returning NULL? @babarosaM33 https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/network/tcpsocket.c#L94

PolynomialDivision, 2020-06-12:
> Looks like the ustream PR has some issues still...
>
> ```
> Fri Jun 12 00:30:49 2020 daemon.err dawn[9032]: Neigbor-Report is null!
> Fri Jun 12 00:30:52 2020 daemon.err dawn[9032]: Neigbor-Report is [...]

Could u look at your memory usage? Maybe somewhere a huge memory leak causes malloc or realloc to return NULL.

jonozzz, 2020-06-12:
I've already reverted to the previous version and I don't think it was a leak. It happened within a couple of minutes after start.

PolynomialDivision, 2020-06-12:
> I've already reverted to the previous version and I don't think it was a leak. It happened within a couple of minutes after start.

I think the `ustream_read` function is used wrong. I will have a deeper look into this. Actually this call needs `ustream_pending_data` before calling read. Let's see.

PolynomialDivision, 2020-06-12:
> I've already reverted to the previous version and I don't think it was a leak. It happened within a couple of minutes after start.

Does the nuller affect previous version? I think I will revert and fix this later.

jonozzz, 2020-06-12:
Yes, I saw lots of `Neigbor-Report is null!` in the previous version, every few seconds. I disabled stderr logging to make it stop.

PolynomialDivision, 2020-06-12:
> Yes, I saw lots of `Neigbor-Report is null!` in the previous version, every few seconds. I disabled stderr logging to make it stop.

you can just increase the 802.11k update time or set it to 0.

jonozzz, 2020-06-12:
`update_client='10'`? Is that in seconds? The time settings are not documented anywhere, I noticed. Also the "Evaluate X" settings, what do they exactly do? For example the "Evaluate Probe Requests" is described as "Evaluate the incoming probe requests." which is not very helpful.

PolynomialDivision, 2020-06-12:
> `update_client='10'`? Is that in seconds? The time settings are not documented anywhere, I noticed. Also the "Evaluate X" settings, what do they exactly do? For example the "Evaluate Probe Requests" is described as "Evaluate the incoming probe requests." which is not very helpful.

Just set this to 0: https://github.com/openwrt/packages/blob/master/net/dawn/files/dawn.config#L27

jonozzz, 2020-06-12:
Set `update_beacon_reports` to `0`, enabled logging, restarted and.......:
```
Fri Jun 12 11:39:09 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:39:10 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:39:10 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:39:10 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:39:17 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:39:17 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:03 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:03 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:16 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:16 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1916, expected len:2292
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1916, expected len:2292
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1916, expected len:2292
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1916, expected len:2292
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1160, expected len:2558959878
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1160, expected len:2558959878
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1160, expected len:2558959878
Fri Jun 12 11:40:17 2020 daemon.err dawn[9943]: not complete msg, len:1160, expected len:2558959878
Fri Jun 12 11:40:19 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:19 2020 daemon.err dawn[9943]: Neigbor-Report is null!
Fri Jun 12 11:40:19 2020 daemon.err dawn[9943]: Neigbor-Report is null!
```

PolynomialDivision, 2020-06-12:
> Set `update_beacon_reports` to `0`, enabled logging, restarted and.......:

Oh. That is actually 802.11v based. I'm not sure why your neighbor-report buffer is null. o.O Actually, my neighbor report string is not copied correct, too. I was not evaluating this a long time since I was fixing other stuff. I will have a look into this. I don't get this error.