berlin-open-wireless-lab/DAWN #70: workaround current umdns error in openwrt trunk

babarosaM33, 2020-05-07:
As of today(2020/05/06), the umdns in openwrt trunk need fix to run normally.

it could not get this script working
ubus call umdns set_config '{ "interfaces": [ "br-lan" ] }'
but output Command failed: Invalid argument.

According to this bug report, https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg50768.html, we should make interface name of lan longer than 6 letters.

Thus, a temporary workaround is to change interface 'lan' in /etc/config/network to 'mylan' or some word longer. As openwrt would add "br-" prefix to this name, and results in "br-mylan" name.

It is necessary to change /etc/config/wireless and /etc/config/dawn accordingly.

In addition, seccomp and jail in openwrt might block umdns working as well. It would be better to remove related lines in /etc/init.d/umdns file.

After these workaround, the dawn and umdns works.

PolynomialDivision, 2020-05-08:
Weird. o.O I can reproduce the issue! I always had `lan` and `wan` in my umdns list and no error. If I delete `wan` again, I have the same error.

babarosaM33, 2020-05-08:
@PolynomialDivision your wan interface name may be longer than 6 letters.

clefru, 2021-04-04:
Just FYI, umdns still fails with the master branch, but seccomp is to blame. Removing or renaming /etc/seccomp/umdns.json fixes the issue, and after a reboot DAWN+umdns work correctly.

PolynomialDivision, 2021-04-04:
> Just FYI, umdns still fails with the master branch, but seccomp is to blame. Removing or renaming /etc/seccomp/umdns.json fixes the issue, and after a reboot DAWN+umdns work correctly.

Is that already somewhere bug reported?

clefru, 2021-04-04:
> > Just FYI, umdns still fails with the master branch, but seccomp is to blame. Removing or renaming /etc/seccomp/umdns.json fixes the issue, and after a reboot DAWN+umdns work correctly.
> 
> Is that already somewhere bug reported?

https://bugs.openwrt.org/index.php?do=details&task_id=3355

PolynomialDivision, 2021-04-05:
> > > Just FYI, umdns still fails with the master branch, but seccomp is to blame. Removing or renaming /etc/seccomp/umdns.json fixes the issue, and after a reboot DAWN+umdns work correctly.
> > 
> > 
> > Is that already somewhere bug reported?
> 
> https://bugs.openwrt.org/index.php?do=details&task_id=3355

Which architecture did u use?

clefru, 2021-04-05:
> Which architecture did u use?

I am seeing this bug both on the ramips-mt7621 and the ath79-generic profile.

richardhd, 2021-04-05:
Here's another angle to this. My router is an R7800 (ipq806x). If I compile my build using "make menuconfig", DAWN runs fine  and there are no problems with umdns.

If I use imagebuilder to make my build, I have the umdns error.

PolynomialDivision, 2021-04-10:
Should be fixed with: https://github.com/openwrt/openwrt/commit/00a85a163405fdf9bee4d8c3f0ee87ca9ed259d6

Could you test again?

@clefru @richardhd @babarosaM33

clefru, 2021-04-10:
I am happy to test once the change has landed in the imagebuilder, which should be at least in 2 days.

clefru, 2021-04-12:
The fix above wasn't enough for me. But with an additional change, it worked: I had simply not specified procd-seccomp as package to install when using the image builder. dawn depends on umdns and umdns depends procd-seccomp. It's sad that the image builder doesn't handle dependencies correctly.

@PolynomialDivision Maybe you can add those two packages as required dependencies somewhere on the DAWN "How to install" page? Thanks!

PolynomialDivision, 2021-04-16:
@clefru I will fix that upstream. Thanks for testing! :)

scottdotau, 2023-05-16:
For anyone visiting this issue in 2023 trying to debug OpenWRT umDNS, above comment helped a lot.

Install the 'procd-seccomp' opkg. As stated, should be a dependency?