berlin-open-wireless-lab/DAWN #74: dawn would get null pointer from iwinfo_backend if hostapd has global interface

babarosaM33, 2020-05-11:
during my daily usage, dawn got killed by kernel sometimes.

I dig into such issue with gdb and finally found the cause.

here is the gdb report:
Program received signal SIGSEGV, Segmentation fault.
get_rssi (ifname=0x77e0e3cb "global",
    client_addr=0x41c87e <client_array+1074> "xxxxx"...) at dawn-2020-04-17-b78a01bf/src/utils/dawn_iwinfo.c:172

and I inserted fprintf(stderr) before this line and found that if ifname was "global" the iwinfo_backend would return a null pointer.

the temporal solution is to delete this file "/var/run/hostapd/global" before dawn startup.

I am not familiar with openwrt and hostapd and have no idea why would hostapd automatically add this interface.

after such workaround, the dawn has been running for several hours without being terminated by kernel.

PolynomialDivision, 2020-05-11:
Thank you!!!
I flashed a new OpenWrt and now I have a global interface, too.

Sorry, currently I'm very busy but in the end of next week I have time to look at all this stuff again.
I just shortly look into DAWN and see if I can exclude this interface.

PolynomialDivision, 2020-05-11:
Actually, I want to rewrite this iwinfo stuff! Since, this is born out of an early stage where I was not able to differentiate between different bssids on the same AP. Now I directly know on which interface a client is sitting.

A fix workaround would be to check `d_name` for `global`...

PolynomialDivision, 2020-05-27:
Fixed from #2d1a6f9c0cc9dfa9e00a7b34541a7c5c3848e6f4

closed 2020-05-27 by PolynomialDivision