berlin-open-wireless-lab/DAWN #79: strange log

babarosaM33, 2020-05-21:
after one day running, these lines appears in system log:

daemon.err dawn[6538]: [RSSI INFO] No hostapd sockets!
daemon.err dawn[6538]: [RSSI INFO] Failed to open dir:freq
daemon.err dawn[6538]: [RSSI INFO] No hostapd sockets!
daemon.err dawn[6538]: [RSSI INFO] Failed to open dir:freq
daemon.err dawn[6538]: [SUBSCRIBING] No hostapd sockets!

it seems dawn get wrong option hostapd_dir from /etc/config/dawn.

thus, variable "hostapd_dir_glob" became "freq".

no idea why such issue could happen.

Ian-Clowes, 2020-05-25:
I saw this as well, but getting random strings for the directory as if it was a buffer overflow or stack corruption.  I put a bit of code in so could see that config was being read correctly.

I was working on various issues to get things running at the time, and I think this just went away as other things started working.

Not a very helpful answer, but at least agreeing that it happens.

Ian-Clowes, 2020-05-26:
I thought back to what I was doing when I saw this are realised how to make a repro.

On my "main" AP I was adjusting some DAWN parameters for no good reason in LUCI.  When configuration is saved the "other" AP starts showing these errors.

I haven't looked at the code, but assume the directory name is being badly unpacked at the remote end, or perhaps unpacked into memory like stack which remains valid to not cause SEGV but is later overwritten.  Or maybe it is never set up properly for transmission.

I'd ask if this parameter should actually be transmitted to the other nodes.  Are there not sme parameters which describe the local device and shouldn't be updated from a different device?

PolynomialDivision, 2020-05-26:
> On my "main" AP I was adjusting some DAWN parameters for no good reason in LUCI. When configuration is saved the "other" AP starts showing these errors.

Thank u! That is some good hint. I will look at this.

Ian-Clowes, 2020-05-26:
More detail on repro...

On device B wth DAWN running as daemon start logread -f.  Observe that all is well.

On device A with DAWN running use LUCI app to edit a parmeter and Save & Apply.

Observe on device B that errors start to apper in logread output.

PolynomialDivision, 2020-05-29:
> On my "main" AP I was adjusting some DAWN parameters for no good reason in LUCI. When configuration is saved the "other" AP starts showing these errors.

Does it also happen if u use

    ubus call dawn reload_config

?

babarosaM33, 2020-05-30:
such issue still exists at latest dawn.
reletive logs:
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: [RSSI INFO] No hostapd sockets!
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: [RSSI INFO] Failed to open dir:�����D�{�%��\%�3�:A�WH��:�C��,�tѣahl����� dh�hay.l=FC�Kp3��KE^.H;U���h_����IE|�e�jM��r��ͩ!�U�?���iZ�����,�a�ho�L�����*%���pd�$���MvOI��#ھ�˶~W��:��2m9��/X!�in�U>瑅��E�+����\Z
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: �B1Ŷ<<(��
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: ?��.�d?��Z&1�7R�l�{eX�lȒ�1��
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: �p�m1D
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: c�����5��콦��!��}���%�1���x�2Wm]�n max_ch
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: [RSSI INFO] No hostapd sockets!
Sat May 30 22:02:05 2020 daemon.err dawn[4102]: [RSSI INFO] Failed to open dir:lization":8,"neighbor_report":"43765000af1900008024090603022a00"}
Sat May 30 22:02:13 2020 daemon.err dawn[4102]: [SUBSCRIBING] No hostapd sockets!
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: [RSSI INFO] No hostapd sockets!
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: [RSSI INFO] Failed to open dir:�����D�{�%��\%�3�:A�WH��:�C��,�tѣahl����� dh�hay.l=FC�Kp3��KE^.H;U���h_����IE|�e�jM��r��ͩ!�U�?���iZ�����,�a�ho�L�����*%���pd�$���MvOI��#ھ�˶~W��:��2m9��/X!�in�U>瑅��E�+����\Z
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: �B1Ŷ<<(��
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: ?��.�d?��Z&1�7R�l�{eX�lȒ�1��
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: �p�m1D
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: c�����5��콦��!��}���%�1���x�2Wm]�
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: [RSSI INFO] No hostapd sockets!
Sat May 30 22:02:15 2020 daemon.err dawn[4102]: [RSSI INFO] Failed to open dir:lization":8,"neighbor_report":"xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"}

PolynomialDivision, 2020-05-30:
Maybe we should just hardcode that into the code as a workaround?

PolynomialDivision, 2020-05-30:
https://forum.openwrt.org/t/correct-way-of-handling-uci-lookup-option-string/65387

PolynomialDivision, 2020-05-30:
I think I have to `strcpy` the directory path.

PolynomialDivision, 2020-05-30:
The datastroage could be effected. That could be very problematic. :S

PolynomialDivision, 2020-05-30:
I think I fixed it: https://github.com/berlin-open-wireless-lab/DAWN/commit/d735c367b5f34481eef107cdc2060e8f67afdb33

But I have to fix it everywhere. I do this tomorrow. :)

PolynomialDivision, 2020-06-01:
Should be fixed with 71536468ff01dc6513e6535face6794705b960c3.
Otherwise reopen again. :)

closed 2020-06-01 by PolynomialDivision