berlin-open-wireless-lab/DAWN #85: memory leak: should call blob_buf_free() after calling blob_buf_init()

babarosaM33, 2020-05-30:
every  blob_buf_init in  src/utils/ubus.c is not followed by blob_buf_free.

according to https://lxr.openwrt.org/source/libubox/blob.c#L89

 88 int
 89 blob_buf_init(struct blob_buf *buf, int id)
 90 {
 91         if (!buf->grow)
 92                 buf->grow = blob_buffer_grow;
 93 
 94         buf->head = buf->buf;
 95         if (blob_add(buf, buf->buf, id, 0) == NULL)
 96                 return -ENOMEM;
 97 
 98         return 0;
 99 }
and
101 void
102 blob_buf_free(struct blob_buf *buf)
103 {
104         free(buf->buf);
105         buf->buf = NULL;
106         buf->buflen = 0;
107 }
blob_buf_init did not reset structure before setup.

as a result, dawn might have serious memory leak.

babarosaM33, 2020-05-30:
issue #79 might be relative with such bug

babarosaM33, 2020-05-30:
maybe I need dig in deeper about the usage of  blob_buf_init

PolynomialDivision, 2020-05-30:
I think u are right? :S Let the issue open, I will have a look at this.

PolynomialDivision, 2020-05-30:
https://forum.openwrt.org/t/should-you-call-blob-buf-free-after-calling-blob-buf-init/65388

jonozzz, 2020-06-15:
I'm seeing a constant memory increase and this only for a couple of hours:

[image]

PolynomialDivision, 2020-06-19:
`blob_buf_init` frees previous memory! :)

closed 2020-06-19 by PolynomialDivision