berlin-open-wireless-lab/DAWN #88: blobmsg_format_json does not support uint8 type

babarosaM33, 2020-06-03:
35 static inline char *blobmsg_format_json(struct blob_attr *attr, bool list)
 36 {
 37         return blobmsg_format_json_with_cb(attr, list, NULL, NULL, -1);
 38 }

from https://lxr.openwrt.org/source/libubox/blobmsg_json.h#L31

322 char *blobmsg_format_json_with_cb(struct blob_attr *attr, bool list, blobmsg_json_format_t cb, void *priv, int indent)
323 {
324         struct strbuf s = {0};
325         bool array;
326         char *ret;
327 
328         setup_strbuf(&s, attr, cb, priv, indent);
329         if (!s.buf)
330                 return NULL;
331 
332         array = blob_is_extended(attr) &&
333                 blobmsg_type(attr) == BLOBMSG_TYPE_ARRAY;
334 
335         if (list)
336                 blobmsg_format_json_list(&s, blobmsg_data(attr), blobmsg_data_len(attr), array);
337         else
338                 blobmsg_format_element(&s, attr, false, false);
339 
340         if (!s.len) {
341                 free(s.buf);
342                 return NULL;
343         }
344 
345         ret = realloc(s.buf, s.pos + 1);
346         if (!ret) {
347                 free(s.buf);
348                 return NULL;
349         }
350 
351         ret[s.pos] = 0;
352 
353         return ret;
354 }

from  https://lxr.openwrt.org/source/libubox/blobmsg_json.c

the actual processing function is blobmsg_format_element

225 static void blobmsg_format_element(struct strbuf *s, struct blob_attr *attr, bool without_name, bool head)
226 {
227         const char *data_str;
228         char buf[317];
229         void *data;
230         int len;
231 
232         if (!blobmsg_check_attr(attr, false))
233                 return;
234 
235         if (!without_name && blobmsg_name(attr)[0]) {
236                 blobmsg_format_string(s, blobmsg_name(attr));
237                 blobmsg_puts(s, ": ", s->indent ? 2 : 1);
238         }
239 
240         data = blobmsg_data(attr);
241         len = blobmsg_data_len(attr);
242 
243         if (!head && s->custom_format) {
244                 data_str = s->custom_format(s->priv, attr);
245                 if (data_str)
246                         goto out;
247         }
248 
249         data_str = buf;
250         switch(blob_id(attr)) {
251         case BLOBMSG_TYPE_UNSPEC:
252                 snprintf(buf, sizeof(buf), "null");
253                 break;
254         case BLOBMSG_TYPE_BOOL:
255                 snprintf(buf, sizeof(buf), "%s", *(uint8_t *)data ? "true" : "false");
256                 break;
257         case BLOBMSG_TYPE_INT16:
258                 snprintf(buf, sizeof(buf), "%" PRId16, (int16_t) be16_to_cpu(*(uint16_t *)data));
259                 break;
260         case BLOBMSG_TYPE_INT32:
261                 snprintf(buf, sizeof(buf), "%" PRId32, (int32_t) be32_to_cpu(*(uint32_t *)data));
262                 break;
263         case BLOBMSG_TYPE_INT64:
264                 snprintf(buf, sizeof(buf), "%" PRId64, (int64_t) be64_to_cpu(*(uint64_t *)data));
265                 break;
266         case BLOBMSG_TYPE_DOUBLE:
267                 snprintf(buf, sizeof(buf), "%lf", blobmsg_get_double(attr));
268                 break;
269         case BLOBMSG_TYPE_STRING:
270                 blobmsg_format_string(s, data);
271                 return;
272         case BLOBMSG_TYPE_ARRAY:
273                 blobmsg_format_json_list(s, data, len, true);
274                 return;
275         case BLOBMSG_TYPE_TABLE:
276                 blobmsg_format_json_list(s, data, len, false);
277                 return;
278         }

from https://lxr.openwrt.org/source/libubox/blobmsg_json.c#L225

babarosaM33, 2020-06-03:
it seems that blobmsg_format_json is used in src/utils/ubus.c

but after converted by blobmsg_format_json ,  every BLOBMSG_TYPE_INT8 would be interpreted as BLOBMSG_TYPE_BOOL.

babarosaM33, 2020-06-03:
25 enum blobmsg_type {
 26         BLOBMSG_TYPE_UNSPEC,
 27         BLOBMSG_TYPE_ARRAY,
 28         BLOBMSG_TYPE_TABLE,
 29         BLOBMSG_TYPE_STRING,
 30         BLOBMSG_TYPE_INT64,
 31         BLOBMSG_TYPE_INT32,
 32         BLOBMSG_TYPE_INT16,
 33         BLOBMSG_TYPE_INT8,
 34         BLOBMSG_TYPE_DOUBLE,
 35         __BLOBMSG_TYPE_LAST,
 36         BLOBMSG_TYPE_LAST = __BLOBMSG_TYPE_LAST - 1,
 37         BLOBMSG_TYPE_BOOL = BLOBMSG_TYPE_INT8,
 38 };

from https://lxr.openwrt.org/source/libubox/blobmsg.h#L25

babarosaM33, 2020-06-03:
many lines of code should be modified to fix such issue.

PolynomialDivision, 2020-06-03:
That u8 stuff is weird, that is why I always use u16 or u32 if I have an actual number.

closed 2020-06-03 by babarosaM33