berlin-open-wireless-lab/DAWN #94: Feature request: AP "nickname"

dwmw2, 2020-06-08:
I've just discovered DAWN, and it's great for seeing the clients connected to the APs in my home network. Thanks.

However, I'm not very good at remembering the MAC addresses of all the APs. They have *names* which I would recognise though.

Would it be possible to use the hostname and wifi interface name (e.g. wlan0) to create a user-readable name which can be displayed in addition to the AP MAC address in the display?

dwmw2, 2020-06-08:
I mocked it up in luci-app-dawn: http://david.woodhou.se/dawn-with-nicknames.png

PolynomialDivision, 2020-06-08:
Yeah of course is that possible. :)
We could do this in different ways. I already looked at this and used some luci plugin that does a reverse dns hostname lookup.
But I would lookup the hostname and wifi interface and send it as information to the other APs. That's easily done.

PolynomialDivision, 2020-06-08:
I will have a look at this tonight.

dwmw2, 2020-06-08:
Hm, what are all those `%d`s doing in the signal column of my mockup, btw? That's there in the real version I started from.

PolynomialDivision, 2020-06-08:
> Hm, what are all those `%d`s doing in the signal column of my mockup, btw? That's there in the real version I started from.

This is because your device does not exchanged a probe request with your AP. Maybe you installed dawn and the device was already connected, or it did an assoc without doing a probe exchange before. I know this problem but it is low priority for me right now. xD Probe exchanges give me important information about the capabilities of a client. I would have to rewrite that and create a probe exchange by myself or do some dummy probe entry. Actually that would not be that hard, but right now I hope the device sends a probe request.

PolynomialDivision, 2020-06-08:
https://github.com/berlin-open-wireless-lab/DAWN/issues/96

dwmw2, 2020-06-08:
I have only just installed dawn in the last few hours; most clients will have been connected already. That sounds like it's the issue. Speaking of which, I believe I saw somewhere that dawn required wpad-full but the OpenWRT package didn't enforce that dependency and on a couple of APs here it's running with wpad-basic. Is that a real dependency and should it be specified in the package? 

(And is there a better place to communicate than abusing github issue reports? Are you on IRC?)

PolynomialDivision, 2020-06-10:
New Feature Merged: https://github.com/openwrt/luci/pull/4147#event-3430512424

closed 2020-06-10 by PolynomialDivision