berlin-open-wireless-lab/DAWN #95: IP communication only works over Legacy IP

dwmw2, 2020-06-08:
Dawn doesn't seem to work over IPv6. This December will mark the 25th anniversary of the publication of RFC1883; the IPv6 standard. That's a full quarter of a century :)

PolynomialDivision, 2020-06-08:
Yeah, I know. ;) I will add this as feature.

dwmw2, 2020-06-08:
Hm, I think there might be a ubus bug to fix first. I only seem to get link-local addresses (which would actually work, but I'm not told which link):

```
root@garden # ubus call umdns browse | grep -A10 _dawn
	"_dawn._tcp": {
		"wndr3800.infradead.org": {
			"ipv4": "90.155.92.219",
			"ipv6": "fe80::841b:5eff:fe77:933",
			"port": 1026
		},
		"lantiq.infradead.org": {
			"ipv4": "90.155.92.193",
			"ipv6": "fe80::1a62:2cff:fe5d:946a",
			"port": 1026
		}

```

In fact the (little-endian) box running OpenWRT master doesn't show up; only the two big-endian boxes running 19.07 are there.

dwmw2, 2020-06-08:
https://bugs.openwrt.org/index.php?do=details&task_id=3167

PolynomialDivision, 2020-08-05:
This is fixed in `68b94f0fb4d5bf659a3eed6045718b8082852966`. Now we have to add support. xD