berlin-open-wireless-lab/DAWN #100: DAWN segfaults

gongsearch, 2020-06-12:
Latest git head (and earlier versions) are segfaulting (on ath79 only?). 

Last messages from logread:

Fri Jun 12 12:18:07 2020 daemon.err dawn[2319]: Neigbor-Report is null!
Fri Jun 12 12:18:07 2020 daemon.err dawn[2319]: Neigbor-Report is null!
Fri Jun 12 12:18:15 2020 daemon.err dawn[2319]: not complete msg, len:1444, expected len:2016
Fri Jun 12 12:18:15 2020 kern.info kernel: [ 2924.963455] do_page_fault(): sending SIGSEGV to dawn for invalid write access to 00000000
Fri Jun 12 12:18:15 2020 kern.info kernel: [ 2924.971938] epc = 77e86a98 in libc.so[77e14000+97000]
Fri Jun 12 12:18:15 2020 kern.info kernel: [ 2924.977197] ra  = 77dfef4d in libubox.so[77dfa000+17000]

dmesg says:

[ 2924.963455] do_page_fault(): sending SIGSEGV to dawn for invalid write access to 00000000
[ 2924.971938] epc = 77e86a98 in libc.so[77e14000+97000]
[ 2924.977197] ra  = 77dfef4d in libubox.so[77dfa000+17000]

PolynomialDivision, 2020-06-12:
Do u installed full wpad? Could u look if your neighbor reports are correct set with

    ubus call dawn get_network

?

jonozzz, 2020-06-12:
Yes I'm using `wpad-mesh-openssl 2020-05-20-dd2daf08-2`, and the output looks ok, I see all my APs and clients in there.

PolynomialDivision, 2020-06-12:
> Yes I'm using `wpad-mesh-openssl 2020-05-20-dd2daf08-2`, and the output looks ok, I see all my APs and clients in there.

Especially this field in the AP `"neighbor_report": "27fb4ea6ef0900007e9d070603019f00",` is important. Did u already try the PR?

gongsearch, 2020-06-15:
For me the above error seems solved with fix/nullpointer. Now I sometimes get something like:

Mon Jun 15 12:36:25 2020 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 10:98:c3:04:74:ff 173 ack=1
Mon Jun 15 12:36:25 2020 daemon.notice hostapd: wlan1: BEACON-RESP-RX 10:98:c3:04:74:ff 0 04
Mon Jun 15 12:36:25 2020 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 48:88:ca:e1:c8:7e 171 ack=1
Mon Jun 15 12:36:25 2020 daemon.notice hostapd: wlan1: BEACON-RESP-RX 48:88:ca:e1:c8:7e 171 04 0000000000000000000000008000000000000000000000000000
Mon Jun 15 12:36:25 2020 daemon.err dawn[1430]: Received NULL MAC! Client is strange!
Mon Jun 15 12:36:25 2020 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 94:65:2d:2f:df:1b 172 ack=1
Mon Jun 15 12:36:30 2020 kern.info kernel: [ 3795.226593] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77dc9000
Mon Jun 15 12:36:30 2020 kern.info kernel: [ 3795.235018] epc = 77e56a8c in libc.so[77de4000+97000]
Mon Jun 15 12:36:30 2020 kern.info kernel: [ 3795.240184] ra  = 00405049 in dawn[400000+c000]

Ian-Clowes, 2020-07-10:
I was seeing what may be that 2nd style pf SEGV yesterday.  Via gdb I found a line that was using memcpy() to copy a maximum length neighbor report.  Looked a bit suspicious so I changed it to strncpy() and my crashes went away.  Noy sure how to check if the otiginal intent of the copy is eotking though.  It's included in a PR I'm hoping tp submit in next day or twp.

PolynomialDivision, 2020-07-10:
> I was seeing what may be that 2nd style of SEGV yesterday. Via gdb I found a line that was using memcpy() to try and copy a maximum length neighbor report. Looked a bit suspicious so I changed it to strncpy() and my crashes went away. Not sure how to check if the original intent of the copy is working though. It's included in a PR I'm hoping tp submit in next day or twp.

Ah thanks! I will have a look at this. Nice hint.

Ian-Clowes, 2020-07-10:
Just double checked that as I'm writing the commit comment.  It was copying the client signature in dump_client() that I chnaged to strncpy().

PolynomialDivision, 2020-08-08:
Is this issue still existing in latest dawn master?

gongsearch, 2020-08-09:
For me the problem is gone, thanks. :)

closed 2020-08-09 by PolynomialDivision