berlin-open-wireless-lab/DAWN #109: Problem compiling with GCC 10

finasfinas, 2020-07-12:
When trying GCC 10 ( it was enabled today on OpenWRT master ) it fails to build with the following error:~
`make[6]: Entering directory '/home/finas/OpenWRT.WPA8630/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4'
[ 11%] Building C object CMakeFiles/umdns.dir/service.c.o
/home/finas/OpenWRT.WPA8630/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c: In function 'service_load_blob':
/home/finas/OpenWRT.WPA8630/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c:240:10: error: 'strcpy' offset 6 from the object at 'b' is out of the bounds of referenced subobject 'name' with type 'uint8_t[]' {aka 'unsigned char[]'} at offset 6 [-Werror=array-bounds]
  240 |  s->id = strcpy(d_id, blobmsg_name(b));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/finas/OpenWRT.WPA8630/staging_dir/target-mips_24kc_musl/usr/include/libubus.h:23,
                 from /home/finas/OpenWRT.WPA8630/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c:23:
/home/finas/OpenWRT.WPA8630/staging_dir/target-mips_24kc_musl/usr/include/libubox/blobmsg.h:42:10: note: subobject 'name' declared here
   42 |  uint8_t name[];
      |          ^~~~
cc1: all warnings being treated as errors
CMakeFiles/umdns.dir/build.make:131: recipe for target 'CMakeFiles/umdns.dir/service.c.o' failed
make[6]: *** [CMakeFiles/umdns.dir/service.c.o] Error 1
make[6]: Leaving directory '/home/finas/OpenWRT.WPA8630/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4'
CMakeFiles/Makefile2:92: recipe for target 'CMakeFiles/umdns.dir/all' failed
make[5]: *** [CMakeFiles/umdns.dir/all] Error 2
`

Ian-Clowes, 2020-07-12:
That looks like an error building umdns doesn't it?

PolynomialDivision, 2020-07-12:
> That looks like an error building umdns doesn't it?

Yep. That is an issue umdns.

finasfinas, 2020-07-12:
> 
> 
> > That looks like an error building umdns doesn't it?
> 
> Yep. That is an issue umdns.

finasfinas, 2020-07-12:
> 
> 
> > > That looks like an error building umdns doesn't it?
> > 
> > 
> > Yep. That is an issue umdns.

I thought they were related. So should I open an issue with that project?

closed 2020-07-12 by finasfinas

PolynomialDivision, 2020-07-12:
> I thought they were related. So should I open an issue with that project?

Yep. I would suggest the mailing-list. :)

finasfinas, 2020-07-12:
is this the repository?
https://github.com/grover/uMDNS

PolynomialDivision, 2020-07-12:
> is this the repository?
> https://github.com/grover/uMDNS

No. umdns is not maitnained on github. Just use official openwrt devel mailing list. ;)

finasfinas, 2020-07-12:
> 
> 
> > is this the repository?
> > https://github.com/grover/uMDNS
> 
> No. umdns is not maitnained on github. Just use official openwrt devel mailing list. ;)

I really don't like mailing lists and don't want to register. Will wait for someone else to report it. :/

PolynomialDivision, 2020-07-13:
I will reporte it.

```
[ 11%] Building C object CMakeFiles/umdns.dir/service.c.o
/tmp/build/nick/openwrt/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c: In function 'service_load_blob':
/tmp/build/nick/openwrt/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c:240:10: error: 'strcpy' offset 6 from the object at 'b' is out of the bounds of referenced subobject 'name' with type 'uint8_t[]' {aka 'unsigned char[]'} at offset 6 [-Werror=array-bounds]
  240 |  s->id = strcpy(d_id, blobmsg_name(b));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/build/nick/openwrt/staging_dir/target-mips_24kc_musl/usr/include/libubus.h:23,
                 from /tmp/build/nick/openwrt/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c:23:
/tmp/build/nick/openwrt/staging_dir/target-mips_24kc_musl/usr/include/libubox/blobmsg.h:42:10: note: subobject 'name' declared here
   42 |  uint8_t name[];
      |          ^~~~
cc1: all warnings being treated as errors
make[6]: *** [CMakeFiles/umdns.dir/service.c.o] Error 1
```

finasfinas, 2020-07-13:
> I will reporte it.
>
> ```
> [ 11%] Building C object CMakeFiles/umdns.dir/service.c.o
> /tmp/build/nick/openwrt/build_dir/target-mips_24kc_musl/umdns-2020-06-08-d13290b4/service.c: In function [...]

many thanks!