berlin-open-wireless-lab/DAWN #113: hearingmap: already connected clients

PolynomialDivision, 2020-08-06:
In the current version clients get kicked if they have no value in the probe entry database ([code](https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/storage/datastorage.c#L413-L420)).

We have "two magic numbers"
- **-1**: no entry found -> kick
- **-2**: very bad ap

Solution?
- don't kick
- craft probe entry

I discussed with @Ian-Clowes that we will remove that "magic numbers" so all socres will be > 0.
Further, I prefer the solution to craft a probe entry.

XiaoliChan, 2020-08-26:
Hearingmap also cause kick function could not work better (`Chosen AP` function could not do choose best AP correctly).

**Here is my ap kick client logs**

#Note : 02:9E:2F:C3:BD:C5 is my phone_

```
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: -------- KICKING CLIENTS!!!---------
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: EVAL 20:76:93:3D:81:A2
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: Calculating own score!
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: Score: 100 of:
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 02:9E:2F:C3:BD:C5, signal: -63, freq: 5180, counter: 1, vht: 1, min_rate: 34, max_rate: 44
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: Own Score! Skipping!
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 02:9E:2F:C3:BD:C5, signal: -63, freq: 5180, counter: 1, vht: 1, min_rate: 34, max_rate: 44
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: Chosen AP
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: AP is best. Client will stay:
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 02:9E:2F:C3:BD:C5, freq: 5180, ht_supported: 1, vht_supported: 1, ht: 1, vht: 1, kick: 0
Wed Aug 26 23:33:43 2020 daemon.info dawn[3273]: ---------------------------
```

PolynomialDivision, 2020-08-26:
> could not work better

? U mean it works good?

In your log is described that no other hearing map entry exists. :/
Does your wifi device report back 802.11k beacon frames? So do u have rcpi or rsni? Then I would make a quick algorithm adaption that uses this values. Then u are able to steer the client successfully.

PolynomialDivision, 2020-08-26:
Do u have full wpad installed?

XiaoliChan, 2020-08-26:
> ? U mean it works good?

 Sry, I type worng word, I mean "it did not work good".😂

> In your log is described that no other hearing map entry exists. :/
> Does your wifi device report back 802.11k beacon frames? So do u have rcpi or rsni? Then I would make a quick algorithm adaption that uses this values. Then u are able to steer the client successfully.

I think my AP report back the beacon frames, because I saw a lot of bunch of this

```
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 4, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 4, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 194, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 194, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 181, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 181, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 4, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 4, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 3, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 3, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3D:81:A2, freq: 5180, ht: 1, vht: 1, chan_utilz: 3, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933d81a2ef1900008024090603022a00
Thu Aug 27 00:21:26 2020 daemon.info dawn[3273]: ssid: Hydrogen-5G, bssid_addr: 20:76:93:3E:3A:F2, freq: 5260, ht: 1, vht: 1, chan_utilz: 37, col_d: -1, bandwidth: -1, col_count: 117 neighbor_report: 933e3af2ef1900008034090603023a00
```

XiaoliChan, 2020-08-26:
Well, I executed this  `ubus call dawn get_hearing_map`  on the ap , and it reported `rcpi` and `rsni` = `-1`

```
        "Hydrogen-5G": {
                "02:9E:2F:C3:BD:C5": {
                        "20:76:93:3D:81:A2": {
                                "signal": -77,
                                "rcpi": -1,
                                "rsni": -1,
                                "freq": 5180,
                                "ht_capabilities": true,
                                "vht_capabilities": true,
                                "channel_utilization": 4,
                                "num_sta": 1,
                                "ht_support": true,
                                "vht_support": true,
                                "score": 100
                        }
                },
```

XiaoliChan, 2020-08-26:
> Do u have full wpad installed?

Yes, my all ap get `wpad-openssl` installed.

PolynomialDivision, 2020-08-26:
> `rcpi` and `rsni` = `-1`

`-1` is currently our magic number for no entry... ^^
So your client did not report any measurement back? You have `option update_beacon_reports` activated or? 
You can set it to `option update_beacon_reports '60'` or similar.

PolynomialDivision, 2020-08-26:
You can try to find the right 802.11k parameter for your client by using the ubus cmd

    ubus call hostapd.wlan0 rrm_beacon_req '{"addr":"00:xx:xx:xx:xx:xx", "op_class":0, "channel":1,"duration":1,"mode":2,"bssid":"ff:ff:ff:ff:ff:ff", "ssid":""}'

Try changing
- mode
- op_class
- duration
- channel
- ...

Further, you have to set `log_level` in `/etc/config/wireless` to 
```
option log_level '0'
```

Further, stop dawn.

If u send this ubus cmd to your client and then see weird random strings, you found the correct 802.11k parameters.

PolynomialDivision, 2020-08-26:
I thought about steering a client to 5 GHz if an AP is a dualband and the rssi is about some threshold.
Then we don't need a hearing map entry, but we can not be sure, if the client has a good 5 GHz signal strength.

XiaoliChan, 2020-08-26:
> So your client did not report any measurement back?  You have option update_beacon_reports activated or?

I check the log again, it did not seen any measurement come from my device.
I think my client did not support 802.11k 😢

PolynomialDivision, 2020-08-26:
>  I think my client did not support 802.11k cry

:/

U can check this invoking

    ubus call hostapd.wlan0 get_clients

and look at the `rrm` array. If all values are `0` your device does not support 802.11k.

PolynomialDivision, 2020-08-26:
@XiaoliChan U renamed your 5 GHz WiFi to `Hydrogen-5G` and your 2.4 GHz Wifi is probably called differently, or?
Please rename that to the same name and try again. Roaming happens only between the same ssid-named networks.
That could be the reason why you only have 1 entry in your hearingmap.

XiaoliChan, 2020-08-26:
> 
> U can check this invoking
> 
> ```
> ubus call hostapd.wlan0 get_clients
> ```
> 
> and look at the `rrm` array. If all values are `0` your device does not support 802.11k.

I try it now.

PolynomialDivision, 2020-08-26:
Your client tries to stick as long as it can to the same ssid.

XiaoliChan, 2020-08-26:
> @XiaoliChan U renamed your 5 GHz WiFi to `Hydrogen-5G` and your 2.4 GHz Wifi is probably called differently, or?
Yes, my 5GHz WiFi call `Hydrogen-5G`, and my 2.4GHz wifi call `Hydrogen-2.4G`

XiaoliChan, 2020-08-26:
@PolynomialDivision Here is my rrm result

```
root@AP-LivingRoom:/# ubus call hostapd.wlan1 get_clients
{
        "freq": 5180,
        "clients": {
                "02:9e:2f:c3:bd:c5": {
                        "auth": true,
                        "assoc": true,
                        "authorized": true,
                        "preauth": false,
                        "wds": false,
                        "wmm": true,
                        "ht": true,
                        "vht": true,
                        "wps": false,
                        "mfp": false,
                        "rrm": [
                                115,
                                16,
                                145,
                                0,
                                4
                        ],
                        "aid": 2,
                        "signature": "wifi4|probe:0,1,45,127,191,221(0050f2,8),255,127,255,221(506f9a,22),221(8cfdf0,1),htcap:01ef,htagg:13,htmcs:0000ffff,vhtcap:339199fa,vhtrxmcs:030cfffa,vhttxmcs:030cfffa,extcap:00000a02004000000001|assoc:0,1,33,45,48,54,59,70,127,191,221(0050f2,2),221(8cfdf0,1),htcap:01ef,htagg:13,htmcs:0000ffff,vhtcap:338091b2,vhtrxmcs:030cfffa,vhttxmcs:030cfffa,txpow:1708,extcap:0400080200000040"
                }
        }
}
root@AP-LivingRoom:/# 
```

PolynomialDivision, 2020-08-26:
> Yes, my 5GHz WiFi call `Hydrogen-5G`, and my 2.4GHz wifi call `Hydrogen-2.4G`

Then no roaming between 5 GHz and 2.4 GHz will appear frequently. ;)
And DAWN does not steer a client between different ssids. That is sometihng, the wifi stack is not allowing.

PolynomialDivision, 2020-08-26:
> Here is my rrm result

Looks like supporting 802.11k.

XiaoliChan, 2020-08-26:
> Then no roaming between 5 GHz and 2.4 GHz will appear frequently. ;)

 Because my building have lots of people using 2.4 GHz, that was really bad, and I use 2.4GHz for some old device.

PolynomialDivision, 2020-08-26:
> Because my building have lots of people using 2.4 GHz, that was really bad, and I use 2.4GHz for some old device.

Okay. :) That is something DAWN would take care off, because it always prefers 5 GHz APs if they have an reasonable signal strength. (But there is still this hearing map problem :/ )
If your 5 GHz cells overlap it is also fine and dawn would steer the clients between the different 5 GHz radios of your different APs.

Possible reasons, you only have one entry in your database:
- mac randomization while scanning
- no background scanning of your wifi device
(this should be solved with 802.11k and u don't have to care anymore)

I would suggest you try to find the setting for your client device to which it responds to a beacon report, and I will change some code to allow some rudimentary steering decisions based on the reported values.

PolynomialDivision, 2020-08-26:
Just looked at your pictures and u have a channel utilization of nearly 60% without any client connected in AP-Room1. O.O

XiaoliChan, 2020-08-26:
> Just looked at your pictures and u have a channel utilization of nearly 60% without any client connected in AP-Room1. O.O

Yes, because in mainland china, many people are using old WiFi device, such as old TP-LINK, Mercury and so on, they don't care about WiFi channel utilization

XiaoliChan, 2020-08-26:
So, maybe in the future, I will disable 2.4GHz. 😂

XiaoliChan, 2020-08-26:
I saw this come from my notebook, but I still can not see any probe message come from my phone.

`Thu Aug 27 01:43:38 2020 daemon.info dawn[3273]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"20:76:93:3E:3A:F2\",\"address\":\"6C:6A:77:5F:5F:73\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-54,\"freq\":5260,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":1}"}`

PolynomialDivision, 2020-08-26:
> Yes, because in mainland china, many people are using old WiFi device, such as old TP-LINK, Mercury and so on, they don't care about WiFi channel utilization (That was really terrible)

Lol. Why not buying Xiaomi or Redmi? Or even all that other cheap routers that I can only import from china. xD They just have the same SOC as all that TP-Link stuff and some even ship with OpenWrt. ^^ The routers that are based `MT7621A` SOC look nice.

If u want to see your phone being steered, I would suggest you try disabling mac randomization (if this is okay for your privacy needs) or open the wifi settings and scan actively (if no randomization is active).
Meanwhile I (or you? ;) ) will look through the 802.11 protocol and look which bits are indicating which 802.11k support... ;)

PolynomialDivision, 2020-08-26:
At least DAWN does feed hostapd with `neighbor_report`s of all your APs, allowing the client to ask APs for possible roaming candidates. So it could be that your roaming is a bit smoother. xD

XiaoliChan, 2020-08-26:
> Lol. Why not buying Xiaomi or Redmi? Or even all that other cheap routers that I can only import from china. xD They just have the same SOC as all that TP-Link stuff and some even ship with OpenWrt. ^^ The routers that are based `MT7621A` SOC look nice.

I ask the question to my friend early (He also using a old tp-link device), He said "whatever you say, my wifi work very well, right?" 🤣

XiaoliChan, 2020-08-26:
I check this bellow, mac randomization was on, now I disable it.

XiaoliChan, 2020-08-26:
> I check this bellow, mac randomization was on, now I disable it.

Well, I saw some probe message come from my phone after I do this.

`
Thu Aug 27 02:09:42 2020 daemon.info dawn[3273]: Method new: probe : {"address":"58:85:a2:04:ab:91","target":"ff:ff:ff:ff:ff:ff","signal":-60,"freq":2472,"ht_capabilities":{"ht_capabilities_info":429,"supported_mcs_set":{"a_mpdu_params":19,"ht_extended_capabilities":0,"tx_bf_capability_info":0,"asel_capabilities":0,"supported_mcs_set":[255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}},"vht_capabilities":{"vht_capabilities_info":865180050,"vht_supported_mcs_set":{"rx_map":-6,"rx_highest":780,"tx_map":-6,"tx_highest":780}}}
`

`Thu Aug 27 02:11:20 2020 daemon.info dawn[3273]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"20:76:93:3E:3A:F2\",\"address\":\"58:85:A2:04:AB:91\",\"target\":\"20:76:93:3E:3A:F2\",\"signal\":-57,\"freq\":5260,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":1}"}
`

XiaoliChan, 2020-08-26:
I also think you should edit a wiki in this project.

PolynomialDivision, 2020-08-26:
Do u now have more than one entry for your handy?

> I also think you should edit a wiki in this project.

Yeah... ^^ Documentation is missing for some features.
Still you only have to opkg install dawn and u will see some result. ;)

> He said "whatever you say, my wifi work very well, right?" rofl

Works until u want to use IPv6... xD Just bought some EAP 225 and I was not able to use ipv6 xD

XiaoliChan, 2020-08-26:
> Do u now have more than one entry for your handy?

I will check it tomorrow, because now I in the bed. xD

> Works until u want to use IPv6... xD Just bought some EAP 225 and I was not able to use ipv6 (with stock firmware) xD

Yep, stock firmware is very important

In fact, ipv6 also is really bad to use in china, the network vendor did not optimize the ipv6 route and many website did not support ipv6 in china

PolynomialDivision, 2020-08-26:
> Yep, stock firmware is very important

You mean OpenWrt support is very important. ;)

XiaoliChan, 2020-08-27:
`58:85:A2:04:AB:91` is my phone true mac.

```
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: -------- KICKING CLIENTS!!!---------
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: EVAL 20:76:93:3D:81:A2
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Calculating own score!
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Score: 100 of:
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 04:72:95:DF:EF:C6, signal: -57, freq: 5180, counter: 1, vht: 1, min_rate: 0, max_rate: 0
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Own Score! Skipping!
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 04:72:95:DF:EF:C6, signal: -57, freq: 5180, counter: 1, vht: 1, min_rate: 0, max_rate: 0
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Chosen AP
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: AP is best. Client will stay:
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 04:72:95:DF:EF:C6, freq: 5180, ht_supported: 1, vht_supported: 1, ht: 1, vht: 1, kick: 0
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Calculating own score!
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Score: 100 of:
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 58:85:A2:04:AB:91, signal: -67, freq: 5180, counter: 4, vht: 1, min_rate: 0, max_rate: 0
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Own Score! Skipping!
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 58:85:A2:04:AB:91, signal: -67, freq: 5180, counter: 4, vht: 1, min_rate: 0, max_rate: 0
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: Chosen AP
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: AP is best. Client will stay:
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: bssid_addr: 20:76:93:3D:81:A2, client_addr: 58:85:A2:04:AB:91, freq: 5180, ht_supported: 1, vht_supported: 1, ht: 1, vht: 1, kick: 0
Thu Aug 27 11:17:13 2020 daemon.info dawn[3273]: ---------------------------
```