berlin-open-wireless-lab/DAWN #116: Segfault on activating/deactivating kicking option

namidairo, 2020-08-12:
```
Wed Aug 12 19:19:39 2020 kern.info kernel: [190163.211805] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 646c6f64
Wed Aug 12 19:19:39 2020 kern.info kernel: [190163.227958] epc = 77e36f24 in libc.so[77e09000+9c000]
Wed Aug 12 19:19:39 2020 kern.info kernel: [190163.233163] ra  = 77d956df in libuci.so[77d94000+16000]
```

Activating/deactivating kicking in luci-app-dawn seems to trigger this on the other dawn instance running on a different router, requiring a restart of dawn on the affected nodes.

Using commit 50d54a625366cffd48f0ec3f23456d5a04c45fd3. (So up to date with master at the moment)

PolynomialDivision, 2020-08-12:
You can also trigger this by

    ubus dawn reload_config

I can confirm that the reconfiguration of other APs will lead to a crash.

PolynomialDivision, 2020-08-14:
The issue is in the reset function

https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/utils/dawn_uci.c#L191-L199

babarosaM33, 2020-08-17:
i ran dawn with gdb for one day, but could not get the crash. so i could not catch the bug.

babarosaM33, 2020-08-17:
the routine to compile dawn with gdb

1. download openwrt sdk.

2. edit rules.mk at the root directory. enable -g3 option
3. disable config_sstrip and enable config no strip
4. build

babarosaM33, 2020-08-17:
btw, my router only has 64MB memory, as a result, it could not run Valgrind to detect memory leak for memory limit. since openwrt's glibc does not have sanitize enabled, i could not check potential memory leak with automatic tools.

XiaoliChan, 2020-08-20:
@PolynomialDivision  Hello guy. I have a question, I want to kick device without check device bit rates, so what permanent should I set for `bandwidth_threshold` ?