berlin-open-wireless-lab/DAWN #117: Dawn Segfault, 4 AP configuration

morhimi, 2020-08-15:
`Sat Aug 15 14:34:03 2020 kern.info kernel: [41490.149494] do_page_fault(): sending SIGSEGV to dawn for invalid write access to 00000000
Sat Aug 15 14:34:03 2020 kern.info kernel: [41490.157830] epc = 77e56f50 in libc.so[77e29000+9c000]
Sat Aug 15 14:34:03 2020 kern.info kernel: [41490.162966] ra  = 77db55b1 in libuci.so[77db4000+16000]
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: A runtime error occurred: [string "[string]"]:8: bad argument #1 to 'pairs' (table expected, got nil)
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: stack traceback:
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	[C]: in function 'pairs'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	[string "[string]"]:8: in main chunk
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: stack traceback:
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	[C]: in function 'error'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	(tail call): ?
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:10: in function 'render'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: stack traceback:
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	[C]: in function 'error'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	(tail call): ?
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/dispatcher.lua:1451: in function '_cbi'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
Sat Aug 15 14:35:11 2020 daemon.err uhttpd[1475]: 	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>
Sat Aug 15 14:35:18 2020 daemon.err uhttpd[1475]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.`

babarosaM33, 2020-08-16:
try to keep  one ap with luci-dawn-app installed.

two ap setup of mine is working well for 4 days.

morhimi, 2020-08-25:
@PolynomialDivision I think this issue can be closed, with the latest DAWN there are no crashes in the last few days

closed 2020-08-25 by PolynomialDivision