berlin-open-wireless-lab/DAWN #122: 802.11k neighbor list broken

ddimension, 2020-09-02:
It seems, that somehow the properties in the neighborlist are broken.
The neiborborlist of 5Ghz:
`
ubus call hostapd.wap-knet1 rrm_nr_list
{
	"list": [
		[
			"ba:a4:b4:d0:b1:53",
			"kalnet",
			"b4d0b153ff1900008028090603022a00"
		],
`

The own record of the 2Ghz device:
`
ubus call hostapd.wap-knet0 rrm_nr_get_own
{
	"value": [
		"ba:a4:b4:d0:b1:53",
		"kalnet",
		"baa4b4d0b153ff1900008028090603022a00"
	]
}
`
Wireshark also tells it is broken:
[image]

Kind regards.

ddimension, 2020-09-02:
Also I noticed the neighbor reports include all SSIDs managed by dawn. Wouldn't it be better to just announces neighbors for the SSID?

PolynomialDivision, 2020-09-02:
> Also I noticed the neighbor reports include all SSIDs managed by dawn. Wouldn't it be better to just announces neighbors for the SSID?

Of course! :)

Could u try setting them manually by using

    ubus call hostapd.wlan0 rrm_nr_get_own

and 

    ubus call hostapd.wlan0 rrm_nr_set ...

Thanks for testing! When I have time I will check this, too.

ddimension, 2020-09-02:
Hi!

I tried it, it works fine when not doing it over dawn. Wireshark shows no errors in that case. I also worked on a central controller, but with more management in mind.
https://github.com/ddimension/apman
But it's alpha stage..

I'm thinking of combining it with dawn, but I just saw you work on an central controller.

PolynomialDivision, 2020-09-02:
> I tried it, it works fine when not doing it over dawn.

Thanks! The logic is happening here: https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/utils/ubus.c#L1558-L1588

I will have a look. Since it failed silently I did not notice that. Maybe even the fast bss transition frame will be wrong. :S

I just wanted to measure all my clients of a broadcast domain on a central server outside that broadcast domain. ;)
Not building anything special, since I lack on time. If u want to add a second centralized mode, and it stays compatible with decentralized mode, we can talk about merging that!

ddimension, 2020-09-02:
What I can see in the logs is this:
Copied Neighborreport: b4d0b153ff1900008028090603022a00,

It seems that this already happens when reading in ubus_get_rrm_cb() !? The first 2 Bytes are missing.

Abount integration I will do further tests after I have dawn running. BTW, great work!

PolynomialDivision, 2020-09-03:
> BTW, great work!

Thanks.

I can reproduce the issue. Indeed, 2 bytes are missing.

PolynomialDivision, 2020-09-03:
@ddimension I think I fixed it. Could u try: https://github.com/berlin-open-wireless-lab/DAWN/pull/123 ?

ddimension, 2020-09-03:
Tried it, bug is fixed. Thank you!

closed 2020-09-03 by ddimension