berlin-open-wireless-lab/DAWN #125: strange behavior on tp-link archer c7

WiredLife, 2020-09-06:
hello, i have a dawn related problem which only affects the 5 ghz band on archer c7 from tp-link (snapshot r14397)
if dawn is running i got connection problems with hanging and stuttering video streams
if i kill the dawn process its running without problems

`vht80
Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a IEEE 802.11: authenticated
Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a IEEE 802.11: associated (aid 1)
Sun Sep  6 21:05:21 2020 daemon.notice hostapd: wlan0-1: AP-STA-CONNECTED 34:29:12:e1:20:9a
Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a RADIUS: starting accounting session 2FA7BF4866CB57BD
Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a WPA: pairwise key handshake completed (RSN)
Sun Sep  6 21:05:29 2020 daemon.err uhttpd[1342]: luci: accepted login on / for root from 192.168.176.35
Sun Sep  6 21:05:40 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 1 ack=1
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042408c04a00e454ca01b972b15b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d11304201ac24a00e454ca01b8a4b15b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042414c64a00e454ca01bdd6b15b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042214b4fbe487067b012b95bd5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d11304201ab6fbe497067b0148a5bd5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042214b6fbe4b7067b015dc5bd5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00307ac8e65b000000004d060424206666b38a470c010cf3e65b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00307ac8e65b000000004d060426206266b38a470c013524e75b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00307ac8e65b000000004d060426206466b38a470c012150e85b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0064ced4f95b000000008a01045c50c66e1f4fcbb501132efb5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00407c7dfd5b00000000a00004443a96f652ffc96e01e8a6fd5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00646882fe5b000000005f00045a54c26e1f4fcbb501add2fe5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00646882fe5b000000005f00045a50c46e1f4fcbb501d657ff5b
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00745140155c00000000affa0426206466b37ba06601f193155c
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00745140155c00000000affa042c206666b37ba06601eec5155c
Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00745140155c00000000affa042c246266b37ba06601e5f7155c
Sun Sep  6 21:06:00 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 2 ack=1
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 002459e7e15c0000000086c7043c32c04a00e454ca01b521e25c
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 002459e7e15c0000000086c7044036c24a00e454ca01cb53e25c
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 002459e7e15c0000000086c7043c36c64a00e454ca01cd85e25c
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 003022f6175d0000000002ba0424206466b38a470c0104ff185d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 003022f6175d0000000002ba0422206666b38a470c011531195d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 003022f6175d0000000002ba042c206266b38a470c010163195d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 006477022b5d000000003fb504785ac66e1f4fcbb501426d2d5d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0040bebc2e5d0000000050b404584c90f652ffc96e0164812f5d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0064f9c22f5d000000000fb4047456c46e1f4fcbb5010f07305d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0064f9c22f5d000000000fb4047858c26e1f4fcbb501c111315d
Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0074d18b465d000000005dae043a326266b37ba06601b4a6465d
Sun Sep  6 21:06:20 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 3 ack=1
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 00247df3175e00000000037a042620c04a00e454ca01c310195e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 00247df3175e00000000037a04261ac24a00e454ca01cc42195e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 00247df3175e00000000037a042420c64a00e454ca01cc74195e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0030ddfe4d5e00000000806c042c246466b38a470c01f35d4e5e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0030ddfe4d5e00000000806c042e246666b38a470c0101904e5e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0030ddfe4d5e00000000806c042c266266b38a470c01ffc14e5e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0064dd09615e00000000bd67047a5cc66e1f4fcbb50187cc625e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0040f0c0645e00000000cf6604564c90f652ffc96e01e9df645e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0040f0c0645e00000000cf6604564c92f652ffc96e01f511655e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0040f0c0645e00000000cf6604564a96f652ffc96e011c44655e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 00648bc8655e000000008d6604785cc26e1f4fcbb501fc70665e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 00648bc8655e000000008d66047a5cc46e1f4fcbb50143f6665e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0074ad8b7c5e00000000dd600438366466b37ba0660178317d5e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0074ad8b7c5e00000000dd600436306666b37ba0660186637d5e
Sun Sep  6 21:06:39 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 3 00 0099c96fa95e00000000a455041c1af09fc24cbba001c2f2b15e
Sun Sep  6 21:06:40 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 4 ack=1
Sun Sep  6 21:06:41 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 4 07
Sun Sep  6 21:07:00 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 5 ack=1
Sun Sep  6 21:07:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 5 00 0024823176600000000073e204201ac04a00e454ca01434e7760
Sun Sep  6 21:07:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 5 00 0024823176600000000073e2042014c24a00e454ca014e807760
Sun Sep  6 21:07:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 5 00 0024823176600000000073e204201ac64a00e454ca014bb27760
Sun Sep  6 21:07:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 5 00 0030ff3fac6000000000f0d40422146466b38a470c016f9bac60
Sun Sep  6 21:07:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 5 00 0030ff3fac6000000000f0d404241a6666b38a470c0177cdac60
[... cut ...]
Sun Sep  6 21:35:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 71 00 00747aa4956300000000961a0422206666b37ba06601a3169663
Sun Sep  6 21:35:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 71 00 00747aa4956300000000961a0426206266b37ba06601a0489663
Sun Sep  6 21:36:00 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 72 ack=1
Sun Sep  6 21:36:00 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 72 ef
Sun Sep  6 21:36:20 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 73 ack=1`

PolynomialDivision, 2020-09-06:
> hello, i have a dawn related problem which only affects a huawei mate 20 pro connected to the 5 ghz band on archer c7 from tp-link (snapshot r14397 latest dawn)
> if dawn is running i got connection problems with hanging and stuttering video streams
> if i kill the dawn process its running without problems
> 
> > vht80
> > Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a IEEE 802.11: authenticated
> > Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a IEEE 802.11: associated (aid 1)
> > Sun Sep  6 21:05:21 2020 daemon.notice hostapd: wlan0-1: AP-STA-CONNECTED 34:29:12:e1:20:9a
> > Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a RADIUS: starting accounting session 2FA7BF4866CB57BD
> > Sun Sep  6 21:05:21 2020 daemon.info hostapd: wlan0-1: STA 34:29:12:e1:20:9a WPA: pairwise key handshake completed (RSN)
> > Sun Sep  6 21:05:29 2020 daemon.err uhttpd[1342]: luci: accepted login on / for root from 192.168.176.35
> > Sun Sep  6 21:05:40 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 1 ack=1
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042408c04a00e454ca01b972b15b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d11304201ac24a00e454ca01b8a4b15b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042414c64a00e454ca01bdd6b15b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042214b4fbe487067b012b95bd5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d11304201ab6fbe497067b0148a5bd5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0024a1bbb05b00000000d113042214b6fbe4b7067b015dc5bd5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00307ac8e65b000000004d060424206666b38a470c010cf3e65b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00307ac8e65b000000004d060426206266b38a470c013524e75b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00307ac8e65b000000004d060426206466b38a470c012150e85b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 0064ced4f95b000000008a01045c50c66e1f4fcbb501132efb5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00407c7dfd5b00000000a00004443a96f652ffc96e01e8a6fd5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00646882fe5b000000005f00045a54c26e1f4fcbb501add2fe5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00646882fe5b000000005f00045a50c46e1f4fcbb501d657ff5b
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00745140155c00000000affa0426206466b37ba06601f193155c
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00745140155c00000000affa042c206666b37ba06601eec5155c
> > Sun Sep  6 21:05:59 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 1 00 00745140155c00000000affa042c246266b37ba06601e5f7155c
> > Sun Sep  6 21:06:00 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 2 ack=1
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 002459e7e15c0000000086c7043c32c04a00e454ca01b521e25c
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 002459e7e15c0000000086c7044036c24a00e454ca01cb53e25c
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 002459e7e15c0000000086c7043c36c64a00e454ca01cd85e25c
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 003022f6175d0000000002ba0424206466b38a470c0104ff185d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 003022f6175d0000000002ba0422206666b38a470c011531195d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 003022f6175d0000000002ba042c206266b38a470c010163195d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 006477022b5d000000003fb504785ac66e1f4fcbb501426d2d5d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0040bebc2e5d0000000050b404584c90f652ffc96e0164812f5d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0064f9c22f5d000000000fb4047456c46e1f4fcbb5010f07305d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0064f9c22f5d000000000fb4047858c26e1f4fcbb501c111315d
> > Sun Sep  6 21:06:19 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 2 00 0074d18b465d000000005dae043a326266b37ba06601b4a6465d
> > Sun Sep  6 21:06:20 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 3 ack=1
[... cut ...]
> > Sun Sep  6 21:36:00 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 72 ack=1
> > Sun Sep  6 21:36:00 2020 daemon.notice hostapd: wlan0-1: BEACON-RESP-RX 34:29:12:e1:20:9a 72 ef
> > Sun Sep  6 21:36:20 2020 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 34:29:12:e1:20:9a 73 ack=1

Try setting `option update_beacon_reports '20'` to `option update_beacon_reports '0'`.
Your smartphone is answering on every beacon report request the AP send to it. And that is every 20s. Set it higher or set it to 0 that disables it. ;)

PolynomialDivision, 2020-09-06:
Could u give me the json output of your smartphone from

    ubus call dawn get_hearing_map

?

I would like to know which values it reports on a beacon request.

WiredLife, 2020-09-06:
ubus call dawn get_hearing_map only shows me the 2,4 ghz band
maybe its not working with the same ssid on both bands?

WiredLife, 2020-09-07:
"34:29:12:E1:20:9A": {
                        "C2:6E:1F:4F:CB:B5": {
                                "signal": 0,
                                "rcpi": 114,
                                "rsni": 92,
                                "freq": 5500,
                                "ht_capabilities": false,
                                "vht_capabilities": false,
                                "channel_utilization": 0,
                                "num_sta": 1,
                                "ht_support": true,
                                "vht_support": true,
                                "score": -2
--
                "34:29:12:E1:20:9A": {
                        "64:66:B3:7B:A0:66": {
                                "signal": 0,
                                "rcpi": 54,
                                "rsni": 44,
                                "freq": 5580,
                                "ht_capabilities": false,
                                "vht_capabilities": false,
                                "channel_utilization": 2,
                                "num_sta": 0,
                                "ht_support": true,
                                "vht_support": false,
                                "score": -2
--
                "34:29:12:E1:20:9A": {
                        "96:F6:52:FF:C9:6E": {
                                "signal": 0,
                                "rcpi": 78,
                                "rsni": 76,
                                "freq": 5320,
                                "ht_capabilities": false,
                                "vht_capabilities": false,
                                "channel_utilization": 2,
                                "num_sta": 0,
                                "ht_support": true,
                                "vht_support": false,
                                "score": -2

looks like dawn cant get the signal level?

PolynomialDivision, 2020-09-07:
>  maybe its not working with the same ssid on both bands?

This should work. 

Can u paste the output of `ubus call dawn get_network`?
I believe your APs don't find each other.

WiredLife, 2020-09-07:
^^ this output is too long
trust me, they see each other, checked it via dawn luci network view on every single ap
i have 17 (dumb) aps and 6 wlans per ap (3 on each band), 3 separate vlans
no problems with other clients
only with this huawei mate 20 pro

PolynomialDivision, 2020-09-07:
I don't understand the highlighting of your output.
Contains the hearing map of one particular AP all 3 entries?

This would be very nice! :) And your phone is very special, since it repors rcpi and rsni! :D Thats very good.

Maybe you could even test a new feature that steers based on those values:
https://github.com/berlin-open-wireless-lab/DAWN/pull/126
**you need to add config values manually**

I'm a bit tired today. I will have a detailed look tomorrow.
Did you set `option update_beacon_reports '360'` or something like this?

There are some 802.11k that can be tuned, too (e.g. the channel to scan, the mode, ...). I have look into the 802.11 standard again.

PolynomialDivision, 2020-09-07:
I added here some example in the commit message how to test beacon reports: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c8ef465e10f46fd017cc9c3e455cc368a2b9da2b

WiredLife, 2020-09-07:
on the network overview i can see all ssids on all bands on all aps
but on the hearing map one ssid is completly empty
and i cant see some connections to the other ssids

i set the value to 360, now waiting for the client to test it

i have no build environment to compile by myself, so i rely on snapshot package build from openwrt

i used the default dawn config on all aps until now because it worked for me "out of the box"
switched on -80 signal between the aps like a charm
only this one client has problems with hanging/stuttering streams and slow connection speed as i know
we have a big network and i'm not so deep into the wifi roaming/management topic

maybe i can text you on a messenger? my german is much better than my english xD

WiredLife, 2020-09-08:
My client says it reduced the stuttering/hanging to first data transmission, after that its running fine for 6 minutes (value 360) than it stutters again and repeats
Looks like the problems coming in sync with the masses of BEACON-RESP-RX

PolynomialDivision, 2020-09-08:
> maybe i can text you on a messenger? my german is much better than my english xD

Are you in openwrt devel irc, or do u have matrix messenger?

WiredLife, 2020-09-08:
> > maybe i can text you on a messenger? my german is much better than my english xD
> 
> Are you in openwrt devel irc, or do u have matrix messenger?

Irc is ok
With the other i'm not familiar

WiredLife, 2020-09-10:
Workaround: option update_beacon_reports '0'