berlin-open-wireless-lab/DAWN #127: crash/segfault on PowerPC

s-er-01, 2020-10-27:
on 'TP-LINK TL-WDR4900 v1' with openwrt master@r14777 and
with DAWN master@b639145 i get sometimes this:

```
[  300.707102] dawn[5513]: segfault (11) at 10 nip 1000da94 lr 1000da90 code 1 in dawn[10000000+14000]
[  300.716206] dawn[5513]: code: 3884f9e8 7c7c1b78 7cbd2b78 81228ff8 9121601c 39200000 48000475 7c7f1b79
[  300.725525] dawn[5513]: code: 41820034 7f83e378 48000665 38810018 <81230010> 7c7b1b78 7f83e378 7d2903a6
```
also:   
after that DAWN is not running anymore, but '/etc/init.d/dawn status' still shows 'running' - seems procd does not restart ist automatically.

seetings are default except these:
```
dawn.@network[0].broadcast_ip='127.0.0.1'
dawn.@network[0].broadcast_port='1025'
dawn.@network[0].server_ip='127.0.0.1'
dawn.@network[0].tcp_port='1026'
dawn.@network[0].network_option='2'
```

s-er-01, 2020-10-27:
maybe it makes sense with more context:
```
[  299.165475] fsl-gianfar soc@ffe00000:ethernet@b0000 eth0: Link is Down
[  300.707102] dawn[5513]: segfault (11) at 10 nip 1000da94 lr 1000da90 code 1 in dawn[10000000+14000]
[  300.716206] dawn[5513]: code: 3884f9e8 7c7c1b78 7cbd2b78 81228ff8 9121601c 39200000 48000475 7c7f1b79
[  300.725525] dawn[5513]: code: 41820034 7f83e378 48000665 38810018 <81230010> 7c7b1b78 7f83e378 7d2903a6
[  303.773395] fsl-gianfar soc@ffe00000:ethernet@b0000 eth0: Link is Up - 1Gbps/Full - flow control off
[  303.815988] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

PolynomialDivision, 2020-10-27:
Please turn on stdout in `/etc/init.d/dawn`. You will see a lot of garbage messages, but it gives me a hint where segfault is happening.

bittorf, 2020-11-08:
OK, i have now a testbed, where it reliable happens on
different architectures. I'am able and willing to test new versions.
Here a crash with the output of logread. (see also: [http://intercity-vpn.de/1.txt](full file))

```
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: --------APs------
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:DE:D9:EB, freq: 2432, ht: 1, vht: 0, chan_utilz: 61, col_d: -1, bandwidth: -1, col_count: 5 neighbor_report: 6666b3ded9ebef0900005105070603000500
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:DE:D9:EC, freq: 5200, ht: 1, vht: 0, chan_utilz: 7, col_d: -1, bandwidth: -1, col_count: 5 neighbor_report: 6666b3ded9ecef0900007328070603002800
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:E8:38:F3, freq: 5200, ht: 1, vht: 0, chan_utilz: 4, col_d: -1, bandwidth: -1, col_count: 5 neighbor_report: 6666b3e838f3ef0900007328070603002800
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:E8:38:F6, freq: 2432, ht: 1, vht: 0, chan_utilz: 29, col_d: -1, bandwidth: -1, col_count: 5 neighbor_report: 6666b3e838f6ef0900005105070603000500
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ssid: weimar.freifunk.net, bssid_addr: A0:F3:C1:74:8D:BA, freq: 2432, ht: 1, vht: 0, chan_utilz: 108, col_d: -1, bandwidth: -1, col_count: 5 neighbor_report: a0f3c1748dbaef0900005105070603000500
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ssid: weimar.freifunk.net, bssid_addr: A2:F3:C1:74:8D:BA, freq: 5200, ht: 1, vht: 0, chan_utilz: 11, col_d: -1, bandwidth: -1, col_count: 5 neighbor_report: a2f3c1748dbaef0900007328070603002800
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: ------------------
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: Received network message: {"method":"clients","data":"{\"freq\":5200,\"clients\":{},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"66:66:B3:E8:38:F3\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":4,\"neighbor_report\":\"6666b3e838f3ef0900007328070603002800\",\"iface\":\"wlan0-1\",\"hostname\":\"ayse-steve\"}"}
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: Network Method new: clients : {"method":"clients","data":"{\"freq\":5200,\"clients\":{},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"66:66:B3:E8:38:F3\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":4,\"neighbor_report\":\"6666b3e838f3ef0900007328070603002800\",\"iface\":\"wlan0-1\",\"hostname\":\"ayse-steve\"}"}
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: Received network message: {"method":"clients","data":"{\"freq\":5200,\"clients\":{},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"A2:F3:C1:74:8D:BA\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":10,\"neighbor_report\":\"a2f3c1748dbaef0900007328070603002800\",\"iface\":\"wlan1-1\",\"hostname\":\"regina2\"}"}
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: Network Method new: clients : {"method":"clients","data":"{\"freq\":5200,\"clients\":{},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"A2:F3:C1:74:8D:BA\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":10,\"neighbor_report\":\"a2f3c1748dbaef0900007328070603002800\",\"iface\":\"wlan1-1\",\"hostname\":\"regina2\"}"}
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: Received network message: {"method":"clients","data":"{\"freq\":2432,\"clients\":{\"e8:9e:b4:69:c2:57\":{\"auth\":true,\"assoc\":true,\"authorized\":true,\"preauth\":false,\"wds\":false,\"wmm\":true,\"ht\":true,\"vht\":false,\"wps\":false,\"mfp\":false,\"rrm\":[0,0,0,0,0],\"aid\":1,\"signature\":\"wifi4|probe:0,1,3,50|assoc:0,1,33,50,221(0050f2,2),45,htcap:01ad,htagg:03,htmcs:0000ffff,txpow:0d06\"}},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"A0:F3:C1:74:8D:BA\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":110,\"neighbor_report\":\"a0f3c1748dbaef0900005105070603000500\",\"iface\":\"wlan0-1\",\"hostname\":\"regina2\"}"}
Sun Nov  8 10:39:57 2020 daemon.info dawn[32195]: Network Method new: clients : {"method":"clients","data":"{\"freq\":2432,\"clients\":{\"e8:9e:b4:69:c2:57\":{\"auth\":true,\"assoc\":true,\"authorized\":true,\"preauth\":false,\"wds\":false,\"wmm\":true,\"ht\":true,\"vht\":false,\"wps\":false,\"mfp\":false,\"rrm\":[0,0,0,0,0],\"aid\":1,\"signature\":\"wifi4|probe:0,1,3,50|assoc:0,1,33,50,221(0050f2,2),45,htcap:01ad,htagg:03,htmcs:0000ffff,txpow:0d06\"}},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"A0:F3:C1:74:8D:BA\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":110,\"neighbor_report\":\"a0f3c1748dbaef0900005105070603000500\",\"iface\":\"wlan0-1\",\"hostname\":\"regina2\"}"}
Sun Nov  8 10:39:59 2020 daemon.info dawn[32195]: Method new: probe : {"address":"14:49:e0:63:a0:58","target":"ff:ff:ff:ff:ff:ff","signal":-78,"freq":2432,"ht_capabilities":{"ht_capabilities_info":-21503,"supported_mcs_set":{"a_mpdu_params":2,"ht_extended_capabilities":0,"tx_bf_capability_info":0,"asel_capabilities":0,"supported_mcs_set":[255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}} [... cut ...]
Sun Nov  8 10:40:07 2020 daemon.info dawn[32195]: mem-audit: Releasing (F) memory we hadn't registered (dawn-2020-09-03-b639145c/src/utils/msghandler.c:430)...
Sun Nov  8 10:40:12 2020 daemon.notice hostapd: wlan1-1: AP-STA-POLL-OK 78:00:9e:e8:96:ac
Sun Nov  8 10:40:19 2020 daemon.notice hostapd: wlan1-1: AP-STA-POLL-OK 88:87:17:da:cf:33
Sun Nov  8 10:40:37 2020 daemon.notice hostapd: wlan1-1: AP-STA-POLL-OK 88:87:17:da:cf:33
Sun Nov  8 10:40:52 2020 daemon.notice hostapd: wlan1-1: AP-STA-POLL-OK 78:00:9e:e8:96:ac

```

s-er-01, 2020-11-08:
sorry, i'am the same person with 2 accounts.
for the record: this crash was on a 'TP-LINK TL-WDR4900 v1' with
OpenWRT r14869 and DAWN @b639145ce90230e693c41e71624ea0c0798c424d (Sep3)

s-er-01, 2020-11-08:
here again, same versions, but on WDR4300:
full file: http://intercity-vpn.de/2.txt

```
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: 66:66:B3:DE:D9:EC, client_addr: 40:4E:36:9B:A8:8E, freq: 5200, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: 66:66:B3:E8:38:F6, client_addr: 78:00:9E:E8:96:AC, freq: 2432, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: 66:66:B3:E8:38:F6, client_addr: 88:87:17:DA:CF:33, freq: 2432, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: 66:66:B3:E8:38:F6, client_addr: C8:BC:C8:CF:F1:7C, freq: 2432, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: A0:F3:C1:74:8D:BA, client_addr: 00:67:E5:9B:03:BE, freq: 2432, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: A0:F3:C1:74:8D:BA, client_addr: 8C:A9:82:8E:FF:0C, freq: 2432, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: bssid_addr: A0:F3:C1:74:8D:BA, client_addr: E8:9E:B4:69:C2:57, freq: 2432, ht_supported: 1, vht_supported: 0, ht: 1, vht: 0, kick: 0
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ------------------
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: --------APs------
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:DE:D9:EB, freq: 2432, ht: 1, vht: 0, chan_utilz: 62, col_d: -1, bandwidth: -1, col_count: 7 neighbor_report: 6666b3ded9ebef0900005105070603000500
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:DE:D9:EC, freq: 5200, ht: 1, vht: 0, chan_utilz: 10, col_d: -1, bandwidth: -1, col_count: 7 neighbor_report: 6666b3ded9ecef0900007328070603002800
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:E8:38:F3, freq: 5200, ht: 1, vht: 0, chan_utilz: 8, col_d: -1, bandwidth: -1, col_count: 7 neighbor_report: 6666b3e838f3ef0900007328070603002800
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ssid: weimar.freifunk.net, bssid_addr: 66:66:B3:E8:38:F6, freq: 2432, ht: 1, vht: 0, chan_utilz: 29, col_d: -1, bandwidth: -1, col_count: 7 neighbor_report: 6666b3e838f6ef0900005105070603000500
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ssid: weimar.freifunk.net, bssid_addr: A0:F3:C1:74:8D:BA, freq: 2432, ht: 1, vht: 0, chan_utilz: 77, col_d: -1, bandwidth: -1, col_count: 7 neighbor_report: a0f3c1748dbaef0900005105070603000500
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ssid: weimar.freifunk.net, bssid_addr: A2:F3:C1:74:8D:BA, freq: 5200, ht: 1, vht: 0, chan_utilz: 13, col_d: -1, bandwidth: -1, col_count: 7 neighbor_report: a2f3c1748dbaef0900007328070603002800
Sun Nov  8 11:11:23 2020 daemon.info dawn[783]: ------------------
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: [Thread] : Removing old client entries!
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Received network message: {"method":"probe","data":"{\"bssid\":\"66:66:B3:E8:38:F6\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-79,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"66:66:B3:E8:38:F6\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-79,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Received network message: {"method":"probe","data":"{\"bssid\":\"66:66:B3:DE:D9:EB\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-84,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"66:66:B3:DE:D9:EB\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-84,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Received network message: {"method":"probe","data":"{\"bssid\":\"66:66:B3:DE:D9:EB\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-85,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"66:66:B3:DE:D9:EB\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-85,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Received network message: {"method":"probe","data":"{\"bssid\":\"66:66:B3:E8:38:F6\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-80,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":0}"}
Sun Nov  8 11:11:25 2020 daemon.info dawn[783]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"66:66:B3:E8:38:F6\",\"address\":\"14:49:E0:63:A0:58\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-80,\"freq\":2432,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilit [... cut ...]
Sun Nov  8 11:11:35 2020 daemon.info kalua: /usr/sbin/cron.wireless_: wifi_signal_get() dev: wlan0-1 mac: e8:9e:b4:69:c2:57 try: 1 signal: 0 file: '/tmp/stations_ap.txt-full-wlan0-1'
Sun Nov  8 11:11:35 2020 daemon.info kalua: /usr/sbin/cron.wireless_: iwevent() worker(6078) dev:wlan0-1 ch:5 MODE: del MAC: e8:9e:b4:69:c2:57 - native (no route del)
Sun Nov  8 11:11:43 2020 daemon.info hostapd: wlan0-1: STA e8:9e:b4:69:c2:57 IEEE 802.11: authenticated
Sun Nov  8 11:11:43 2020 daemon.info hostapd: wlan0-1: STA e8:9e:b4:69:c2:57 IEEE 802.11: associated (aid 1)
Sun Nov  8 11:11:43 2020 daemon.notice hostapd: wlan0-1: AP-STA-CONNECTED e8:9e:b4:69:c2:57
```

PolynomialDivision, 2020-11-08:
Something strange happens if u look at the `neighbor_report` line.
```
Sun Nov  8 11:11:33 2020 daemon.info dawn[783]: Network Method new: clients : {"method":"clients","data":"{\"freq\":5200,\"clients\":{},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"A2:F3:C1:74:8D:BA\",\"ssid\":\"weimar.freifunk.net\",\"ht_supported\":true,\"vht_supported\":false,\"ap_weight\":0,\"channel_utilization\":13,\"neighbor_report\":\"a2f3c1748dbaef0900007328070603002800\",\"iface\":\"wlan1-1\",\"hostname\":\"regina2\"}"}
Sun Nov  8 11:11:33 2020 kern.info kernel: [25166.912888] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 0000045a
Sun Nov  8 11:11:33 2020 kern.info kernel: [25166.921526] epc = 00403551 in dawn[400000+c000]
Sun Nov  8 11:11:33 2020 kern.info kernel: [25166.926300] ra  = 004035b5 in dawn[400000+c000]
Sun Nov  8 11:11:33 2020 daemon.info dawn[783]: ssid: , bssid_addr: 00:00:02:20:02:32, freq: 36865288, ht: 2, vht: 50, chan_utilz: 2010699768, col_d: 0, bandwidth: 36852272, col_count: 0 neighbor_report: wØ×Ðÿÿÿÿ
```

s-er-01, 2020-11-08:
for completness, the only things i have changed in config are:
dawn.@network[0].network_option='0'
dawn.@network[0].broadcast_ip='6.6.255.255'

a ping 6.6.255.255 shows, the everybody is answering:

```
root@F36-stube:~ :) ping 6.6.255.255
PING 6.6.255.255 (6.6.255.255): 56 data bytes
64 bytes from 6.6.66.19: seq=0 ttl=64 time=0.304 ms
64 bytes from 6.6.1.1: seq=0 ttl=64 time=2.117 ms (DUP!)
64 bytes from 6.6.67.236: seq=0 ttl=64 time=3.856 ms (DUP!)
64 bytes from 6.6.68.139: seq=0 ttl=64 time=4.505 ms (DUP!)
64 bytes from 6.6.66.3: seq=0 ttl=64 time=5.929 ms (DUP!)
```

PolynomialDivision, 2020-11-08:
I think the segfault happens in the "print aps" funtction when reading the wrong "neighbort_report" string.
```
Sun Nov  8 11:11:33 2020 daemon.info dawn[783]: ssid: , bssid_addr: 00:00:02:20:02:32, freq: 36865288, ht: 2, vht: 50, chan_utilz: 2010699768, col_d: 0, bandwidth: 36852272, col_count: 0 neighbor_report: wØ×Ðÿÿÿÿ
```
The values are just garbage.
That it important information:
```
 dawn.@network[0].network_option='0'
```
Could u please switch back to TCP option and see if it will fail?

s-er-01, 2020-11-08:
ok, will switch to tcp

PolynomialDivision, 2020-11-08:
I think I know why it fails.
We concentrated on the tcp socket:
https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/network/tcpsocket.c#L94-L195

https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/network/networksocket.c#L64-L85

PolynomialDivision, 2020-11-08:
@s-er-01 u have to be careful with umdns, since sometimes it is a bit buggy. For me it works all the time, but for some people they have to change something in umdns config.

https://bugs.openwrt.org/index.php?string=umdns

s-er-01, 2020-11-08:
when switching to TCP i dont see any outgoing packets?!

tcpdump -ni $LANDEV port 1025

```
root@F36-stube:~ :) uci show umdns
umdns.@umdns[0]=umdns
umdns.@umdns[0].jail='1'
umdns.@umdns[0].network='lan'
```

s-er-01, 2020-11-08:
ok, so i have to fix umdns first, i have an idea...

PolynomialDivision, 2020-11-08:
> ok, so i have to fix umdns first, i have an idea...

Aaaand? :)

bittorf, 2020-11-08:
still not working on my side, i need more time, sorry

s-er-01, 2020-11-14:
sorry, the problematic vxlan-support in netifd is a
showstopper for me, because umdns-service cannot run
in mesh-mode, but needs an overlay network in out setup:

bug:
https://bugs.openwrt.org/index.php?do=details&task_id=3426&string=vxlan&search_name=&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=

i will work on it and report back.

question:
If I understand it correctly, umdns-service detects avahi-services in the
network and delivers them using unicast to every router in the e.g. 'lan' network.
it makes these informations available via ubus.
dawn uses this list to hold a TCP-session open (in tcp-mode) to every dawn-router.
correct?

s-er-01, 2020-11-23:
I have managed to get my setup working in TCP-mode.
Now i have a new crash:

```
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable
Mon Nov 23 20:44:37 2020 kern.info kernel: [ 4641.530978] dawn[9424]: segfault (11) at 10 nip 1000da94 lr 1000da90 code 1 in dawn[10000000+140
00]
Mon Nov 23 20:44:37 2020 kern.info kernel: [ 4641.540088] dawn[9424]: code: 3884f9e8 7c7c1b78 7cbd2b78 81228ff8 9121601c 39200000 48000475 7c7
f1b79
Mon Nov 23 20:44:37 2020 kern.info kernel: [ 4641.549534] dawn[9424]: code: 41820034 7f83e378 48000665 38810018 <81230010> 7c7b1b78 7f83e378 7
d2903a6

```

PolynomialDivision, 2021-07-19:
Please try latest dawn version.

bittorf, 2021-07-24:
on vacation, give me ~4 weeks

grrr2, 2021-08-19:
Hi, I have a similar problem on a Netgear R6220. Would you rather track it under another issue number or maybe it is not hardware specific? 

`Thu Aug 19 19:03:02 2021 kern.info kernel: [  515.293086] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000000
Thu Aug 19 19:03:02 2021 kern.info kernel: [  515.309702] epc = 77e3f674 in libc.so[77dc7000+9c000]
Thu Aug 19 19:03:02 2021 kern.info kernel: [  515.319881] ra  = 77e403e4 in libc.so[77dc7000+9c000]`