berlin-open-wireless-lab/DAWN #128: build failed with glibc

gw826943555, 2020-12-23:
I tried to build a custom firmware with glibc and encounterred the following error. It's fine with musl libc.

```
2020-12-22T15:58:25.6345760Z [ 96%] Building C object src/CMakeFiles/dawn.dir/utils/ieee80211_utils.c.o
2020-12-22T15:58:25.6647596Z [100%] Linking C executable dawn
2020-12-22T15:58:25.6929559Z src/CMakeFiles/dawn.dir/build.make:312: recipe for target 'src/dawn' failed
2020-12-22T15:58:25.6932548Z /mnt/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.4.0_glibc_eabi/lib/gcc/arm-openwrt-linux-gnueabi/8.4.0/../../../../arm-openwrt-linux-gnueabi/bin/ld: CMakeFiles/dawn.dir/network/networksocket.c.o: in function `init_socket_runopts':
2020-12-22T15:58:25.6934738Z make[6]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/build_dir/target-arm_cortex-a7+neon-vfpv4_glibc_eabi/dawn-2020-09-03-b639145c'
2020-12-22T15:58:25.6936847Z networksocket.c:(.text+0x23c): undefined reference to `pthread_create'
2020-12-22T15:58:25.6937838Z CMakeFiles/Makefile2:170: recipe for target 'src/CMakeFiles/dawn.dir/all' failed
2020-12-22T15:58:25.6939179Z collect2: error: ld returned 1 exit status
2020-12-22T15:58:25.6940447Z make[5]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/build_dir/target-arm_cortex-a7+neon-vfpv4_glibc_eabi/dawn-2020-09-03-b639145c'
2020-12-22T15:58:25.6941784Z make[6]: *** [src/dawn] Error 1
2020-12-22T15:58:25.6942472Z Makefile:148: recipe for target 'all' failed
2020-12-22T15:58:25.6943629Z make[5]: *** [src/CMakeFiles/dawn.dir/all] Error 2
2020-12-22T15:58:25.6944919Z make[4]: Leaving directory '/home/runner/work/Actions-OpenWrt/Actions-OpenWrt/build_dir/target-arm_cortex-a7+neon-vfpv4_glibc_eabi/dawn-2020-09-03-b639145c'
2020-12-22T15:58:25.6946472Z make[4]: *** [all] Error 2
2020-12-22T15:58:25.6947527Z Makefile:50: recipe for target '/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_glibc_eabi/dawn-2020-09-03-b639145c/.built' failed
2020-12-22T15:58:25.6949947Z make[3]: *** [/mnt/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_glibc_eabi/dawn-2020-09-03-b639145c/.built] Error 2
2020-12-22T15:58:25.6951009Z make[3]: Leaving directory '/mnt/openwrt/feeds/packages/net/dawn'
2020-12-22T15:58:25.6951824Z time: package/feeds/packages/dawn/compile#2.87#0.51#3.45
2020-12-22T15:58:25.6956236Z     ERROR: package/feeds/packages/dawn failed to build.
2020-12-22T15:58:25.6971806Z make[2]: *** [package/feeds/packages/dawn/compile] Error 1
2020-12-22T15:58:25.6972938Z package/Makefile:112: recipe for target 'package/feeds/packages/dawn/compile' failed
```

PolynomialDivision, 2020-12-23:
Try adding `lpthread` to the linker:
https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/CMakeLists.txt#L79-L80

PolynomialDivision, 2020-12-24:
@gw826943555 Why did u close?

gw826943555, 2020-12-25:
> @gw826943555 Why did u close?

I thought it could be fixed by adding `lpthread`. However I didn't have much time to test it. I will try it later and give my response.

gw826943555, 2020-12-26:
solved by adding `pthread` to the link libraries 

```
SET(LIBS
        ubox ubus json-c blobmsg_json uci gcrypt iwinfo pthread)
```

PolynomialDivision, 2020-12-31:
Fixed.

closed 2020-12-31 by PolynomialDivision