berlin-open-wireless-lab/DAWN #129: Luci - Hearing Map - 500 Internal Server Error

elcovv, 2020-12-26:
Hi,

Sometimes I get the error message below when trying to display the hearing map using luci. Restarting the dawn service resolves the issue. Please let me know if you need additional info on this issue.

I got this error before on other routers, but this one is on a TP-Link WR902AC running a snapshot image that I've build yesterday using the image builder:

OpenWrt SNAPSHOT r15350-b9389186b0 / LuCI Master git-20.339.75073-e54708a 

-------------------------------
Sorry, the server encountered an unexpected error.

/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occurred: [string "[string]"]:7: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
    [C]: in function 'pairs'
    [string "[string]"]:7: in main chunk
stack traceback:
    [C]: in function 'error'
    /usr/lib/lua/luci/template.lua:97: in function 
    (tail call): ?
    ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
    /usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
    [string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
    [C]: in function 'error'
    /usr/lib/lua/luci/template.lua:97: in function 
    (tail call): ?
    /usr/lib/lua/luci/cbi.lua:257: in function 'render'
    /usr/lib/lua/luci/cbi.lua:440: in function 'render'
    /usr/lib/lua/luci/dispatcher.lua:1451: in function '_cbi'
    /usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
    /usr/lib/lua/luci/dispatcher.lua:479: in function

/var/log/messages:
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]: A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]: A runtime error occurred: [string "[string]"]:7: bad argument #1 to 'pairs' (table expected, got nil)
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]: stack traceback:
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     [C]: in function 'pairs'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     [string "[string]"]:7: in main chunk
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]: stack traceback:
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     [C]: in function 'error'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     (tail call): ?
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     [string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]: stack traceback:
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     [C]: in function 'error'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     (tail call): ?
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/cbi.lua:257: in function 'render'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/cbi.lua:440: in function 'render'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/dispatcher.lua:1451: in function '_cbi'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
Sat Dec 26 10:13:48 2020 daemon.err uhttpd[1758]:     /usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

PolynomialDivision, 2020-12-26:
Can u please try using ubus to access the maps`

    ubus call dawn get_hearing_map

and

    ubus call dawn get_hearing_network

elcovv, 2021-01-15:
Hi,

Got the same error and ran the commands as you requested:

root@Router:~# `service dawn status`
running
root@Router:~# `ubus call dawn get_hearing_map`
Command failed: Not found
root@Router:~# `ubus call dawn get_hearing_network`
Command failed: Not found

The command '`ubus list`' didn't list dawn...

I found the following in the log:
root@Router:~# `logread | grep dawn`
<removed some lines>
Thu Jan 14 21:29:32 2021 daemon.err dawn[2632]: Failed to invoke: Request timed out
Thu Jan 14 21:29:41 2021 kern.info kernel: [  300.555523] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000010
Thu Jan 14 21:29:41 2021 kern.info kernel: [  300.564130] epc = 004089d7 in dawn[400000+c000]
Thu Jan 14 21:29:41 2021 kern.info kernel: [  300.568865] ra  = 004089d5 in dawn[400000+c000]
Thu Jan 14 22:35:41 2021 daemon.err uhttpd[2082]: 	/usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:10: in function 'render'
Thu Jan 14 22:35:44 2021 daemon.err uhttpd[2082]: 	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'

After a 'service dawn restart' everything was back to normal.

The full logging, dawn config and build version are here: [https://pastebin.com/rpD53PZA ](url)

MrWGT, 2021-01-31:
Had the same issue today on a Mi Router 3G

PolynomialDivision, 2021-01-31:
Thanks for reporting.
It would be nice if you enable the `stdout` in `/etc/init.d/dawn`.

MrWGT, 2021-02-01:
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: --------Connections------
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Connecting to Port: 1026, Connected: True
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: ------------------
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Ustream send: 244
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading 240 bytes to add to 4 of 244...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: ...and we're back, now have 244 bytes
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: message completed
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: processing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"54:48:E6:A4:02:C6\",\"address\":\"B8:E9:37:41:CF:0F\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-55,\"freq\":2412,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":0,\"vht_capabilities\":0}"}
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: not complete msg, len:0, expected len:4
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: leaving
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading 240 bytes to add to 4 of 244...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: ...and we're back, now have 244 bytes
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: message completed
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: processing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"56:48:E6:A4:02:C6\",\"address\":\"B8:E9:37:41:CF:0F\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-55,\"freq\":2412,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":0,\"vht_capabilities\":0}"}
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: not complete msg, len:0, expected len:4
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: leaving
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading 240 bytes to add to 4 of 244...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: ...and we're back, now have 244 bytes
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: message completed
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: processing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"52:48:E6:A4:02:C6\",\"address\":\"B8:E9:37:41:CF:0F\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-55,\"freq\":2412,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":0,\"vht_capabilities\":0}"}
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: not complete msg, len:0, expected len:4
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: leaving
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading 240 bytes to add to 4 of 244...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: ...and we're back, now have 244 bytes
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: message completed
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: processing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"54:48:E6:A4:02:C6\",\"address\":\"B8:E9:37:41:CF:0F\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-57,\"freq\":2412,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":0,\"vht_capabilities\":0}"}
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: not complete msg, len:0, expected len:4
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: leaving
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: reading 240 bytes to add to 4 of 244...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: ...and we're back, now have 244 bytes
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: message completed
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: processing message...
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"56:48:E6:A4:02:C6\",\"address\":\"B8:E9:37:41:CF:0F\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-57,\"freq\":2412,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":0,\"vht_capabilities\":0}"}
Mon Feb  1 13:15:29 2021 daemon.info dawn[8425]: tcp_socket: commencing message...
[... cut ...] ap\":{\"rx\":{\"1ss\":9,\"2ss\":-1,\"3ss\":-1,\"4ss\":-1,\"5ss\":-1,\"6ss\":-1,\"7ss\":-1,\"8ss\":-1},\"tx\":{\"1ss\":9,\"2ss\":-1,\"3ss\":-1,\"4ss\":-1,\"5ss\":-1,\"6ss\":-1,\"7ss\":-1,\"8ss\":-1}}}}}},\"collision_domain\":-1,\"bandwidth\":-1,\"bssid\":\"54:48:E6:A4:02:C7\",\"ssid\":\"FRITZ!Box Susi5\",\"ht_supported\":true,\"vht_supported\":true,\"ap_weight\":0,\"channel_utilization\":19,\"neighbor_report\":\"5448e6a402c7ef1900008028090603022a00\",\"iface\":\"wlan1\",\"hostname\":\"MI4AGE\"}"}

PolynomialDivision, 2021-02-01:
@MrWGT In your log is not the error you describe, or?

MrWGT, 2021-02-01:
I've uploaded a larger log to my onedrive:

https://1drv.ms/u/s!Assv5fIy-dXzgpsFkQrH9vzuoXth5w?e=KFspAG

---------------
...
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: --------Connections------
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Connecting to Port: 1026, Connected: True
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: ------------------
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Ustream send: 244
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Calculating own score!
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Score: -2 of:
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:85, client_addr: 2C:78:0E:CE:CB:AF, signal: -77, freq: 5200, counter: 2, vht: 1, min_rate: 9, max_rate: 2
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Own Score! Skipping!
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:85, client_addr: 2C:78:0E:CE:CB:AF, signal: -77, freq: 5200, counter: 2, vht: 1, min_rate: 9, max_rate: 2
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Calculating score to compare!
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Score: -2 of:
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:86, client_addr: 2C:78:0E:CE:CB:AF, signal: -56, freq: 2437, counter: 2, vht: 0, min_rate: 58, max_rate: 50
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Comparing own 2 to 0
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Comparing own station count 2 to 0
Mon Feb  1 13:59:50 2021 daemon.err dawn[8584]: **Neigbor-Report is NULL!**
Mon Feb  1 13:59:50 2021 daemon.info dawn[8584]: Method new: probe : {"address":"f8:63:3f:a4:57:f9","target":"ff:ff:ff:ff:ff:ff","signal":-69,"freq":5200,"ht_capabilities":{"ht_capabilities_info":25353,"supported_mcs_set":{"a_mpdu_params":23,"ht_extended_capabilities":0,"tx_bf_capability_info":0,"asel_capabilities":0,"supported_mcs_set":[255,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}},"vht_capabilities":{"vht_capabilities_info":829521971,"vht_supported_mcs_set":{"rx_map":-257,"rx_highest":0,"tx_map":-257,"tx_highest":0}}}
...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: --------Connections------
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Connecting to Port: 1026, Connected: True
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: ------------------
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Ustream send: 244
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Calculating own score!
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Score: -2 of:
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:85, client_addr: 2C:78:0E:CE:CB:AF, signal: -80, freq: 5200, counter: 3, vht: 1, min_rate: 9, max_rate: 2
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Own Score! Skipping!
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:85, client_addr: 2C:78:0E:CE:CB:AF, signal: -80, freq: 5200, counter: 3, vht: 1, min_rate: 9, max_rate: 2
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Calculating score to compare!
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Score: -2 of:
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:86, client_addr: 2C:78:0E:CE:CB:AF, signal: -58, freq: 2437, counter: 4, vht: 0, min_rate: 58, max_rate: 50
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Comparing own 2 to 0
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Comparing own station count 2 to 0
Mon Feb  1 13:59:56 2021 daemon.err dawn[8584]: **Neigbor-Report is NULL!**
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: commencing message...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: reading message...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: reading 240 bytes to add to 4 of 244...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: ...and we're back, now have 244 bytes
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: message completed
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: processing message...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Network Method new: probe : {"method":"probe","data":"{\"bssid\":\"54:48:E6:A4:02:C7\",\"address\":\"2C:78:0E:CE:CB:AF\",\"target\":\"FF:FF:FF:FF:FF:FF\",\"signal\":-69,\"freq\":5200,\"rcpi\":-1,\"rsni\":-1,\"ht_capabilities\":1,\"vht_capabilities\":1}"}
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: commencing message...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: not complete msg, len:0, expected len:4
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: tcp_socket: leaving
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Method new: probe : {"address":"2c:78:0e:ce:cb:af","target":"ff:ff:ff:ff:ff:ff","signal":-80,"freq":5200,"ht_capabilities":{"ht_capabilities_info":-4351,"supported_mcs_set":{"a_mpdu_params":23,"ht_extended_capabilities":0,"tx_bf_capability_info":0,"asel_capabilities":0,"supported_mcs_set":[255,255,0,0,0,0,0,0,0,0,0,0,1,0,0,0]}},"vht_capabilities":{"vht_capabilities_info":-177176318,"vht_supported_mcs_set":{"rx_map":-1281,"rx_highest":-25081,"tx_map":-1281,"tx_highest":-25049}}}
...
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Connecting to Port: 1026, Connected: True
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: ------------------
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Ustream send: 244
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Calculating own score!
Mon Feb  1 13:59:56 2021 daemon.info dawn[8584]: Score: -2 of:
[... cut ...]
Mon Feb  1 14:01:56 2021 daemon.info dawn[8584]: bssid_addr: 14:CC:20:E5:BE:85, client_addr: FE:5B:7C:59:49:10, signal: -58, freq: 5200, counter: 2, vht: 1, min_rate: 0, max_rate: 0
Mon Feb  1 14:01:56 2021 daemon.err dawn[8584]: **Neigbor-Report is NULL!**
Mon Feb  1 14:01:56 2021 daemon.info dawn[8584]: Method new: probe : {"address":"fe:5b:7c:59:49:10","target":"ff:ff:ff:ff:ff:ff","signal":-60,"freq":2437,"ht_capabilities":{"ht_capabilities_info":11584,"supported_mcs_set":{"a_mpdu_params":27,"ht_extended_capabilities":0,"tx_bf_capability_info":0,"asel_capabilities":0,"supported_mcs_set":[255,255,0,0,0,0,0,0,0,0,0,0,0,0,0,0]}}}
Mo

MrWGT, 2021-02-01:
Oh... @Grmpfl... My fault.... Thought "Neigbor-Report is NULL!" issue

elcovv, 2021-02-12:
Hi,

Just have been able to reproduce the issue. After giving the 'wifi' command on the router, I got the following messages in syslog:

Fri Feb 12 10:45:50 2021 daemon.err dawn[3264]: [SUBSCRIBING] No hostapd sockets!
Fri Feb 12 10:45:50 2021 kern.info kernel: [375467.721227] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000010
Fri Feb 12 10:45:50 2021 kern.info kernel: [375467.737907] epc = 00408917 in dawn[400000+c000]
Fri Feb 12 10:45:50 2021 kern.info kernel: [375467.747119] ra  = 00408915 in dawn[400000+c000]

After that I wasn't able to get the hearing_map or network:

root@Router:~# `ubus call dawn get_hearing_map`
Command failed: Not found
root@Router:~# `ubus call dawn get_network`
Command failed: Not found

And I got the error described in the earlier messages trying to get the hearing_map or network using LuCI.

`service dawn restart` resolved the issue

Dawn only crashed once after giving the `wifi` command, subsequent attempts didn't affect dawn.

Unfortunately I had to disable dawn stdout earlier because the CPU of the router couldn't cope with the amount of logging.

The coming days I'll try to reproduce the issue with dawn stdout enabled.

elcovv, 2021-02-13:
Managed to reproduce the same issue, using the `wifi` command on my other router has stdout logging enabled.

Sat Feb 13 10:05:38 2021 daemon.err dawn[3002]: [SUBSCRIBING] No hostapd sockets!                                                                                    
Sat Feb 13 10:05:38 2021 daemon.info dawn[3002]: [Thread] : Removing old client entries!                                                                             
Sat Feb 13 10:05:41 2021 daemon.err dawn[3002]: No survey results                                                                                                    
Sat Feb 13 10:05:41 2021 daemon.err dawn[3002]:                                                                                                                      
Sat Feb 13 10:05:41 2021 kern.info kernel: [708801.766240] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000010                            
Sat Feb 13 10:05:41 2021 kern.info kernel: [708801.774923] epc = 004089d7 in dawn[400000+c000]                                                                       
Sat Feb 13 10:05:41 2021 kern.info kernel: [708801.779727] ra  = 004089d5 in dawn[400000+c000]

and a second time on the same router:

Sat Feb 13 10:17:49 2021 daemon.info dawn[7004]: Ustream send: 756                                                                                                   
Sat Feb 13 10:17:49 2021 daemon.info dawn[7004]: -------- IW INFO UPDATE!!!---------                                                                                 
Sat Feb 13 10:17:49 2021 daemon.info dawn[7004]: EVAL 50:C7:BF:9A:91:33                                                                                              
Sat Feb 13 10:17:49 2021 kern.info kernel: [709529.949760] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000060                            
Sat Feb 13 10:17:49 2021 kern.info kernel: [709529.958467] epc = 004086e9 in dawn[400000+c000]                                                                       
Sat Feb 13 10:17:49 2021 kern.info kernel: [709529.963282] ra  = 004086e9 in dawn[400000+c000] 

third time: (`service dawn restart` didn't work)

Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Connected to ubus                                                                                                   
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Adding socket!                                                                                                      
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Failed to lookup ID!Failed to lookup ID!Failed to lookup ID!Copied Neighborreport: 56c7bf9a9133ef0900005104070603000
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Subscribed to: wlan1-2                                                                                              
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Copied Neighborreport: 52c7bf9a9133ef0900005104070603000400,                                                        
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Copied Neighborreport: 56c7bf9a9133ef0900005104070603000400,                                                        
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Subscribed to: wlan1-1                                                                                              
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Copied Neighborreport: 50c7bf9a9133ef0900005104070603000400,                                                        
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Copied Neighborreport: 52c7bf9a9133ef0900005104070603000400,                                                        
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Copied Neighborreport: 56c7bf9a9133ef0900005104070603000400,                                                        
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: Subscribed to: wlan1                                                                                                
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: [Thread] : Removing old probe entries!                                                                              
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: [Thread] : Removing old entries finished!                                                                           
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: [Thread] : Removing old client entries!                                                                             
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: [ULOOP] : Removing old ap entries!                                                                                  
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: [ULOOP] : Processing denied authentication!                                                                         
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: --------Connections------                                                                                           
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: ------------------                                                                                                  
Sat Feb 13 10:23:14 2021 daemon.info dawn[7825]: -------- IW INFO UPDATE!!!---------                                                                                 
[... cut ...]
Sat Feb 13 10:23:14 2021 kern.info kernel: [709854.875958] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000060                            
Sat Feb 13 10:23:14 2021 kern.info kernel: [709854.884654] epc = 004086e9 in dawn[400000+c000]                                                                       
Sat Feb 13 10:23:14 2021 kern.info kernel: [709854.889452] ra  = 004086e9 in dawn[400000+c000]   

after a `service dawn stop` and `service dawn start`, it worked again.

Not every time the wifi command is given dawn crashes.

PolynomialDivision, 2021-02-14:
Could u turn of 802.11k in dawn config and see if this error is still happening?

PolynomialDivision, 2021-07-19:
Should be fixed. If not feel free to reopen.

closed 2021-07-19 by PolynomialDivision