berlin-open-wireless-lab/DAWN #130: What exactly is this project?

rubo77, 2021-02-02:
I stumbled across this and it sounded very interesting. Though I wonder what it exactly does. What is the difference to a Freifunk mesh network?

Maybe you can add some words to the Readme?

PolynomialDivision, 2021-02-03:
> Though I wonder what it exactly does. What is the difference to a Freifunk mesh network?

I try influencing the roaming behavior of clients and basically do client-steering.
Maybe just read what I did here: https://polynomialdivision.github.io/ ;)
It is a bit outdated, since I added 802.11v/k to have a more modern steering.

> Maybe you can add some words to the Readme?

Yeah I should add some stuff. ;)

rubo77, 2021-02-04:
So is this a whole framework like gluon? or is it just a package for openwrt, that you can use to controll which client connects where?

PolynomialDivision, 2021-02-04:
It is just a package.

CodeFetch, 2021-02-14:
@rubo77 Lemoer and I are looking into implementing this in Gluon.

PolynomialDivision, 2021-02-14:
> @rubo77 Lemoer and I are looking into implementing this in Gluon.

Cool. :) Maybe we can do a jitsi call. As u maybe notice, I'm not that active right now, since we have other things to do for Freifunk (IPv6 Stuff). Maybe in 1 1/2 months I can continue, here.

CodeFetch, 2021-02-15:
@PolynomialDivision I'm often (and now) in the leinelab Mumble (leineserver.leinelab.org) in the evening.

I think we first need to identify mesh realms to use DAWN efficiently:
https://pad.freifunk.net/p/meshrealmd

And I still need to figure out a good algorithm, so it's still a draft...

PolynomialDivision, 2021-07-17:
On some sites dawn is now running in our freifunk network.

CodeFetch, 2021-07-17:
@PolynomialDivision So you run it on whole sites? Aren't there any scaling issues?

PolynomialDivision, 2021-07-18:
@CodeFetch No? I talk here about 6 APs.