berlin-open-wireless-lab/DAWN #131: Remove duplicate column headers

UltraHKR, 2021-02-22:
In the following image I think the areas which are in red, should be eliminated...

In the case of the subtitle is pretty obvious i think. (Title repeated twice)

In the case of the AP table, the description is repeated per every BSSID which makes the page much longer than needed.

In the case of the VHT column I think this column should (could?) be omitted in AP's that dont support VHT.

Improvement: Please sort the whole BSSID list by frequency, otherwise its harder to differentiate 2.4 & 5GHz BSSID's.

[image]

namidairo, 2021-02-22:
I do agree that the HT/VHT columns do look inefficient and could probably be merged, since VHT capability would probably imply HT capability, just as HE would imply VHT. (At least on 5ghz anyway, HE wouldn't imply VHT on 2.4ghz as that usually was a vendor extension. I don't believe DAWN has support for HE at the moment anyway)
So it would be something like N/A, HT, VHT, HE?

Taking a look at https://github.com/openwrt/luci/tree/master/applications/luci-app-dawn and I don't believe it would be too hard of a change to implement.

Although, I think the author(s) may be looking at rewriting all of the dom stuff anyway in javascript right now...

PolynomialDivision, 2021-02-24:
Feel free to make a PR. :)

> Although, I think the author(s) may be looking at rewriting all of the dom stuff anyway in javascript right now...

Yep.