berlin-open-wireless-lab/DAWN #133: Failed to execute template 'cbi/map'

rutkas, 2021-05-30:
At the beginning it was working, but after some hours of work I've got in luci-dawn:

```
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occurred: [string "[string]"]:8: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	[string "[string]"]:8: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1449: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1022: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function 
```
OpenWrt SNAPSHOT r16739-f2844a6338 / LuCI Master git-21.124.24916-0faf9a4
dawn | 2020-12-31-8ce09d64-3
luci-app-dawn | git-21.088.35499-32f0ff2

DAWN configuration is default:

```
config network
    option broadcast_ip         '10.0.0.255'
    option broadcast_port       '1025'
    option server_ip            ''
    option tcp_port             '1026'
    option network_option       '2' # 0 udp broadcast, 1 multicast, 2 tcp
    option shared_key           'Niiiiiiiiiiiiiik'
    option iv                   'Niiiiiiiiiiiiiik'
    option use_symm_enc         '1'
    option collision_domain     '-1'     # enter here aps which are in the same collision domain
    option bandwidth            '-1'     # enter network bandwidth

config ordering
    option sort_order           'cbfs'

config hostapd
    option hostapd_dir          '/var/run/hostapd'

config times
    option update_client         '10'
    option denied_req_threshold  '30'
    option remove_client         '15'
    option remove_probe          '30'
    option remove_ap             '460'
    option update_hostapd        '10'
    option update_tcp_con        '10'
    option update_chan_util      '5'
    option update_beacon_reports '20'

config metric
    option ap_weight            '0'
    option ht_support           '0'
    option vht_support          '0'
    option no_ht_support        '0'
    option no_vht_support       '0'
    option rssi                 '10'
    option low_rssi             '-500'
    option freq                 '100' # some kind of band-steering
    option chan_util            '0'
    option max_chan_util        '-500'
    option rssi_val             '-60'
    option low_rssi_val         '-80'
    option chan_util_val        '140'
    option max_chan_util_val    '170'
    option min_probe_count      '0'
    option bandwidth_threshold  '6'
    option use_station_count    '1'
    option max_station_diff     '1'
    option eval_probe_req       '0'
    option eval_auth_req        '0'     # no real reasoncode...
    option eval_assoc_req       '0'     # just deny assocs...
    option kicking              '0'
    option deny_auth_reason     '1'     # unspecified
    option deny_assoc_reason    '17'    # assoc rejected can't handle new station
    option use_driver_recog     '1'
    option min_number_to_kick   '3'
    option chan_util_avg_period '3'
    option set_hostapd_nr       '1'
    option op_class             '0' #802.11k stuff...
    option duration             '0'
    option mode                 '0'
    option scan_channel         '0'
```

PolynomialDivision, 2021-05-30:
Probably DAWN is crashed? You can look into `logread` to find information.

rutkas, 2021-05-30:
here's everything about dawn:

root@OpenWrt-EW72:~# logread | grep dawn
Sun May 30 11:30:02 2021 daemon.err dawn[8621]: Failed to invoke: Request timed out
Sun May 30 11:30:04 2021 daemon.err dawn[8621]: Failed to invoke: Request timed out
Sun May 30 11:30:07 2021 daemon.err dawn[8621]: Failed to invoke: Request timed out
Sun May 30 11:30:07 2021 kern.info kernel: [412748.524493] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000044
Sun May 30 11:30:07 2021 kern.info kernel: [412748.533222] epc = 00408b05 in dawn[400000+c000]
Sun May 30 11:30:07 2021 kern.info kernel: [412748.538026] ra  = 00408b05 in dawn[400000+c000]
Sun May 30 13:56:30 2021 daemon.err uhttpd[1519]:       ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Sun May 30 13:56:57 2021 daemon.err uhttpd[1519]:       ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Sun May 30 13:57:00 2021 daemon.err uhttpd[1519]:       /usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:10: in function 'render'
Sun May 30 13:57:05 2021 daemon.err uhttpd[1519]:       ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Sun May 30 13:59:27 2021 daemon.err uhttpd[1519]:       /usr/lib/lua/luci/model/cbi/dawn/dawn_network.lua:10: in function 'render'
Sun May 30 14:03:53 2021 daemon.err uhttpd[1519]:       ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Sun May 30 21:18:49 2021 daemon.err uhttpd[1519]:       ...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'

ktgeek, 2021-06-16:
I'm getting this right now with dawn running.  Dawn off the openwrt-21.02 packages branch and running on 21.02-rc1 on a Turris Omnia, an r7800 and an archerc7v2

PolynomialDivision, 2021-06-17:
Does dawn run without any problems if you do not use/access luci?

ktgeek, 2021-06-17:
I had erased my earlier comment of the network overview not working on 21.02rc3 because I noticed there was a package update and it worked right after i did that... then a little bit later it was back to not working.

On all 3 access points the dawn daemon seems to be running without problem.  The lua runtime error for network overview happens *most* of the time, with occasional times when it works just fine. (Just keep reloading until it shows me something is my current horrible strategy.)

I haven't been able to determine the why it works at some times and not at others.  The hearing map always works, though.

PolynomialDivision, 2021-06-18:
There was a lot of rewrite of the "datastorage" code. I guess some race condition... That is why I'm intersted if you do not call anything and just let dawn run at it is, if it works. ;)

cotequeiroz, 2021-07-05:
I think the cause of this is that the `ap_array_find_first_entry` function does not take the SSID into consideration.  So, if you have more than one SSID, the first one--at least in my case--will have a correct list of APs, while the rest gets flooded with duplicate entries.  I have a tentative fix for this, and will open a PR if it works here.  When you have enough of them, the ubus call will fail:
```
# ubus call dawn get_network
Command failed: Connection failed
```
It will eventually succeed.  If you're patient enough, you may get a huge map with:
```
# while ! ubus call dawn get_network; do true; done
```

PR #136 (2021-07-08): Take SSID into account when building the network map and neighbor reports
The first commit in this series adds an extra byte to SSID strings to hold the NULL-byte termination, avoiding problems with some string functions.  Specifically, `get_ssid()` would write the termination null-byte outside of the ssid string if the ssid length is 32; I've changed `strcpy` to `strncpy` and remove the unnecessary `memcpy` above it to strengthen it some more.

Currently, `ap_array_get_ap` returns an entry by matching just the bssid, without checking if the ssid matches.  `ap_array_find_first_entry` does a bi [... cut]
closed 2021-07-08 by PolynomialDivision