berlin-open-wireless-lab/DAWN #134: Securing DAWN messaging

stevekerrison, 2021-05-31:
I've recently taken an interest in this project as I'm building an OpenWRT-based home network with several APs.

I've noted that the TCP comms is encrypted with `gcrypt`. However I think there's some issues with the way it's done right now, and have some ideas about how to improve that.

Note that I haven't yet taken the time to attempt to break/exploit any of these potential issues. I thought it might be good to discuss (privately or openly) with @PolynomialDivision and other interested parties, to align with the general direction/intent of the project's future before proposing any fixes/changes.

Let me know if that's a conversation worth having!

PolynomialDivision, 2021-05-31:
> Let me know if that's a conversation worth having!

I'm always happy about contributions. :)
I thought about ssl encrypted TCP sessions, but I had an issues with the corresponding ssl-usock. Some people use also the multicast option. However, I always feel like I should rewrite a huge part of the code to simplify everything...

stevekerrison, 2021-06-01:
> I'm always happy about contributions. :)

Great!

> I thought about ssl encrypted TCP sessions, but I had an issues with the corresponding ssl-usock. Some people use also the multicast option. However, I always feel like I should rewrite a huge part of the code to simplify everything...

Yeah TLS was my first thought, which has its own challenges and precludes it from the multicast/broadcast method.

Anyway, day-job will have priority, but I'll try to look into it soon.