berlin-open-wireless-lab/DAWN #138: Error with latest dawn

MrWGT, 2021-07-21:
I'm running dawn on 2 TP C2600 and one EA8300 with openwrt 21.07rc3. Now the following error occurred:

/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occurred: [string "[string]"]:8: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	[string "[string]"]:8: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1451: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function 

Syslog shows:

Wed Jul 21 21:25:06 2021 daemon.notice hostapd: Failed to send assoc resp: Resource temporarily unavailable

Any idea how to solve the problem?

Thanks Gerd

PolynomialDivision, 2021-07-21:
DAWN version is `dawn_2021-07-11`? You can findout by `opkg list-installed | grep dawn`.

MrWGT, 2021-07-21:
dawn - 2021-07-11-ec9a3a96-1
luci-app-dawn - git-21.088.35541-d627abe

Access to tp c2600 was slow. Never had such a syslog entry before. But updated all modules which where offered for update. Second c2600 and ea8300 where also updated and work (up to now) flawless.

PolynomialDivision, 2021-07-21:
Did you run an older version of dawn before?

MrWGT, 2021-07-21:
Yes. Running dawn on them since snapshot, rc1 and rc2.

1. C2600: 5ghz r/v/k, 2.4ghz iot 1, mt7612usb r/v/k
2. EA8300: 2x5ghz r/v/k
3. C2600: 5ghz r/v/k, 2.4ghz iot 2

Since last dawn view hearing map shows combined APs for r/v/k net and both iot nets with clients. Previously only one iot net showed clients. The second one was just a header line with correct said.

PolynomialDivision, 2021-07-22:
Could you try turning off 802.11k beacon requesting? Set the update interval to 0.

MrWGT, 2021-07-22:
Is this "update client information" over web interface? Or how should I do it?

MrWGT, 2021-07-22:
Set above parameter and rebooted. ~1h later:

Thu Jul 22 19:40:43 2021 daemon.err uhttpd[1083]: luci: accepted login on /admin/network/network for root from 192.168.10.34
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: A runtime error occurred: [string "[string]"]:8: bad argument #1 to 'pairs' (table expected, got nil)
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: stack traceback:
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	[C]: in function 'pairs'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	[string "[string]"]:8: in main chunk
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: stack traceback:
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	[C]: in function 'error'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	(tail call): ?
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: stack traceback:
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	[C]: in function 'error'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	(tail call): ?
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/dispatcher.lua:1451: in function '_cbi'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/dispatcher.lua:1024: in function 'dispatch'
Thu Jul 22 19:40:46 2021 daemon.err uhttpd[1083]: 	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

MrWGT, 2021-07-22:
C2600 showing error has high top usage for dawn:
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 4939     1 root     R     2632   1%  49% /usr/sbin/dawn

Other C2600 has low usage:
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 3521     2 root     IW       0   0%   2% [kworker/1:2-eve]
 1258     1 root     S     2664   1%   1% /usr/sbin/dawn

PolynomialDivision, 2021-07-24:
Sry, I can not reproduce it. Can you please activate stdout output in /etc/init.d/dawn? If crash happens give me the whole logread output? :)

MrWGT, 2021-07-25:
I took a third c2600 laying around, installed 21.02rc3+updates and everything else, made a backup of settings of "error c2600", restored them on 3. and no problems.

So i made the same with "error c2600" (reinstall 21.02rc3, updates, restore backup) and it works again too. It looks like something "not in settings backup" was the problem.

MrWGT, 2021-07-25:
As the "always 0" channel utilization with qca99x0 in c2600 was something which annoyed me I swapped the two c2600 with one ea8300 and one mr8300. So currently dawn runs on 2 ea8300 and 1 mr8300 but this was also not easy as they supply one "r" ssid, two none "r" ssids and one ea8300 had a disabled upper 5ghz card.

But this is a different story 😅

closed 2021-07-25 by MrWGT