berlin-open-wireless-lab/DAWN #139: Correct mode for beacon request?

MrWGT, 2021-07-25:
ubus call hostapd.wlan0 rrm_beacon_req accepts 3 modes 0 (BEACON_REPORT_MODE_PASSIVE), 1 (BEACON_REPORT_MODE_ACTIVE) and 2 (BEACON_REPORT_MODE_TABLE). 

Which mode is the correct one for dawn?

Tested with an iPhone 11 pro max.

mode = 0
`ubus call hostapd.wlan0 rrm_beacon_req '{"addr":"CC:D2:81:41:72:90", "op_class":1, "channel":0,"duration":1,"mode":0,"bssid":"ff:ff:ff:ff:ff:ff", "ssid":""}'

Sun Jul 25 23:20:34 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS cc:d2:81:41:72:90 2 ack=1
Sun Jul 25 23:20:34 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX cc:d2:81:41:72:90 2 00 01748adfce42a0cc9600010000c923ea9f8015f472000000000001e0015ce12d0100000043001110000f465249545a21426f7820537573693501069824b048606c030174050401020000070c494e20640b1890010095051e30180100000fac040100000fac040200000fac02000fac040c000b0501000700004605720000000036036215013b0280002d1aef091bffff0000000000000000000001000000000000000000003d16740504000000000000000000000000000000000000007f080400080200000140bf0cb2198833faff0000faff0000c005017a00fcffc30402303030dd180050f2020101810003a4000027a4000042435e0062322f00`

mode = 1
`ubus call hostapd.wlan0 rrm_beacon_req '{"addr":"CC:D2:81:41:72:90", "op_class":1, "channel":0,"duration":1,"mode":1,"bssid":"ff:ff:ff:ff:ff:ff", "ssid":""}'

Sun Jul 25 23:20:46 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS cc:d2:81:41:72:90 3 ack=1
Sun Jul 25 23:20:46 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX cc:d2:81:41:72:90 3 00 017418b48d43a0cc9600010000c923ea9f8015f472000000000001e001fca12e0100000043001110000f465249545a21426f7820537573693501069824b048606c030174050401020102070c494e20640b1890010095051e30180100000fac040100000fac040200000fac02000fac040c000b0501000600004605720000000036036215013b0280002d1aef091bffff0000000000000000000001000000000000000000003d16740504000000000000000000000000000000000000007f080400080200000140bf0cb2198833faff0000faff0000c005017a00fcffc30402303030dd180050f2020101810003a4000027a4000042435e0062322f00
Sun Jul 25 23:20:46 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX cc:d2:81:41:72:90 3 00 200b18b48d43a0cc9600010000a70368b6fcff1ba8000000000001d6d49d50a3580000006400110c00084175656e6c616e64010882848b961224486c03010b32040c1830600706555320010b143308200102030405060733082105060708090a0b050400010006dd270050f204104a0001101044000102104700102880288028801880a88068b6fcff1ba8103c0001012a01042d1a6e1017ffff0000010000000000000000000000000c00000000003d160b0700000000000000000000000000000000000000004a0e14000a002c01c8001400050019007f0101dd160050f20101000050f20401000050f20401000050f2021`

mode = 2
`ubus call hostapd.wlan0 rrm_beacon_req '{"addr":"CC:D2:81:41:72:90", "op_class":1, "channel":0,"duration":1,"mode":2,"bssid":"ff:ff:ff:ff:ff:ff", "ssid":""}'
Sun Jul 25 23:20:53 2021 daemon.notice hostapd: Beacon request: cc:d2:81:41:72:90 does not support table beacon report`

Gerd

MrWGT, 2021-07-25:
The reason i ask is that if i issue `ubus call hostapd.wlan0 rrm_beacon_req '{"addr":"CC:D2:81:41:72:90", "op_class":1, "channel":0,"duration":1,"mode":1,"bssid":"ff:ff:ff:ff:ff:ff", "ssid":""}'` my iphone shows up in the hearing map as

`CC:D2:81:41:72:90 | EA:9F:80:15:F4:72 | 5.580 GHz Channel: 116 | True | True | -70 | 202 | 35 | 1.57 % | 1 | 285`

But running dawn without issuing the above command the rcpi and rsni stays at -1. But i think i've set the params at dawns config correct:
`option op_class '1'
option duration '1'
option mode '1'
option scan_channel '0'`

PolynomialDivision, 2021-07-26:
Thanks for bringing that up. Please have also a look at this:
https://github.com/berlin-open-wireless-lab/DAWN/pull/126

We can derive the correct mode automatically:
https://github.com/openwrt/packages/blob/master/utils/prometheus-node-exporter-lua/files/usr/lib/lua/prometheus-collectors/hostapd_ubus_stations.lua#L19-L55

That is a plugin that I wrote, to know what the clients are capable of. I actually wanted to put that also in dawn. However, then I stopped working on DAWN because having no time anymore.

MrWGT, 2021-07-26:
If I set some config parameters over web interface scan channel is always set to 1. So the correct setting 0 is overwritten.

Maybe the setting 0 is seen as invalid on startup and set internally to 1. But with 1 beacon report doesn't work with ea8300, mr8300 and c2600.

This could be the reason why sending the cmd on shell with channel 0 leads to valid rcpi  values. But running dawn alone does not.

Maybe you can check even if no time 😅

PolynomialDivision, 2021-07-26:
> Maybe the setting 0 is seen as invalid on startup and set internally to 1. But with 1 beacon report doesn't work with ea8300, mr8300 and c2600.

Sry, what is your dawn config? This line are interesting for me:
https://github.com/openwrt/packages/blob/master/net/dawn/files/dawn.config#L59-L62

They configure what rrm parameters the AP uses.

PolynomialDivision, 2021-07-26:
@MrWGT Are you willing to test my PR, if I merge DAWN PR into trunk?

PolynomialDivision, 2021-07-26:
What is https://github.com/openwrt/packages/blob/master/net/dawn/files/dawn.config#L28?

PolynomialDivision, 2021-07-26:
@MrWGT Please enable stdout in /etc/init.d/dawn and grep for "Invoking beacon report" in the log

MrWGT, 2021-07-26:
> @MrWGT Are you willing to test my PR, if I merge DAWN PR into trunk?

Yes but I never tried to compile openwrt for ea8300 or dawn. Should be no problem - just time. Lot of work and 3 weeks vacation start of August 😅 (Schulferien in Bayern). So would be after 21th of August.

PS: compiled openwrt and dawn for c2600 half year ago just to see if my virtual machine is ok.

PPS: I'm a sw ing. but real small embedded systems like stm32 without os for e.g. motor drive (pedelec) and associated displays.

MrWGT, 2021-07-26:
> @MrWGT Please enable stdout in /etc/init.d/dawn and grep for "Invoking beacon report" in the log

Will try tomorrow evening.

MrWGT, 2021-07-27:
Enabled stdout, set scan_channel '0' (as needed for ea8300?!?), "logread -f | grep Invok" showed nothing. Time was standard setting 20.

This evening I'll try with scan_channel '1'. But with channel 1 ubus cmd on shell won't work.

PolynomialDivision, 2021-07-27:
Thanks. I check here for enables rrm:
https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/storage/datastorage.c#L397-L402

Could you check the rrm string in `ubus call hostapd.wlan0 get_clients` or whataever. ;)

MrWGT, 2021-07-27:
A old S7 and a Amazon echo at home checked over Vpn:

ubus call hostapd.wlan0 get_clients
{
        "freq": 5580,
        "clients": {
                "64:a2:00:63:32:e6": {
                        "auth": true,
                        "assoc": true,
                        "authorized": true,
                        "preauth": false,
                        "wds": false,
                        "wmm": true,
                        "ht": true,
                        "vht": true,
                        "wps": false,
                        "mfp": false,
                        "rrm": [
                                115,
                                16,
                                145,
                                0,
                                4
                        ],
                        "aid": 2,
                        "signature": "wifi4|probe:0,1,45,191,221(0050f2,8),127,255,221(506f9a,22),221(8cfdf0,1),htcap:01ef,htagg:13,htmcs:0000ffff,vhtcap:3391f9b2,vhtrxmcs:0362fffa,vhttxmcs:0362fffa,extcap:00000a82004000000001|assoc:0,1,33,70,48,54,55,59,45,127,191,221(0050f2,2),221(8cfdf0,1),htcap:01ef,htagg:13,htmcs:0000ffff,vhtcap:339119b2,vhtrxmcs:0362fffa,vhttxmcs:0362fffa,txpow:1408,extcap:0400080200000040",
                        "bytes": {
                                "rx": 798509,
                                "tx": 23759470
                        },
                        "airtime": {
                                "rx": 497412,
                                "tx": 1266963
                        },
                        "packets": {
                                "rx": 5605,
                                "tx": 18008
                        },
                        "rate": {
                                "rx": 12000,
                                "tx": 6500
                        },
                        "signal": -92,
                        "capabilities": {
                                "vht": {
                                        "su_beamformee": true,
                                        "mu_beamformee": true,
                                        "mcs_map": {
                                                "rx": {
                                                        "1ss": 9,
                                                        "2ss": 9,
                                                        "3ss": -1,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                },
                                                "tx": {
                                                        "1ss": 9,
                                                        "2ss": 9,
                                                        "3ss": -1,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                }
                                        }
                                }
[... 47 lines cut ...]
                                                        "3ss": -1,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                },
                                                "tx": {
                                                        "1ss": 9,
                                                        "2ss": 9,
                                                        "3ss": -1,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                }
                                        }
                                }
                        }
                }
        }
}

rrm is populated.

MrWGT, 2021-07-27:
ubus call hostapd.wlan2 get_clients
{
        "freq": 5200,
        "clients": {
                "cc:66:0a:c8:83:b4": {
                        "auth": true,
                        "assoc": true,
                        "authorized": true,
                        "preauth": false,
                        "wds": false,
                        "wmm": true,
                        "ht": true,
                        "vht": true,
                        "wps": false,
                        "mfp": false,
                        "rrm": [
                                49,
                                8,
                                1,
                                0,
                                0
                        ],
                        "aid": 4,
                        "bytes": {
                                "rx": 6587442,
                                "tx": 162734411
                        },
                        "airtime": {
                                "rx": 2566080,
                                "tx": 4618747
                        },
                        "packets": {
                                "rx": 42139,
                                "tx": 167275
                        },
                        "rate": {
                                "rx": 585100,
                                "tx": 866700
                        },
                        "signal": -67,
                        "capabilities": {
                                "vht": {
                                        "su_beamformee": true,
                                        "mu_beamformee": false,
                                        "mcs_map": {
                                                "rx": {
                                                        "1ss": 9,
                                                        "2ss": 9,
                                                        "3ss": -1,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                },
                                                "tx": {
                                                        "1ss": 9,
                                                        "2ss": 9,
                                                        "3ss": -1,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                }
                                        }
                                }
                        }
                },
                "94:f6:d6:d8:df:3d": {
[... 173 lines cut ...]
                                                        "3ss": 9,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                },
                                                "tx": {
                                                        "1ss": 9,
                                                        "2ss": 9,
                                                        "3ss": 9,
                                                        "4ss": -1,
                                                        "5ss": -1,
                                                        "6ss": -1,
                                                        "7ss": -1,
                                                        "8ss": -1
                                                }
                                        }
                                }
                        }
                }
        }
}

2. 5ghz wifi of ea8300. Containing a iPad air

cotequeiroz, 2021-07-27:
There's a bug when parsing the rrm table.  In `dump_client`, `dump_rrm_table` [is being called with the table data](https://github.com/berlin-open-wireless-lab/DAWN/blob/ec9a3a96e8c708a0ad85287fd3ed42cc0527e869/src/utils/msghandler.c#L409), instead of the pointer to the table itself.
Instead of looping through the table, and aborting after the first pass, we can get the data straight away, by changing the call from `dump_rrm_table` to `dump_rrm_data`.  I'm adding a debug line to show the parsed values with it as well.

PolynomialDivision, 2021-07-27:
Thanks a lot.

closed 2021-07-27 by PolynomialDivision

MrWGT, 2021-07-27:
Compiled openwrt and dawn, scped to mr8300 and works. Values shown 👍🏻

PolynomialDivision, 2021-07-27:
Nice!

MrWGT, 2021-07-28:
After one day runtime i found the following:

Both MACs cc:d2:81:41:72:90 and 4c:66:41:75:9d:49 answer beacon requests but both MACs are shown as RCPI RSNI as -1 at hearing map. Are there no values inside?

Wed Jul 28 21:07:29 2021 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS cc:d2:81:41:72:90 87 ack=1
Wed Jul 28 21:07:29 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 4c:66:41:75:9d:49 218 ack=1
Wed Jul 28 21:07:29 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:66:41:75:9d:49 0 00 012a4f6af2ab13000000010000bb15b8bef43dcb6f000000000001bef8243ed00b00000064001108000b574c414e2d34303530323901088c129824b048606c0301282d1aef0117ffff0000010000000000000000000000000000000000003d162807000000000000000000000000000000000000000030180100000fac020200000fac02000fac040100000fac0200007f080000000000000000dd180050f2020101800003a4000027a4000042435e0062322f000706474220241310bf0c3000c031faff0c03faff0c03c005012a00faffdd07000c4303000000
Wed Jul 28 21:07:30 2021 daemon.notice hostapd: wlan2: BEACON-RESP-RX cc:d2:81:41:72:90 87 00 0124cca5d33963fbc400010000c422e89f8015f471000000000001de01783d0c1000000043001110000f465249545a21426f7820537573693501069824b048606c030124050401020000070a494e2024041e3404180030180100000fac040100000fac040200000fac02000fac040c000b0501000c00004605720000000036036215013b0280002d1aef091bffff0000000000000000000001000000000000000000003d16240504000000000000000000000000000000000000007f080400080200000140bf0cb2198833faff0000faff0000c005012a00fcffc304023c3c3cdd180050f2020101810003a4000027a4000042435e0062322f00
Wed Jul 28 21:07:30 2021 daemon.notice hostapd: wlan2: BEACON-RESP-RX cc:d2:81:41:72:90 87 00 0105cca5d33963fbc400010000b30b60e3279b84d8000000000001d35744b7f23d00000064003104000c54502d4c696e6b5f38443141010882848b960c121824030105050400010000070c5553200b0114010a140c02142a010030180100000fac020200000fac04000fac020100000fac02000032043048606c2d1a2c0003ff000000000000000000000000000000000000000000003d1605080000000000000000000000000000000000000000dd1a0050f20101000050f20202000050f2040050f20201000050f202dd180050f2020101030003a4000027a4000042435e0062322f00dd0900037f01010000ff7f
Wed Jul 28 21:07:49 2021 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS cc:d2:81:41:72:90 88 ack=1
Wed Jul 28 21:07:50 2021 daemon.notice hostapd: wlan2: BEACON-RESP-RX cc:d2:81:41:72:90 88 00 01244996073b63fbc400010000c723e89f8015f471000000000001de0140710d1000000043001110000f465249545a21426f7820537573693501069824b048606c030124050401020000070a494e2024041e3404180030180100000fac040100000fac040200000fac02000fac040c000b0501000b00004605720000000036036215013b0280002d1aef091bffff0000000000000000000001000000000000000000003d16240504000000000000000000000000000000000000007f080400080200000140bf0cb2198833faff0000faff0000c005012a00fcffc304023c3c3cdd180050f2020101810003a4000027a4000042435e0062322f00
Wed Jul 28 21:07:50 2021 daemon.notice hostapd: wlan2: BEACON-RESP-RX cc:d2:81:41:72:90 88 00 01014996073b63fbc400010000aa057c4fb5cc5fc4000000000001cf8ad1b6437001000064001104000b574c414e2d434335463432010882848b962430486c0301010504000101002a01042f010430140100000fac040100000fac040100000fac02000032040c1218602d1a6c181fffff0000000000000000000000000000000000000000003d16010804000000000000000000000000000000000000004a0e14000a002c01c8001400050019007f0101dd090010180206f0050000dd180050f2020101000003a4000027a4000042435e0062322f00dd130050f204104a0001101044000102103c000103
Wed Jul 28 21:07:50 2021 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS 4c:66:41:75:9d:49 219 ack=1
Wed Jul 28 21:07:50 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:66:41:75:9d:49 0 00

MrWGT, 2021-07-28:
After dawn restart rcpi and rsni are show. logread -f shows

Wed Jul 28 21:52:40 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:66:41:75:9d:49 0 00 012a943fe23300000000020000cf23e89f8015f471000d0085ce01d886ecd2ad1000000043001110000f465249545a21426f7820537573693501069824b048606c030124070a494e2024041e3404180030180100000fac040100000fac040200000fac02000fac040c000b0502000400004605720000000036036215013b0280002d1aef091bffff0000000000000000000001000000000000000000003d16240506000000000000000000000000000000000000007f080400080200000140bf0cb2198833faff0000faff0000c005012a00fcffc304023c3c3cdd180050f2020101810003a4000027a4000042435e0062322f00
Wed Jul 28 21:52:40 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:66:41:75:9d:49 0 00 012a943fe23300000000020000bb15b8bef43dcb6f002a43000001be0c38d4710c00000064001108000b574c414e2d34303530323901088c129824b048606c0301282d1aef0117ffff0000010000000000000000000000000000000000003d162807000000000000000000000000000000000000000030180100000fac020200000fac02000fac040100000fac0200007f080000000000000000dd180050f2020101800003a4000027a4000042435e0062322f000706474220241310bf0c3000c031faff0c03faff0c03c005012a00faffdd07000c4303000000
Wed Jul 28 21:52:40 2021 daemon.notice hostapd: wlan0: BEACON-RESP-RX 4c:66:41:75:9d:49 0 00 012a943fe23300000000020000bd173023036ef5bb000000000001d8c8c366112800000043001110000f465249545a21426f7820537573693501069824b048606c030128070a494e2024041e3404180030180100000fac040100000fac040200000fac02000fac040c000b0504000500004605720000000036036215013b0280002d1aef091bffff0000000000000000000001000000000000000000003d16280704000000000000000000000000000000000000007f080400080200000140bf0cb2198833faff0000faff0000c005012a00fcffc304023c3c3cdd180050f2020101810003a4000027a4000042435e0062322f00