berlin-open-wireless-lab/DAWN #141: Entry with %d in Network Overview (non r wifi)

MrWGT, 2021-07-27:
The network overview shows my "r" ssid and also my 2 non "r" ssid. In one "non r" ssid there is one client (out of ~10) showing %d as signal strength (if network overview is viewed on the two other aps which do not have this ssid).

68:C6:3A:F4:38:E7 | available | not available | %d

On the ap with the "non r" ssid the signal is shown correct.

cotequeiroz, 2021-07-28:
Got the HD back.  Never mind.  It is dawn that can't find the client in the probe array, so it can't fetch the signal information, and sends the network map json without the signal information.  [Here](https://github.com/berlin-open-wireless-lab/DAWN/blob/009aab9ca4d60c570e97772a05adabd083ce590c/src/utils/ubus.c#L1579)'s where the signal gets added to the json.

The "clients" method shows the signal, so the AP has gotten it.  I suspect the [call to `probe_get_array_entry`](https://github.com/berlin-open-wireless-lab/DAWN/blob/009aab9ca4d60c570e97772a05adabd083ce590c/src/utils/ubus.c#L1575) does not find it even though it may be there.  Last time I saw something like that this happening, it was because the search function was not observing the SSID, but I have to check this one.

cotequeiroz, 2021-07-28:
The network map will only show the signal received from a `probe` method.  That's the only place the signal information is saved.  It does not do anything with the `signal` field received from a `client` method.  @PolynomialDivision, can you shed some light into this?

PolynomialDivision, 2021-07-28:
> The network map will only show the signal received from a `probe` method. That's the only place the signal information is saved. It does not do anything with the `signal` field received from a `client` method. @PolynomialDivision, can you shed some light into this?

I assumed I would receive a probe from an already connected client. However, this is not the case for all clients. [Here](https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/storage/datastorage.c#L667-L700) I update the rssi for all connected clients. I do this by updating the probe entry. If I have no probe_entry available, [I don't create one](https://github.com/berlin-open-wireless-lab/DAWN/blob/master/src/storage/datastorage.c#L933-L949). That is why the rssi is never updated. I can just create a probe entry. I would need to construct the information myself. Maybe it is also a stupid design decision to use the probe database as "main database". I don't know.

I'm very happy to see you involved in the code. I'm the sorry I'm currently in exam period. :/ Feel free to ask a lot of questions, and I hope I can find the time to answer in a good way. :)

PolynomialDivision, 2021-07-28:
Updating the Probe-Database periodically was needed since a client does not send a probe to the AP it is already connected. That was my inital intention.

PolynomialDivision, 2021-07-28:
I first steered only legacy clients with dawn, then I started to bring 802.11v and 802.11k to dawn. And now I'm unsure if all the probe stuff is still needed... ;)

PolynomialDivision, 2021-07-28:
But basically, it is enough to include the signal information I get from the "get_clients" call and use that for adding the network overview. DAWN somehow envolved and that is why some datastructures and stuff is written like it is. ^^