berlin-open-wireless-lab/DAWN #145: unknown configuration item 'bss_transition' - dawn on release 21.02.0

elcovv, 2021-09-13:
The dawn package is available for the 21.02.0, but 'wpad' doesn't seem to accept 'bss_transition':

After installing dawn and wpad (full) and configuring following commands to the wifi-iface sections of /etc/config/wireless:
```
    option ieee80211k '1'
    option ieee80211v '1'
    option bss_transition '1'
```

I got the following error messages:

```
Mon Sep 13 10:33:19 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Mon Sep 13 10:33:19 2021 daemon.err hostapd: Line 45: unknown configuration item 'bss_transition'
Mon Sep 13 10:33:19 2021 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy0.conf'
Mon Sep 13 10:33:19 2021 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy0.conf
Mon Sep 13 10:33:19 2021 daemon.notice netifd: radio0 (5792): Command failed: Invalid argument
Mon Sep 13 10:33:19 2021 daemon.notice netifd: radio0 (5792): Device setup failed: HOSTAPD_START_FAILED
Mon Sep 13 10:33:19 2021 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Mon Sep 13 10:33:19 2021 daemon.err hostapd: Line 42: unknown configuration item 'bss_transition'
Mon Sep 13 10:33:19 2021 daemon.err hostapd: 1 errors found in configuration file '/var/run/hostapd-phy1.conf'
Mon Sep 13 10:33:19 2021 daemon.err hostapd: Failed to set up interface with /var/run/hostapd-phy1.conf
Mon Sep 13 10:33:19 2021 daemon.notice netifd: radio1 (5793): Command failed: Invalid argument
Mon Sep 13 10:33:19 2021 daemon.notice netifd: radio1 (5793): Device setup failed: HOSTAPD_START_FAILED
```
Is it possible to get dawn working in 21.02.0? Any suggestions?

XiaoliChan, 2021-09-23:
@PolynomialDivision I got something confusing, should I set these parameters with dawn installed? If it needs this, please add these into ReadMe :)
```text
option ieee80211k '1'
option ieee80211v '1'
option bss_transition '1'
```

PolynomialDivision, 2021-09-23:
No, you don't need it. :) I activate that via ubus calls. You only need `full wpad installation`.

elcovv, 2021-09-23:
@PolynomialDivision Ok, great, thanks for the update.

closed 2021-09-23 by elcovv