berlin-open-wireless-lab/DAWN #146: [Q] 'kick' devices to 5hgz

tiagogbarbosa, 2021-09-14:
Can you please share a better config to 'force' devices to 5ghz?

My config
`config network
    option broadcast_ip         '192.168.0.255'
    option broadcast_port       '1025'
    option server_ip            ''
    option tcp_port             '1026'
    option network_option       '2' # 0 udp broadcast, 1 multicast, 2 tcp
    option shared_key           'Niiiiiiiiiiiiiik'
    option iv                   'Niiiiiiiiiiiiiik'
    option use_symm_enc         '1'
    option collision_domain     '-1'     # enter here aps which are in the same collision domain
    option bandwidth            '-1'     # enter network bandwidth

config ordering
    option sort_order           'cbfs'

config hostapd
    option hostapd_dir          '/var/run/hostapd'

config times
    option update_client         '10'
    option denied_req_threshold  '30'
    option remove_client         '15'
    option remove_probe          '30'
    option remove_ap             '460'
    option update_hostapd        '10'
    option update_tcp_con        '10'
    option update_chan_util      '5'
    option update_beacon_reports '20'

config metric
    option ap_weight            '0'
    option ht_support           '0'
    option vht_support          '0'
    option no_ht_support        '0'
    option no_vht_support       '0'
    option rssi                 '10'
    option low_rssi             '-500'
    option freq                 '100' # some kind of band-steering
    option chan_util            '0'
    option max_chan_util        '-500'
    option rssi_val             '-45'
    option low_rssi_val         '-50'
    option chan_util_val        '140'
    option max_chan_util_val    '170'
    option min_probe_count      '0'
    option bandwidth_threshold  '6'
    option use_station_count    '0'
    option max_station_diff     '1'
    option eval_probe_req       '0'
    option eval_auth_req        '0'     # no real reasoncode...
    option eval_assoc_req       '0'     # just deny assocs...
    option kicking              '1'
    option deny_auth_reason     '1'     # unspecified
    option deny_assoc_reason    '17'    # assoc rejected can't handle new station
    option use_driver_recog     '1'
    option min_number_to_kick   '3'
    option chan_util_avg_period '3'
    option set_hostapd_nr       '1'
    option op_class             '0' #802.11k stuff...
    option duration             '0'
    option mode                 '0'
    option scan_channel         '0'
    option rssi_weight '100'
    option rssi_center '-40'`

josteink, 2022-01-11:
What's the best way to do that? And how far behind would one be of you are running plain OpenWRT 21.02.x (ie "stable")?

josteink, 2022-01-15:
Thanks for trying to help out. 

When trying to improve my setup, I was linked endless Cisco-articles and other very technical information. 

So I gave up on DAWN for now.  

Hopefully these things will be more zero-config-ish in the future 😄