berlin-open-wireless-lab/DAWN #147: FR: resolve MAC addresses

tobi-nmx, 2021-10-02:
In the standard Luci "Wireless" view, hostnames from dhcp leases and /etc/ethers are displayed along with the MAC address. Would it be possible to add this feature to the DAWN network overview and hearing map in Luci as well? This would help me to understand which client is connected where/how - without the need of looking up dhcp leases manually. 
Thanks for considering!

PolynomialDivision, 2022-01-17:
Yes it is possible.

marazmarci, 2022-01-17:
Since DAWN is often used on dumb APs which don't directly interact with clients on the network layer (IP) because they just forward data link layer (Ethernet) frames, they wouldn't have the clients' IPs and MACs in their ARP table. I suspect this can vary by router hardware architecture, but I'm sure mine is working like that.

I wanted to have my WiFi clients listed by hostname and IP, not just by MAC in LuCI's `Wireless` view.
I was able to implement this by setting up a cron job running `fping` every 15 minutes, that "pings through" my whole subnet, filling the AP's ARP cache.
How to set it up:

1. `opkg update && opkg install fping`
2. add this crontab entry: `*/15 * * * * fping --count 3 --period 2000 --random --generate --quiet 192.168.1.0/24 >/dev/null 2>&1`

Note: When a newcomer client joins the network, its IP will be unknown until the next `fping` scan. Scans happen at minutes divisible by 15, eg. 10:45.

Source of the idea: https://forum.openwrt.org/t/associated-stations-list-in-ap-how-to-show-host-names/63475/5

turboproc, 2022-04-18:
Just exploring options, an alternative would be to translate the MAC address to a notation that includes the manufaturer name such that it would look like SamsungE_00:00:00. Not sure whether a LUA class exists to handle this easy.

tobi-nmx, 2022-04-18:
> Just exploring options, an alternative would be to translate the MAC address to a notation that includes the manufaturer name such that it would look like SamsungE_00:00:00. Not sure whether a LUA class exists to handle this easy.

Thanks for looking into this! I have lots of ESP8266 devices on my WIFI. All by the same vendor. The families (and companies) I know seem to have a preference for a specific device brand. My laptop uses some WIFI module vendor I have never heard about and cannot remember. So I'm not sure how helpful this would be. I'd like to vote for using the same lookups OpenWRT does in the Wireless view (names from DHCP leases or /etc/ethers).

Ian-Clowes, 2022-04-20:
There's a few things at play here:
- In the most current versions the goal is to share "enough" network information that every DAWN node can make a valide decision
- That doesn't mean that every node knows everything: in particular BEACON reports are not sent from one node to another, because they are only used on the node that a client is currently conected to
- So the hearing map of each node can differ to that of another node
- The DHCP names work simply on the device that is running DHCP, a bit less so on others
- I haven't thought what this would mean for multi-SSID / DHCP networks
- The ubus calls are intended to be "raw", so that LUCI has that info to work with

So overall it is something I've thought about a bit, but not rushing to do

doreks, 2023-07-25:
any update here with arp resolution?