berlin-open-wireless-lab/DAWN #156: DAWN logging

Ian-Clowes, 2022-01-02:
There has been comment by a few users that DAWN's logging is "noisy" but despite (or because of...) this they can't tell whether the behaviour they are seeing is correct.  I'm preparing a PR to address this.

I realise it'll be easier to comment when the PR is available, but I thought I'd invite input on the proposed logging functions since there may be things on your mind that I can add in advance and now could be a good time to make adjustments 😄.

One of the key aims is that the default level of logging should show necessary errors and warnings, plus a stripped back amount of information to confirm that DAWN is working.  That has become three levels of message: ERROR, WARNING and ALWAYS.  All that is seen on my small home network is a message every hour or so that a client is being kicked.

```
root@WN604:~# logread|grep dawn
Sat Jan  1 22:14:40 2022 daemon.info dawn[1101]: Station XX:XX:XX:XX:XX:XX: Kicking as probably NOT in active transmisison.
Sat Jan  1 23:51:15 2022 daemon.info dawn[1101]: Station YY:YY:YY:YY:YY:YY: Kicking as probably NOT in active transmisison.
Sun Jan  2 00:06:43 2022 daemon.info dawn[1101]: Station XX:XX:XX:XX:XX:XX: Kicking as probably NOT in active transmisison.
Sun Jan  2 00:21:09 2022 daemon.info dawn[1101]: Station ZZ:ZZ:ZZ:ZZ:ZZ:ZZ: Kicking as probably NOT in active transmisison.
```

If I then adjust the logging up a notch (via a new config item) to INFO and restart DAWN I see some extra logging, indicating which clients are being considered for kicking every 10 seconds and how many alternate AP candidates there were - this would include the kick neighbour report if I had caught one:
```
root@WN604:~# head -4 /etc/config/dawn
config local
    option loglevel             '1'

config network

root@WN604:~# /etc/init.d/dawn restart && logread|grep dawn
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: AP BSSID AP:AP:AP:AP:AP:AP: Looking for candidates to kick
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: Station MM:MM:MM:MM:MM:MM: Compared 0 alternate AP candidates
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: Station MM:MM:MM:MM:MM:MM: Current AP is best. Client will stay:
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: Station XX:XX:XX:XX:XX:XX: Compared 1 alternate AP candidates
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: Station XX:XX:XX:XX:XX:XX: Current AP is best. Client will stay:
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: Station NN:NN:NN:NN:NN:NN: Compared 0 alternate AP candidates
Sun Jan  2 00:33:24 2022 daemon.info dawn[1784]: Station NN:NN:NN:NN:NN:NN: Current AP is best. Client will stay:

```
I won't clutter this post with examples, but there are then two more logging levels TRACE and DEBUG that can be used for high- or low-level code path / outcome tracing detail.

All of this is achieved via a set of variadic macros which which wrap a few new logging functions to keep the impact on core code low, while allowing useful messages to be generated via format strings.  For example:

```
dawnlog_error("Memory allocation failed!\n");

dawnlog_info("Station " MACSTR ": Compared %d alternate AP candidates", MAC2STR(j->u8), ap_count);
```
Or for lower level tracing which would expand to include function name, file and line number:
`dawnlog_debug_func("Testing signal strength...\n");`

Is there anything I've missed and should consider before submitting?

namidairo, 2022-01-02:
> That has become three levels of message: ERROR, WARNING and ALWAYS

Do we need fatal as well, for the showstoppers that require dawn to exit?

> Is there anything I've missed and should consider before submitting?

Skimmed through the current master and put some thoughts below.

Since you're breaking it all out into the different log levels, you can easily get rid of DAWN_NO_OUTPUT if you haven't already.

I feel like this here https://github.com/berlin-open-wireless-lab/DAWN/blob/ddc007e32ced2df48212c0bc65ce2f77a8617bd2/src/storage/datastorage.c#L1667-L1689 probably either needs more specificity or just a little clean up as well.

I think there were a couple typos floating around in there too. Eg. https://github.com/berlin-open-wireless-lab/DAWN/blob/ddc007e32ced2df48212c0bc65ce2f77a8617bd2/src/network/tcpsocket.c#L314

I think this https://github.com/berlin-open-wireless-lab/DAWN/blob/ddc007e32ced2df48212c0bc65ce2f77a8617bd2/src/network/multicastsocket.c#L55 might output a double newline currently, so you may have to normalise them across the project. Which brings me to ask if we don't want to just output the newline in the logging functions, since even you forgot one in your example and one would probably use it in the majority of cases.

Ian-Clowes, 2022-01-02:
Thanks, there are some really good points you raise there.

The note I've put against ERROR is that it should be used for significant issues like OS exhaustion.  Whether and how DAWN then exits is a local code decision, so personally I don't see a need for FATAL - a relevant note like "quitting" can be added to the message.  The extra level can be added quite easily later if someone wants to.

All DAWN_NO_OUTPUT has been removed.  It was never really anything to do with managing output verbosity.  I originally added it to help do a compile time check on which core structure members were only referenced in print_... functions (rather than decision logic).  A similar effect can now be achieved using a concept of "minimum compile level" which is already in the new logging code if anyone ever wants to revisit removing redundant structure fields.

I'd missed the perror() uses.  I've a bit of new code that hopefully handles them appropriately using strerror_r() in the unlikely case of them being called.  This might be overkill as I can't recall ever seeing those messages occuring.

I haven't reviewed every message to see how unique / useful it is, but the dawnlog_error() macro expands to use __FILE__ and __LINE__ which should make them easy to align with triggering circumstances.  My intention is that ERROR is for technical failure, while WARNING is for user awareness - hence the rationale for making the former more "unfriendly".

Similarly, I haven't checked everything for typos.  I have fixed a few where I spot them and tidied up the messages that I now expect most users to see (ALWAYS and INFO).

The default revised behaviour is to use syslog() for output, so newlines are generally handled automatically.  I do provide a means to use stdio if required (eg running DAWN as non-daemon).  All it does is ensure that there is a newline on the end of the output format specifier, so it will be less graceful in the infrequent case of that type of use.

Ian-Clowes, 2022-01-05:
PR for new logging now here: #157

PolynomialDivision, 2022-01-05:
Wow, thanks a lot for your efforts! It is definitely completely useful, to rework the debugging! What i think could be also useful is, what @blocktrron already started in several commits for the hostapd: https://github.com/openwrt/openwrt/commit/54cfe0774ce4203d23ffcb397ba15bd36968d098

We could also add statistic counters for dawn, and display them using ubus commands. (how often is a station kicked? how often is a request denied...) What do you think?

Ian-Clowes, 2022-01-06:
> Wow, thanks a lot for your efforts! It is definitely completely useful, to rework the debugging! What i think could be also useful is, what @blocktrron already started in several commits for the hostapd: [openwrt/openwrt@54cfe07](https://github.com/openwrt/openwrt/commit/54cfe0774ce4203d23ffcb397ba15bd36968d098)
> 
> We could also add statistic counters for dawn, and display them using ubus commands. (how often is a station kicked? how often is a request denied...) What do you think?

Yes, it would be good to develop both those.

I'm planning to spend a bit of time making sure I understand 802.11 and DAWN a little better so I can understand what is expected behaviour rather than a bug.  For example, my Android phone only shows in one hearing map for a long time, but if I do a wifi scan from it then it quickly appears in more.  Is that right or should DAWN be getting the hearing map another way?  Also now the messages are stripped back I see warnings about "NULL MAC, strange client" and "Current AP not in array".  I'd like to also make sure they are "by design".

Once I'm comfortable on those then happy to look at the other things.

PolynomialDivision, 2022-01-06:
> For example, my Android phone only shows in one hearing map for a long time, but if I do a wifi scan from it then it quickly appears in more.

With a scan you suddenly create a lote of probe requests. :) And we collect probe requests to create the hearing map. However, we also add 802.11k feature called beacon-requests, that requests from clients to do a scan (active,passive,table...). The `"NULL MAC, strange client"` indicates, that your phone has issues with the beacon request.

Ian-Clowes, 2022-01-06:
> However, we also add 802.11k feature called beacon-requests, that requests from clients to do a scan (active,passive,table...). The `"NULL MAC, strange client"` indicates, that your phone has issues with the beacon request.

Yes, it's that lack of BR that I wanted to look at.  It makes transitioning for my phones very delayed, and is one of the key areas I want DAWN to work for.

Ian-Clowes, 2022-01-06:
> With a scan you suddenly create a lote of probe requests. :) And we collect probe requests to create the hearing map. However, we also add 802.11k feature called beacon-requests, that requests from [...]

I watched the logs a bit.  I see this:

```
Thu Jan  6 14:34:51 2022 daemon.notice hostapd: wlan0: BEACON-REQ-TX-STATUS XX:XX:XX:XX:XX:XX:139 ack=1
Thu Jan  6 14:34:51 2022 daemon.notice hostapd: wlan0: BEACON-RESP-RX XX:XX:XX:XX:XX:XX:139 04 00000...
Thu Jan  6 14:34:51 2022 daemon.warn dawn[13985]: Received NULL MAC! Client is strange!
Thu Jan  6 14:34:51 2022 daemon.notice hostapd: wlan0: BEACON-RESP-RX XX:XX:XX:XX:XX:XX:139 00 5106c...
```
So after the request I get a reply with 04 "indicator" and NULL MAC, and then one with 00 and proper data.  Do we know what the 04 means - should those responses simply be ignored if another one is following with good info?

Ian-Clowes, 2022-01-07:
> So after the request I get a reply with 04 "indicator" and NULL MAC, and then one with 00 and proper data. Do we know what the 04 means - should those responses simply be ignored if another one is following with good info?

From a quick read of the comment [here](https://w1.fi/cgit/hostap/commit/src/common?id=ec45927b8ac8bc77699988dc76cbdbf4f7a5a1b8) I assume the 04 is "<report mode>".  Whether that means they can / should be ignored I don't know yet.

Ian-Clowes, 2022-01-07:
> So the 04 with data appears to be a device speciifc quirk.

I found more to support this in the spec fragments that I could find, and the hostapd source code.

hostapd has this: `#define MEASUREMENT_REPORT_MODE_REJECT_REFUSED BIT(2)` - which agrees that 04 is saying refused.

I also found this, and the code shows that although the report mode is available to be put on the ubus message it isn't.  I don't know how disruptive it would be to add a "report-mode" element:
```
void hostapd_ubus_notify_beacon_report(
        struct hostapd_data *hapd, const u8 *addr, u8 token, u8 rep_mode,
        struct rrm_measurement_beacon_report *rep, size_t len)
```

PolynomialDivision, 2022-01-07:
> I also found this, and the code shows that although the report mode is available to be put on the ubus message it isn't. I don't know how disruptive it would be to add a "report-mode" element:

Awesome! I would love to add this. Should I, or do you want to add this?

Ian-Clowes, 2022-01-07:
> > I also found this, and the code shows that although the report mode is available to be put on the ubus message it isn't. I don't know how disruptive it would be to add a "report-mode" element:
> 
> Awesome! I would love to add this. Should I, or do you want to add this?

I saw you were connected to 600-ubus_support.patch which I think does this.  Prob nest for you to do it if you have some familiarity.

Ian-Clowes, 2022-01-07:
> ...so assume hostapd doesn't pass on empty messages

If I read things correctly we don't currently see 04 with no further data.   It might also be worth checking if this is a hostapd convention, or is a speciality of the way BEACON REPs are handled.  I think it might be interesting for DAWN to know which stations are refusing to reply.

Ian-Clowes, 2022-01-07:
> It might also be worth checking if this is a hostapd convention, or is a speciality of the way BEACON REPs are handled.

Looks like it is special to BEACON REPS.  I'm not sure this line is quite correct anyway unless the message is exactly the same length as a compiler aligned structure, but allowing "short" messages onto UBUS would allow all refused entries to be seen:

```
        if (len < sizeof(struct rrm_measurement_beacon_report))
                return;
        hostapd_ubus_notify_beacon_report(hapd, addr, token, rep_mode, (struct rrm_measurement_beacon_report*) pos, len);
```

But as I mentioned above, there seem to be some entries with 00 (indicating valid response) and no data, so they would beed ot be handled somehow as well.

PolynomialDivision, 2022-01-08:
@Ian-Clowes Can you maybe comment on the PR? https://github.com/openwrt/openwrt/pull/4934 Do you think it is good to do it like that? I can also modify it.

PolynomialDivision, 2022-01-08:
I just did this PR on the fly. I have time in the evening today, to analyze and runtest it.

Ian-Clowes, 2022-01-08:
> @Ian-Clowes Can you maybe comment on the PR? [openwrt/openwrt#4934](https://github.com/openwrt/openwrt/pull/4934) Do you think it is good to do it like that? I can also modify it.

At this level of bus communication I don't think expanding the bits is required, so for that part would just add the reported value.  That makes it easier to add a simple tests on the DAWN end as well.

The case that led us to this of "refused" with data is "special", and I think as a happy accident it keeps the access to rep-> from SEGVing.  So usually the message buffer will be too short and the other parts of building the message should be protected.  But I also saw those ACCEPT responses with no data as well, so think the test from the length test in the calling function has to be moved here as well (and deleted there otherwise it will still stop some REFUSED reporting).

Based on those wrong looking ACCEPT messages we would then see a new type of empty BSSID on the DAWN side, and should trap them before getting too far into processing them.  If they are very common we should perhaps make them INFO messages rather than WARNING so they have to be turned on by the user.

So I end up with:
```
        blobmsg_add_u8(&b, "report-mode", rep_mode);

	if (rep_mode == MEASUREMENT_REPORT_MODE_ACCEPT &&
                len >= sizeof(struct rrm_measurement_beacon_report)) {
            ...
            blobmsg_add_u16(&b, "antenna-id", rep->antenna_id);
	    blobmsg_add_u16(&b, "parent-tsf", rep->parent_tsf);
	} else {
            // Not sure anything goes here.  Are empty field values needed?
        }

```

And removal of this in the caller:
```
            if (len < sizeof(struct rrm_measurement_beacon_report))
                return;
```

Ian-Clowes, 2022-01-08:
> and should trap them...

We should be explicit on that.  A field something like one of these would help:
```
    blobmsg_add_u16(&b, "report-len", len);
    blobmsg_add_u8(&b, "report-fully-formed", (len >= sizeof(struct rrm_measurement_beacon_report)) ? 1 : 0);
```

Ian-Clowes, 2022-01-08:
It was useful that you copied the spec text.  I re-read the section on the PARALLEL bit.  I'm not sure it reflects an error condition, more of a "be aware".  Maybe the `rep_mode == MEASUREMENT_REPORT_MODE_ACCEPT &&` test above can be removed while we get some real world experience of inspecting that bit in DAWN?

Ian-Clowes, 2022-01-08:
User might also update DAWN without wanting to do the same with hostapd in the short term.  Making DAWN compatible with old and new messages would be a good thing.

Ian-Clowes, 2022-01-11:
Reason for opening this issue has been addressed by the PR, so closng it,

closed 2022-01-11 by Ian-Clowes