berlin-open-wireless-lab/DAWN #159: Developer and technical documentation

Ian-Clowes, 2022-01-11:
I've done a couple of rounds of work on DAWN, mainly revising existing code to make it more resilient, user friendly, etc.  Along the way I've found (and indeed created...) a few bugs, but that was largely secondary.  I've now paused a bit to look at some of the fundamentals of how DAWN works, the data structures it maintains, how they get updated, etc, so that I can double-check that all my local devices are behaving as expected, and whether there might be some bad behaviour that suggests bugs to chase down.

I intend to document those so that I have a reference, and others might be able to jump in more easily.  This thread / issue is to help refine that so it will be useful, so comments to that end are welcomed.

Things I intend to write, perhaps in a DEVELOPER.md document:

- Basics of 802.11k and 802.11r that DAWN makes use of
- How DAWN instances form a network (UMDNS)
- The data that is gathered by each DAWN instance from:
  - hostapd (associations, BEACON, NEIGHBOR REPORT, etc)
  - iwinfo (channel utilisation, etc)
  - Other DAWN instances
- How that data is represented internally to DAWN as lists of AP, client, probes / beacons, etc
- How kicking works by inspecting that data and using configuration parameters

What else might be useful?

Flole998, 2022-01-15:
The protocol used between access points. When things don't work as expected I'd take a look at that, so it'd be nice if it's documented somehow (how access points are found, how data is then transmitted between them and so on).

Also if you have lots of free time what should be done in my opinion is run Coverity's static analysis on this. It does a great job in finding and explaining potential issues, including memory leaks and memory corruptions. You'd have to fork the project and then sign up for Coverity, download the build tool and then compile dawn with it. You don't need any cross compiling or whatever, it just needs the compilation to go through, then you get a file which you submit to Coverity and you'll then see all issues in the Webinterface, pretty cool stuff. I know that's not what you actually intended to do but maybe you'll like that idea and want to have a look. I'll definitely soon (whenever that might be) do it if nobody else has done it until then just to see if there are any critical bugs in it before I actually get ready to deploy it on my production network ;)

Ian-Clowes, 2022-01-15:
Thanks.  Explaining how DAWN instances share data bewteen themselves is on the list.

Similar to you perhaps I am interested in overall code quality and robustness.  I added wrappers for standard memory functions to help see if there might be undiscovered problems, and test scripts to excercise some of the core functions with large amounts of data to see if they work OK.  I'm actually just looking at a memory leak I seem to have introduced while refactoring some code (so not in any pused code AFAIK), and that I can easily see the source of via the added memory auditing feature of sending the process a HUP signal:

```
...
Sat Jan 15 09:32:11 2022 daemon.info dawn: mem-audit:   457815=M - ubus.c@307: 72
Sat Jan 15 09:32:11 2022 daemon.info dawn: mem-audit:   456446=M - ubus.c@307: 72
Sat Jan 15 09:32:11 2022 daemon.info dawn: mem-audit:   456454=M - ubus.c@307: 72
Sat Jan 15 09:32:11 2022 daemon.info dawn: mem-audit:   459781=M - ubus.c@307: 72
...
```
Although it sounds interesting I don't expect to find time to look at the Coventry thing imminently.  If you do then that would be great.

One take away from this for me is to add some docuemntation notes on compile and run time features that support code quality, like the memory auditing.

Flole998, 2022-01-15:
Coverity found 26 issues, some are false positives/precautionary warnings (rand() should not be used) but others are real, for example in networksocket.c:106 the passed argument might not be properly null-terminated. Right before there is a useless comparison: "Comparing an array to null is not useful: recv_string == NULL, since the test will always evaluate as true".

In broadcastsocket.c:21 there is a ressource leak of `sock`, it needs to be closed in that case.

In datastorage.c:1563 there should be a return, entry must not be accessed afterwards but that could happen in the current version.

In test_storage.c:936 it warns about "Accessing pr0->time without holding lock probe_array_mutex. Elsewhere, probe_entry_s.time is accessed with probe_array_mutex held 2 out of 3 times (1 of these accesses strongly imply that it is necessary)."

In datastorage.c:1440 fp is accessed even though it was freed in the line right before it....

In dawn_uci.c:336 (and following lines aswell) we have "Calling strncpy with a maximum size argument of 46 bytes on destination array ret.broadcast_ip of size 46 bytes might leave the destination string unterminated."

If you sign up for coverity and request access to https://scan.coverity.com/projects/flole998-dawn you can have a look at the issue list aswell.

I simply love this tool, although I am not sure why it didn't find your memory leak.....

Ian-Clowes, 2022-01-16:
Great stuff - many thanks.  BTW, I think you've run theat on HEAD of master which makes sense, but the "rework_pr" branch is headed into master and is almost certianly a better base to build from if you're planning to deploy.

I'm also planning a new PR that will have revised standard messages to syslog that may make exploring what is happening easier if the network you're targetting is non-trivial.  It will also restructure some code and include tweaks driven by your findings, which leads nicely to...

>... in networksocket.c:106 the passed argument might not be properly null-terminated

I think line 104 takes care of that:
```
        recv_string[recv_string_len] = '\0';

        char *base64_dec_str = dawn_malloc(B64_DECODE_LEN(strlen(recv_string)));
```

> Right before there is a useless comparison: "Comparing an array to null is not useful: recv_string == NULL, since the test will always evaluate as true".

Yes, that had already been spotted but the required resolution was unclear.  Good prompt to update it to a test on the length though:
`// TODO: recv_string is a fixed array.  Should test be here?`

> 
> In broadcastsocket.c:21 there is a ressource leak of `sock`, it needs to be closed in that case.

I think failure there leads to exit(), but in case the OS doesn't free all process resources I've add that to broadcast and multicast code.  I think it's fair to say they are used far less than TCP, so there are probably other niggles in them as well though.

> 
> In datastorage.c:1563 there should be a return, entry must not be accessed afterwards but that could happen in the current version.

I think that function is protected by never being called 😄 .  I've removed it.

> In test_storage.c:936 it warns about "Accessing pr0->time without holding lock probe_array_mutex. Elsewhere, probe_entry_s.time is accessed with probe_array_mutex held 2 out of 3 times (1 of these accesses strongly imply that it is necessary)."

The test harnesses are single-thread, so not required.  Interesting that it suggests it though.
 
> In datastorage.c:1440 fp is accessed even though it was freed in the line right before it....

Looks like that might be a reference to fclose(fp) on branch master.  If so that's the semantics of the memory tracking - we're passing the pointer value to indicate we have indeed finished with it - no access to the pointer is made..
 
> In dawn_uci.c:336 (and following lines aswell) we have "Calling strncpy with a maximum size argument of 46 bytes on destination array ret.broadcast_ip of size 46 bytes might leave the destination string unterminated."

Not sure what the desired semanics of those sizes being with / without NUL are, but I've done a +1 on the array allocations which should be safe.

> I simply love this tool, although I am not sure why it didn't find your memory leak.....

The leak hadn't been pushed - it was just in my current workset.

Flole998, 2022-01-16:
We were probably looking at different versions of the source:
> I think line 104 takes care of that:
> 
> ```
>         recv_string[recv_string_len] = '\0';
> 
>         char *base64_dec_str = dawn_malloc(B64_DECODE_LEN(strlen(recv_string)));
> ```

The problematic line is right before that:

```
       if (strlen(recv_string) <= 0) {
            return 0;
        }
        recv_string[recv_string_len] = '\0';

        char *base64_dec_str = dawn_malloc(B64_DECODE_LEN(strlen(recv_string)));

```
Not entirely sure what the point of that strlen even is, but it could potentially read beyond the buffer. So that null-termination should be moved up before that.

I'll re-run it on rework_pr (and then I'll get a summary of fixed/new issues aswell ;) ). If that's what you're working with and what the new master will be soon(ish) then it makes sense to analyze that.

Ian-Clowes, 2022-01-16:
> The problematic line is right before that: ...

The whole thing looks a bit mangled.  For now I've made that `strlen(recv_string, recv_string_len)`.  Can't be any worse, can it 😕 ?

Flole998, 2022-01-16:
It seems to be an attempt to return on an empty string (the less-than operator is useless, strlen never returns a negative value, so it just checks if the length is 0), so even swapping those lines and null-terminating it shouldn't be an issue, but your approach works aswell I guess. I assume that both have the same result (on empty string with proper 0-termination both cases return, when the null byte is missing then both return recv_string_len (which might be 0, in which case it returns)). 

Just checked the other branch: 5 new defects, 1 fixed. It picked up the test cases for memory leak and segfault and so on ;)

But it also picked up a mistake in mac_utils.c, after line 52 a return is missing, otherwise the close would be called with a null-pointer.

It's still quite a long list so I'm not going through it here completely, especially since many issues are occuring multiple times (msghandler.c 560/568/572, strncpy might leave the destination array unterminated due to the array length and the limit being the same, still better than in 531 where there is no lenght limit at all).

Just one real nice find: dawn_uci:255 checks if name is null and then still uses it in the strcmp, there should be an additional check before the strcmp I guess. Even though that might not occur during normal operation it's still a good idea to fix that so in case the code is modified later on by someone else they don't accidentally introduce a bug.

Ian-Clowes, 2022-01-16:
> Just one real nice find: dawn_uci:255 checks if name is null and then still uses it in the strcmp, there should be an additional check before the strcmp I guess. Even though that might not occur [...]

I'm not sure if that is relying on knowledge of how the parametes will be set and the C language behaviour that the second test of the logical-OR won't be evaluated if the first part is TRUE.  That may be reasonable for any sensible config file, but it looks like there could be a path to the potential bug if it was written in a certain way - I'll try and trip it up so I know what I'm fixing :smile:.

Flole998, 2022-01-16:
There are 7 places (iirc) where the 0-termination could potentially be missing.

Don't forget, we only have 23 findings (after I dismissed the intentional/false positive ones), so it's really not that bad. I believe you already fixed at least 2 of them in your code, so we could easily get it down to 0. If you don't want to work on the arrays then I'd come up with a PR that increases the size by 1 for all of the ones that are currently found by Coverity. We just need to make sure we aren't working on the same thing at the same time, otherwise that'd be a waste of time ;)

PolynomialDivision, 2022-01-17:
> I'm actually just looking at a memory leak I seem to have introduced while refactoring some code (so not in any pushed code AFAIK), and that I can easily see the source of via the added memory auditing feature of sending the process a HUP signal:

Whups. I pushed it to master. However, I will not push it to OpenWrt packages master until we find the bug (https://github.com/openwrt/packages/pull/17627) Or did you only introduce the memory leak in your own branch (I am not sure anymore after reading whole discussion)?

> If you sign up for coverity and request access to https://scan.coverity.com/projects/flole998-dawn you can have a look at the issue list aswell.

Awesome. I will do that.

> The leak hadn't been pushed - it was just in my current workset.

Sorry, I was very ill and now I have to fetch up all discussions and what was happening.

@Ian-Clowes Did you already addressed some of the mentioned issues from @Flole998 ? Could you maybe do some PR, so we can sync again? Please, if anything is important just mention me again. I got somehow out of sync what was happening here.

Thanks for taking so much time contributing to DAWN!

Ian-Clowes, 2022-01-17:
> > I'm actually just looking at a memory leak I seem to have introduced while refactoring some code (so not in any pushed code AFAIK), and that I can easily see the source of via the added memory auditing feature of sending the process a HUP signal:
> 
> Whups. I pushed it to master.

I think I'm confusing you.  It was in the code I'm now working on - not the previous PR.  I don't know of any leaks or significant bugs in the last PR.

Ian-Clowes, 2022-01-17:
> @Ian-Clowes Did you already addressed some of the mentioned issues from @Flole998 ? Could you maybe do some PR, so we can sync again? Please, if anything is important just mention me again. I got somehow out of sync what was happening here.

Although the things found by @Flole998 are certainly worth addressing I think (without going back to review comments above) they are all in parts of the code that are never visited or are edge cases for things like full-length SSID.  Someone could hit the latter, but haven't so far AFAIK.  So I don't think any of them need rushing through.

I have added them to my latest workset, but it also contains a whole bunch of layered changes that I am hoping to refactor to a set of coherent commits in the same way as last time - pick a "topic" generate a diff for it; repeat until all changes are handled in a set of topic-specific patches that add up to all the revisions.

I know my updates touch various parts of code and even if it is just whitespace where I have restructured something into a if / else then it may clash with updates to things like strncpy() arguments.  If I have my next set of revisions in a few days I think that would be a good time to sync around a Coverity issue based update while avoiding commit clashes.

edrikk, 2022-01-21:
I see that these have been merged into openwrt master package which is great given the improvements and fixes.
Wondering about 2 things:

- Any thoughts about multi-AP environment where some may be running master (e.g. due to support only existing in Master) and APs running in stable?  Not sure if the fixes depend on all APs running the same version to "cooperate" or not.
- Related to above question, I'm not sure of the process or intentional lag period before also pushing to stable branch.  Any thoughts on that?

Thanks!

edrikk, 2022-01-22:
@PolynomialDivision @Ian-Clowes 

Want to share that as I've just noted [here](https://forum.openwrt.org/t/dawn-a-decentralized-wireless-controller/108768/92?u=edrikk), I compiled Robi's latest codebase for AX3600 (Kernel 5.15).  This compiled in the updated DAWN from Openwrt master.

I noticed that the 2 clients that weren't able to connect with the old version (which was also pre-this update) are now able to connect without having to first turn off DAWN.  I think that the fixes here are likely the reason why.

I'll keep an eye on it, but great job! :)

Ian-Clowes, 2022-01-24:
>     * Any thoughts about multi-AP environment where some may be running openwrt master (e.g. due to support only existing in Master) and APs running in stable?  Not sure if the fixes depend on all APs running the same version to "cooperate" or not.

It's something I think about when pushing changes.  I think this is pretty safe, although I don't have enough devices and / or time to test it hard.

The messages between DAWN instances are effectively textual key / value pairs, and so any mismatches just get silently absorbed by the other end ignoring extra and / or defaulting missing values.  So that should prevent crashes that might occur if "one view of the world" binary type messages were being used.

I also don't think there are issues with AP A wanting to push a device to AP B, and AP B wanting to go the other way due to differerent algorithms, etc.  They aren't changing much and there is a scoring hysterisis / buffer where one AP will "tolerate" a less-than-perfect device before wanting to kick it back.  This is perhaps more likley to happen if two APs have configs different enough (scoring weights, values, etc) to create an impance.

I'll leave the 2nd Q to @PolynomialDivision, as I don't deal with that bit at all 😄 .

Ian-Clowes, 2022-01-25:
Context switch...

Can anyone help me understand the significance of threading for DAWN a little better so I can add a paragraph on that to dev docos?

I know there are mutex in the code around data structures and that makes perfect sense if different threads could be accessing them due to various bits of code being fired by timers, ubus events, etc.  To help ensure they were being used appropriately I added some dev-time macros to show how they were being used, and spot access to "protected" structures outside a lock.

I used pthread_self() for this, but always get the same value back.  Is that an artefact of my particular device, where plib doesn't really do any threading at all?  Are other platforms likely to be "more threaded"?  Should I use a different methof to identify different threads?

Or is it a reality of OpenWRT that ubus, timers, etc are really just managing which code "entry point" / callback will run next, and as long as nothing is cached from one scheduled slice to the next (eg storing a pointer to a client record) nothing else will jump in and mess things up?  Which would lead to the quesiton of whether the mutex are really needed.

Ian-Clowes, 2022-01-28:
For anyone awaiting the developer notes, I've been visiting pretty much all parts of the code to help me build them.  That has led to work on various items that roll-up into the PR that I've proposed ( #160).  I'll see if I can find time to do first version of the notes to accompany that, otherwise it will follow shortly.

PolynomialDivision, 2022-01-31:
> whether the mutex are really needed.

Good question, I have to look into documentation, or ask people.

> Related to above question, I'm not sure of the process or intentional lag period before also pushing to openwrt stable branch. Any thoughts on that?

I am always waiting a bit for reports from users and then push to stable. I can do it. :)

Ian-Clowes, 2022-02-07:
For those who are interested, "late draft" documentation updates are here: https://github.com/Ian-Clowes/DAWN

I'll tidy up some typos, add more if you comment, etc before the PR.

edrikk, 2022-02-07:
Looks great!  Just a minor typo, and a thought:

Within the [Client Connection Scoring section of the Configure Page](https://github.com/Ian-Clowes/DAWN/blob/master/CONFIGURE.md#client-connection-scoring), there's a typo "These values ar ehten" instead of "are then".

The document's "Note 1" regarding Legacy Clients is really helpful.  But in the overall I wonder if it can be enhanced a little.  Basically it currently says "If you enable this legacy client behaviour via parameters" and lists the parameters as footnotes.
The table below (where the parameters are) really nicely ties back to this section as well.  However, it's not really clear how to disable this behaviour for legacy clients.   Is it just to set the "kicking" value to 0?  Or would the user have to disable all the other parameters but leave kicking on and set the other parameters to 0 to disable?   Would that have any impact on the non-legacy clients?

PolynomialDivision, 2022-02-07:
>  If DAWN continues to try to not accept the client it is effectively denied wifi access.
There is actually something missing. In the `remove_old_denied_req_entries` it is checked if a denied client is already connected to the network somewhere. Otherwise the client is seen as problematic and is no longer steered. However, I need to test that again and it introduces also new issues and should maybe removed completely.

Ian-Clowes, 2022-02-07:
> it's not really clear how to disable this behaviour for legacy clients

Putting the `enable_...` parameters to zero means legacy clients get no special treatment.  They will recieve (but ignore as they don't understand it) the RMM DISASSOCIATE message.  Then hostapd will do the DISASSOCIATE anyway and they will have to find somewhere to connect to.  They may scan and find the better AP that DAWN wanted them to go to, or do the simple thing and come straight back to where they have just lost the connection to.

I'm going to look at a "state machine" type model for legacy clients to try and make the steering more robust.  There is the counter that is mentioned in the parameters, but I suspect its intended behaviour has been broken along the way.

Ian-Clowes, 2022-02-07:
> it is checked if a denied client is already connected to the network somewhere

My reading of the code was that the denied_req list wasn't being referenced for any decisions, so the whole list was removed in a recent commit.  It could come back if needed...

Flole998, 2022-10-26:
Is there any progress on this? So far there is no PR yet?

PolynomialDivision, 2022-10-27:
> Is there any progress on this? So far there is no PR yet?

What exactly do you reference?

Flole998, 2022-10-27:
The improved documentation by @Ian-Clowes, his repo is ahead of this one with several good commits. Or does the rework-pr branch needs to be merged first?

PolynomialDivision, 2022-10-27:
I pulled in all changes?

Flole998, 2022-10-27:
Indeed, I missed that. What about the pr-rework branch then? Is that also pulled in? I wanted to deploy DAWN on a system for testing but I want to be sure it's the "best"/latest version with all the improvements that have recently been done.

PolynomialDivision, 2022-10-27:
Just go for master branch. :)

Flole998, 2022-10-28:
Alright, done!

I noticed the configuration manual at https://github.com/berlin-open-wireless-lab/DAWN/blob/master/CONFIGURE.md is partially incomplete, for example key and iv aren't used according to that but they occur in the code, so they are used?

PolynomialDivision, 2022-10-28:
Only if you enable symmetric encryption. Default it is turned off.