berlin-open-wireless-lab/DAWN #160: New PR on the way: mainly bugs, NRs and user messages

Ian-Clowes, 2022-01-28:
I'm planning to have a new RP published ovr the weekend. Prior to me doing the final refactoring and tidy up I thought it'd be worth sharing what is in it in case I missed anything significant that can be addressed in time.  Here's a summary of the notes for the overall PR, and what will become the commits witin it.  Comments welcome.

**Main Headlines**
- Local Neighbor Report auto-generated for AP from current clients (works best in large network)
- Revise activity messages to support user awareness and confidence in operation
- Added RRM capability info to hearing map to help understand how device will behave
- Streamline kicking algorithm to help ongoing evolution
- Address potential / actual bugs spotted by Coverity static analysis
- Resource mutex tracking to support code robustness
- Internal data structures simplified
- Continue to "flatten out" code to assist long term scalability
- Remove some unused parts of code (eg denied auth tracking)

**Commits...**

IEEE80211:
- Move some RRM utility functions around to simplify code structure / #includes

CLIENT_REQ:
- Remove the linked list of rejected auth / assoc as it was not used for any decision making
- Rename auth_req to client_req as it is also used by association requests

MAC:
- Simplify linked list search - code was unnecessarily complex for no benefit
- Adjust some MAC address handling to simplify code

AP:
- Simplify linked list search - code was unnecessarily complex for no benefit
- AP record is uniquely identified by BSSID, so remove unnecessary search on SSID as well
- Change some data handling to simplify passing of records, pointers, etc
- Assure sort order to simplify generating network map
- Removed unused fields for collision domain and bandwidth

PROBE / BEACON:
- New skipping linked list structure - possible bug in previous version
- Adjust some function names to better reflect what actually happens
- Remove redundant configurable sort elements
- Remove probe structure SSID field which is not set via BEACON, so is unreliable
- Adjust way the different fields from BEACON and PROBE reports are handled / merged
- Made client / BSSID paramater order consistent for probe / beacon functions to help avoid cut/ paste bugs
- Add RRM capability, RCPI and RSNI to client in hearing map (which is a summary of probe / beacon entries)

CLIENT:
- Simplify linked list search - code was unnecessarily complex for no benefit
- Refactored parse / kick logic to simplify functions

NETWORK:
- Merge encoded / plain messge handling to ensure fixes land in both
- Fix some potential resource leaks spotted by Coverity

MUTEX:
- Wrap mutex calls with traceable messages to help debugging resource locks

MAPS:
- Simplify loops and scans in generating hearing map

CLEANUP:
- Remove a few things that are no longer needed: #includes, etc

MSGS:
- Add messages that help users understand what PROBE and BEACON messages are being used

BUGS:
- Fix a couple of bugs in hardly ever (never?) used code paths that Coverity spotted
- Add comments for things that look a bit odd, but not sure if they are bugs

TESTS:
- Tighten up a bit of test path coding

MUTEX2:
- Add dawn_mutex_require() to indicate when code is accessing resources
- Use pthread_mutex_trylock() to actively test required mutex are locked

SCORING:
- Simplify how score is tracked
- Streamline some code paths for scoring evaluation

NEIGHBOR REPORT:
- Tweak disassociate / kick NR code to reduce build overhead
- Add set_hostapd_nrdynamic == 2 to build of AP local station NR from what current clients hear elsewhere
- Tighten up hex string length usage for strncpy(), etc

Ian-Clowes, 2022-01-31:
I've pushed that to my fork: https://github.com/Ian-Clowes/DAWN

I've made vary minor typo type changes to what I'm currently running, but having been caught out by that before I won't open a PR until I've had the exact same code running for a few days.  If others want to try it to that'd be grand 😃.

PolynomialDivision, 2022-01-31:
Wow! Awesome! Thanks a lot for your contributions. Sorry, I am so busy right now with other things.

edrikk, 2022-02-11:
I've been running this build for a week now (actually on a Robimarko kernel 5.15 build) - it is working flawlessly.
In-fact the current set of changes in Ian-Clowes' git fixes my misbehaving IoT devices getting kicked repeatedly by DAWN if "kicking" is set to 1.
Will share this feedback once the PR is opened, but did want to share that here as well.
Great job!

Ian-Clowes, 2022-02-12:
I'm pleasd it is running well.  I noticed a very slow memory leak from looking at the runtime memory auditing.  I'll fix that plus a couple of other things on my mind and do the PR.

Ian-Clowes, 2022-02-17:
A discussion on the OpenWrt forum led to a couple more changes, which delayed things, but have now pushed again to my fork.  As ever I'm going to run that for a few days, and will then be happy to PR it.

Functional changes:
- New "soft" kick algorithm for 802.11v clients - always invite them to move within NR group if below absolute RSSI level (which is different to original algorithm which required them to share RSSI level with other APs)
- RSSI values handeld in slightly diffeent way to make them exposed more often (and RCPI, but they aren't really used)
- Refresh on parameters, including new default values that work well for me :).  Also included sample config file.

@edrikk, if you're able to test that'd be great

edrikk, 2022-02-17:
Perfect.  Will do a build tonight and report back.  :)

edrikk, 2022-02-18:
@Ian-Clowes I have built and installed.  My setup is:

Xiaomi AX3600 - Running Robimarko's latest build - Kernel 5.15.  This is an AX Device. 
3 Other AC only AP devices.

I can only build from code for the Xiaomi AX3600, so the other devices are all running 21.02 (latest DAWN versions in 21.02 snapshot).

I did a reset of all the configs given your config updates.  The only adjustment I made was set kicking mode to 1 on the AC devices, since they are running the version of DAWN that doesn't support the new default mode 3.  The AX router is set to kicking=3.

Everything appears to be working at first glance.

Only thing I've noticed is that on the AX device, the system log shows repeating:

Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Thu Feb 17 22:11:31 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.

Ian-Clowes, 2022-02-18:
`https://github.com/berlin-open-wireless-lab/DAWN/issues/160#issuecomment-1043808547`

I suspect I know what that is, and it'll be new.  If I have it right it'll be harmless for all normal configurations.

Ian-Clowes, 2022-02-18:
@edrikk, you can try this patch to see if it gets rid of that...

```
diff --git a/src/storage/datastorage.c b/src/storage/datastorage.c
index b847d42..5d08577 100644
--- a/src/storage/datastorage.c
+++ b/src/storage/datastorage.c
@@ -822,6 +822,18 @@ probe_entry* probe_array_update_rssi(struct dawn_mac client_addr, struct dawn_ma
 {
     dawnlog_debug_func("Entering...");

+    int freq = 0;
+
+    dawn_mutex_lock(&ap_array_mutex);
+    dawn_mutex_require(&ap_array_mutex);
+
+    ap *this_ap = ap_array_get_ap(bssid_addr);
+
+    if (this_ap)
+        freq = this_ap->freq;
+
+    dawn_mutex_unlock(&ap_array_mutex);
+
     dawn_mutex_require(&probe_array_mutex);
     probe_entry* probe_req_new = dawn_malloc(sizeof(probe_entry));
     probe_entry* probe_req_updated = NULL;
@@ -831,6 +843,7 @@ probe_entry* probe_array_update_rssi(struct dawn_mac client_addr, struct dawn_ma
         probe_req_new->client_addr = client_addr;
         probe_req_new->bssid_addr = bssid_addr;
         probe_req_new->signal = rssi;
+        probe_req_new->freq = freq;

         // Other fields in case entry is new
         probe_req_new->ht_capabilities = false;

```

edrikk, 2022-02-18:
Ok will do.  Likely will be a couple of days until I can report back though.

It looks like a mutex lock you've added as well, which may be related to what I came to post.  Seeing a lot of the below in syslog.
The ratio of "42A7E8" to "42A790" is probably close to 10 to 1.  These are the only two I see though.

`Fri Feb 18 14:02:09 2022 daemon.warn dawn: MUTEX require = 42A7E8@datastorage.c:120[2514967648l] - appears to be UNLOCKED!
Fri Feb 18 14:02:09 2022 daemon.warn dawn: MUTEX require = 42A7E8@ubus.c:2034[2514967648l] - appears to be UNLOCKED!
Fri Feb 18 14:02:09 2022 daemon.warn dawn: MUTEX require = 42A7E8@datastorage.c:120[2514967648l] - appears to be UNLOCKED!
Fri Feb 18 14:02:10 2022 daemon.warn dawn: MUTEX require = 42A790@ubus.c:1991[2514967648l] - appears to be UNLOCKED!
`

I don't believe I see any combinations other than: 

42A7E8@datastorage.c:120
42A7E8@ubus.c:2034
42A790@ubus.c:1991

Ian-Clowes, 2022-02-18:
> It looks like a mutex lock you've added as well, which may be related to what I came to post.

Thanks.  As far as I can see thay are all related to using `ubus call dawn get_network`, which had slipped out of my manual test cases.

edrikk, 2022-02-20:
Hi @Ian-Clowes 

Compiled and has been running overnight.
The mutex unlocked issue is gone.

Although the number of the occurrences seems to be lower, there are still several frequency is beyond last known band messages.

Sun Feb 20 08:24:22 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 08:24:32 2022 daemon.warn dawn: frequency 3014664 is beyond the last known band. Using '802_11a' band parameters.

elmr91, 2022-02-20:
Hi @Ian-Clowes,

I have a very a similar setup to Edrikk's one:
- 3 dumb AP on same subnet with 802.11r (same SSID/password) on 2.4 Ghz and 5 Ghz
2 AP are running dawn from 21.02.02 repo.
Third one is running 21.02.02 + your dawn version (archer c7v2 with wpad-wolfssl / option kicking=3)

I had the same errors:
Sun Feb 20 16:39:31 2022 daemon.warn dawn: frequency 1645370357 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 16:39:31 2022 daemon.warn dawn: frequency 1645370357 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 16:39:41 2022 daemon.warn dawn: frequency 1645370357 is beyond the last known band. Using '802_11a' band parameters.

So I applied fix you suggested (in addition to the mutex fix).
It fixed the problem.

I have another problem : 
It seems kinking is not always effective:

I have seen dawn trying to kick a device (3/3 kicking message).
kicking counter increased in logs, but device remains connected.
(I'll try to provide logs for this case)

I have another device with low RSSI:
Dawn logs multiple messages:

Sun Feb 20 16:52:00 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 16:53:01 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 16:53:24 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 04:b1:67:27:ba:0a 87 ack=0
Sun Feb 20 16:53:42 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 16:54:12 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 16:54:53 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 16:55:23 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000

No deauthenticate message in the logs....

Clients disconnects by itself few minutes later 
Sun Feb 20 17:05:52 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 17:06:22 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 17:06:53 2022 daemon.info dawn: Client 04:B1:67:27:BA:0A: Kicking as probably NOT in active transmission. RxRate is: 6.000000
Sun Feb 20 17:06:59 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 04:b1:67:27:ba:0a
Sun Feb 20 17:06:59 2022 daemon.info hostapd: wlan1: STA 04:b1:67:27:ba:0a IEEE 802.11: disassociated due to inactivity
Sun Feb 20 17:07:00 2022 daemon.info hostapd: wlan1: STA 04:b1:67:27:ba:0a IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

May be I missed something.
Is there something to configure in addition to kicking=3 in config file ?

elmr91, 2022-02-20:
It may be worth downgrading log level for the "No station connected" message.
It is currently set to warning : dawnlog_warning("No station connected\n");
It may be lowered to info level.
When a SSID is empty (guest SSID for example), logs are filled with  "No station connected" message even when running with minimal log level.

edrikk, 2022-02-21:
> Will bump up logging verbosity in case you need more info.

Bumping up debug verbosity to 1...  Can raise higher if needed.

`
Sun Feb 20 20:27:40 2022 daemon.info dawn: AP BSSID AA:AA:AAA:AA:DE:B0: Looking for candidates to kick
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372232 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client 28:BB:BB:BB:5A:D6: Compared 0 alternate AP candidates
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372232 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client 28:BB:BB:BB:5A:D6: Current AP is best. Client will stay:
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372328 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client CC:CC:CC:06:7A:FF: Compared 1 alternate AP candidates
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372328 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client CC:CC:CC:06:7A:FF: Current AP is best. Client will stay:
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372520 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client DD:DD:DD:64:37:3B: Compared 0 alternate AP candidates
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372520 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client DD:DD:DD:64:37:3B: Current AP is best. Client will stay:
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372632 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client EE:EE:EE:78:E2:62: Compared 0 alternate AP candidates
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372632 is beyond the last known band. Using '802_11a' band parameters.
Sun Feb 20 20:27:40 2022 daemon.info dawn: Client EE:EE:EE:78:E2:62: Current AP is best. Client will stay:
Sun Feb 20 20:27:40 2022 daemon.warn dawn: frequency 4372728 is beyond the last known band. Using '802_11a' band parameters.
`

edrikk, 2022-02-26:
It’s been a week now.  Just wanted to report back that other than what’s noted above, everything is working well.

Even the %d RSSI values in the UI are fixed :)

noblemtw, 2022-03-02:
Before pushing the PR, I think it's worth amending the "Client is probably in active transmission. Don't kick! RxRate is:" log line to include the configured threshold. I ran in to an issue yesterday where some of my clients weren't getting kicked despite the RxRate being under what I thought my configured threshold was. Turned out it wasn't even using the threshold I'd got configured, but that wasn't obvious without the extra bit of logging
I've still no idea why it wasn't working with my original configuration, but I started from scratch and it's all good now...

elmr91, 2022-03-04:
> I think what you're seeing may be cause by "type 2 / 3 kicking". This just advisory, and not followed by a forced deauthentication (which type 1 does do). If the device is not 802.11v enabled it will definitely ignore it, and even enable ddevices might decide they want to stay.
> 
Just to be sure, I changed kicking from 3 to 1 on AP1 (Ian's version) .
When moving the device from AP2 (current dawn from openwrt) to AP1, WMM kicking is used, my test device is effectively kicked out.
When moving from AP1 to AP2 it is not.
I added some logging around kick_type to try to understand: strangely, even with kicking=1, bss_transition_request is called.
It seems not working with this device or this AP - Archer C7 v2

I have to make additional tests to understand what happens...

Ian-Clowes, 2022-03-04:
> even with kicking=1, bss_transition_request is called

Is that based on looking at what hostapd sends, or the DAWN log messages?  Up until recently the log message and actual ubus method used weren't aligned - I think the 802.11 methods had been explored without adjusting the message.

Let me know if you want a hand digging deeper.

elmr91, 2022-03-05:
> Is that based on looking at what hostapd sends, or the DAWN log messages? Up until recently the log message and actual ubus method used weren't aligned - I think the 802.11 methods had been explored without adjusting the message.

I added logs around calling:
- sync_kick = wnm_disassoc_imminent
- bss_transition_request

bss_transition_request was called.
My configuration may be broken. I will add more logs and retry.

Ian-Clowes, 2022-03-06:
I do like it when the resolutuon is logical :)

Fail-Safe, 2022-03-11:
Any updates on ETA for this PR? Awesome work on this, by the way! 💯

supperchym, 2022-03-14:
Hi, thank for your project. I have a question, i see wifi always switch from 5ghz to 2ghz when signal ~ -70dBm, is there a way to make it stay at 5ghz band when the signal is weaker than -70dBm (eg -80dBm ). One more question can add non-openwrt ap in dawn (e.g. unifi ), i can confirm unifi ap uses same 802.11r setting as openwrt, they can detect openwrt ap and roaming from unifi to openwrt very fast and good. Conversely roaming from openwrt to unifi is not so good, i need to manually connect to unifi once for roaming to work ( I will have to do the same thing if I restart unifi ).

edrikk, 2022-03-20:
@Ian-Clowes the build has been running without noticeable items beyond what’s noted above for weeks.  This is on an AX3600 running Kernel 5.15 and an AX6S running Kernel 5.10.

It might be good to push anything further you’ve done for wider testing?

Ian-Clowes, 2022-03-26:
I think I've nailed the rest of the "beyond the last known band" messages.  So as long as I can see kicking still working I'm really-really aiming to get my fork updated this w/e 😄.

Ian-Clowes, 2022-03-26:
> Hi, thank for your project. 

It isn't really mine.  @PolynomialDivision got it all started - I just happen to have more time to refine things at the moment.

> I have a question, i see wifi always switch from 5ghz to 2ghz when signal ~ -70dBm, is there a way to make it stay at 5ghz band when the signal is weaker than -70dBm (eg -80dBm ).

Possibly.  Are you just using the original 'option kicking 1`'?  If so then ensuring that 5GHz stations get a higher score should be possible via the band specific `initial_score` parameter.  But there is always the challenge that when a device is asked to move the highest score should be the first proposed to it, but it may choose something else based on its on-device algorithms.

If using `option kicking 3` (or 2) then this is just a request to the device to find a new AP, so it can choose anything.

Having said that there could well be some bugs in the selection or ordering of proposed AP for the 802.11v messages, and if you / others can trace them at a low enough level it might give insight that I've missed.

> One more question can add non-openwrt ap in dawn (e.g. unifi ), i can confirm unifi ap uses same 802.11r setting as openwrt, they can detect openwrt ap and roaming from unifi to openwrt very fast [...]

DAWN doesn't do anything with 802.11r Fast Transition.  DAWN will suggest a chnage of AP to the device, and it can then use FT if it is configured across old / new AP or use "slow" transition if not.  I'm not sure what would cause your problem, but expect that a packet sniff of relevant packets with Wireshark would show where it was failing.

turboproc, 2022-03-28:
Very interesting and much appreciated. Any idea when this work will be pulled into the main repo?

BTW, would be really helpful to be able to find somewhere whether a station supports 11V and/or 11K. Ideally in the GUI but as a first step, in the command line.

atiensivu, 2022-03-31:
> BTW, would be really helpful to be able to find somewhere whether a station supports 11V and/or 11K. Ideally in the GUI but as a first step, in the command line.

That'd be pretty neat I think, as well!

turboproc, 2022-03-31:
So having @Ian-Clowes code running on 4 APs since a number of days. Had it merged into the master branch of OPNWrt. Good news that DAWN eventually start behaving as expected: score really change with the various input parameters (RSSI, util, etc) and stations start to move to better APs.
One issue that shows is that in the hearing map sometimes a station appears with a Freq of 0. Freq == 0 results in a -481 value of channel and that will result in a server error when the hearing map page is rendered.
Question is where this comes from because the AP where this station is connected to perfectly knows the correct Freq. Any idea how to debug ?

Example below

[image]

Of course this error can be avoided by changing %d by %0.f in the template but that's not solving the problem.

Ian-Clowes, 2022-04-02:
> Question is where this comes from because the AP where this station is connected to perfectly knows the correct Freq. Any idea how to debug ?

It's because some data is based on BEACONs, which don't have the frequency.  But they do have the channel numer, which is now converted to frequency to address the bug.

Ian-Clowes, 2022-04-02:
> BTW, would be really helpful to be able to find somewhere whether a station supports 11V and/or 11K. Ideally in the GUI but as a first step, in the command line.

The `rrm_caps` field from `ubus call dawn get_hearing_map` indicates which 802.11k capabilities are reported yt the client.

Ian-Clowes, 2022-04-02:
Finally pushed those updates to my fork: https://github.com/Ian-Clowes/DAWN.

- Updated approach to sending BEACON REQUEST
- Modifications to BEACON and PROBE handling
- Minor changes to kicking algorithm
- Tidy up a few bits of code
- Documentation fixes and updates

I'm still struggling to my enough of my lovcal devices do enough with 802.11k to help devise new kicking algorithms using RCPI (from BEACONs) rather than RSSI (from PROBEs).  Once a few people have the new code working OK I may ask for help gathering in-field examples to better understand what might be possible.

Tagging a few people who I think have been patiently waiting: @turboproc @edrikk @Fail-Safe @imwhocodes @noblemtw

edrikk, 2022-04-03:
Hi there @Ian-Clowes great stuff.

After 24 hours of uptime things are looking good.  The only things I noticed in the logs that I'll share:

- AX3600 Master Kernel 5.15 --- Main router (also carries most of the wifi clients)
     - The only noticeable messages in the log are about "No method found"
    
>Sun Apr  3 12:58:37 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 88:64:40:XX:XX:d7 171 04
Sun Apr  3 12:58:37 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 40:f9:46:XX:XX:da 169 ack=1
Sun Apr  3 12:58:37 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 40:f9:46:XX:XX:da 169 04
Sun Apr  3 12:58:37 2022 daemon.warn dawn: No method found for: beacon-report
Sun Apr  3 12:58:37 2022 daemon.warn dawn: No method found for: beacon-report
Sun Apr  3 12:58:37 2022 daemon.warn dawn: No method found for: beacon-report
Sun Apr  3 12:58:39 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 00:07:a6:XX:XX:c7 167 ack=1
Sun Apr  3 12:58:39 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 00:07:a6:XX:XX:c7 0 04
Sun Apr  3 12:58:39 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 7c: ab:60:XX:XX:6f 173 ack=1
Sun Apr  3 12:58:39 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 7c: ab:60:XX:XX:6f 173 04

- R6700 22.03 Kernel 5.10 --- AP not many devices on it usually
    -  Log flooded with "no station connected" warning...  I don't think this should even be a warming..  maybe debug.

> Sun Apr  3 11:45:25 2022 daemon.warn dawn: No station connected
Sun Apr  3 11:45:35 2022 daemon.warn dawn: No station connected
Sun Apr  3 11:45:35 2022 daemon.warn dawn: No station connected
Sun Apr  3 11:45:45 2022 daemon.warn dawn: No station connected
Sun Apr  3 11:45:45 2022 daemon.warn dawn: No station connected
Sun Apr  3 11:45:55 2022 daemon.warn dawn: No station connected
Sun Apr  3 11:45:55 2022 daemon.warn dawn: No station connected

Unless I've missed it, haven't noticed any lock or out of band errors as yet.  Great job!

Ian-Clowes, 2022-04-04:
> * R6700 22.03 Kernel 5.10 --- AP not many devices on it usually
>       
>       * Log flooded with "no station connected" warning...  I don't think this should even be a warming..  maybe debug.
> 
> 
> > Sun Apr  3 11:45:25 2022 daemon.warn dawn: No station connected

That's very strange - I don't think that's even a message in the current code.  The text and level were changed to this:

`dawnlog_trace("No station connected to %s\n", ifname);`

See lines 183 and 239 in https://github.com/Ian-Clowes/DAWN/blob/2c6a998c9eb70cb5ed26f558da6564f6d7e925a6/src/utils/dawn_iwinfo.c.

Could that device still be running an older version?

edrikk, 2022-04-04:
Hmmm.  This was the only device I didn’t do a full clean on the code for, so it’s quite possible.
Will recompile.

turboproc, 2022-04-04:
Hi @Ian-Clowes just build latest commit and loaded into Netgear X4S R7800. Works all fine (quick check) but noticing a lot of message appearing in the log like below:
```
Mon Apr  4 20:13:43 2022 daemon.warn dawn: MUTEX require = 36694@ubus.c:2034[3069584280l] - appears to be UNLOCKED!
Mon Apr  4 20:13:43 2022 daemon.warn dawn: MUTEX require = 36694@datastorage.c:120[3069584280l] - appears to be UNLOCKED!
Mon Apr  4 20:13:43 2022 daemon.warn dawn: MUTEX require = 36694@ubus.c:2034[3069584280l] - appears to be UNLOCKED!
Mon Apr  4 20:13:43 2022 daemon.warn dawn: MUTEX require = 36694@datastorage.c:120[3069584280l] - appears to be UNLOCKED!
```
Any idea where this might come from?

turboproc, 2022-04-04:
BTW, @Ian-Clowes , when checking the output of ubus call dawn get-network I noticed the following out
```
                                "signature": "wifi4|probe:0,1,45,127,191,255,221(506f9a,22),htcap:096f,htagg:13,htmcs:000000ff,vhtcap:3390f132,vhtrxmcs:0186fffe,vhttxmcs:018
6fffe,extcap:00000a82004000000001|assoc:0,1,33,36,45,48,54,59,70,127,191,221(0050f2,2),221(8cfdf0,1),htcap:096f,htagg:13,htmcs:000000ff,vhtcap:3390f132,vhtrxmcs:0186fffe,vht
txmcs:0186fffe,txpow:1008,extcap:0400080201000040",
                                "ht": true,
                                "vht": true,
                                "rrm-caps": ".TAP..NL",
                                "signal": -68,
                                "rcpi": -1,
                                "rsni": -1
                        },
```
This comes from a Nokia X20 which is very new, supports WiFi (as far as I know). Being new matches with the RRM support but I would expect WiFi5 in the signature rather than WiFi4. Couldn't discover whether the signature is created by DAWN or comes straight from the station

Ian-Clowes, 2022-04-04:
> Hi @Ian-Clowes just build latest commit and loaded into Netgear X4S R7800. Works all fine (quick check) but noticing a lot of message appearing in the log like below:
>
> ```
> Mon Apr  4 20:13:43 2022 [...]

That looks like is the previous set of commits, not the very latest.

See line 2034 from https://github.com/Ian-Clowes/DAWN/blob/92e5327404d963ac75df1c60e6a8d77a56390b76/src/utils/ubus.c,.

Can you double check what is being built?  Do you have a local branch that might still be on the prior code?

Ian-Clowes, 2022-04-04:
> That looks like is the previous set of commits, not the very latest.
>
> See line 2034 from https://github.com/Ian-Clowes/DAWN/blob/92e5327404d963ac75df1c60e6a8d77a56390b76/src/utils/ubus.c,.
>
> Can you [...]

Those signatures are sent by hostapd IIRC.  Whether it constructs them or gets them direct form the device I do not know.

Ian-Clowes, 2022-04-04:
> Sun Apr 3 12:58:37 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 88:64:40:XX:XX:d7 171 04
> Sun Apr 3 12:58:37 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 40:f9:46:XX:XX:da 169 ack=1 [...]

I think that's a consequence of having current and prior DAWN versions running.  I removed the sharing of BEACON REPORTS, so a newer version doesn't know how to handle them.  But if there is a previous version running it will still be sharing them.  Based on your other message where it looks like a build missed pulling the latest code I suspect that may be the reason.

But I've but an "empty handler" in so that the (now unused) messages are handled gracefully if something does send them.

PolynomialDivision, 2022-04-04:
Maybe you want to push your new changes into this repository and we push the changes to openwrt master?

Ian-Clowes, 2022-04-04:
> Maybe you want to push your new changes into this repository and we push the changes to openwrt master?

Yes, once a few people confirm no problems I'll do that.

edrikk, 2022-04-04:
I've recompiled and all is well...  The messages have all disappeared.
Clearly a bad build that didn't pickup the updated code.
I would personally say this (maybe with the inclusion of "empty handler" and it's ready for 22.03 and master at the very least).
I've been running the code for well over a month now without issue.  :)

Ian-Clowes, 2022-04-05:
@edrikk, it's good that it does no harm 😄.  Do you also see the movement of clients between APs that you'd like?  That is the positive test that it is worth pushing.

edrikk, 2022-04-05:
@Ian-Clowes 
My household is mostly iOS devices, and as far as I've been able to tell, yes they move as they should/I expect.  I've been using the standard "1" kicking method.

I don't often / if ever go into the hearing map, but noticed this today half way down the list.  The MAC address was for one of my APs:

`0.000 GHz Channel:
500 Internal Server Error

Sorry, the server encountered an unexpected error.

/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template '[string]'.
A runtime error occurred: [string "[string]"]:16: bad argument #2 to '?' (integer expected, got number)
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/util.lua:49: in function 
	[string "[string]"]:16: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	...usr/lib/lua/luci/model/cbi/dawn/dawn_hearing_map.lua:8: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:1478: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:1051: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function `

turboproc, 2022-04-06:
> > Couldn't discover whether the signature is created by DAWN or comes straight from the station
> 
> Those signatures are sent by hostapd IIRC. Whether it constructs them or gets them direct form the device I do not know.

@Ian-Clowes, Indeed, checked hostapd source and the signature is created in src/ap/taxonomy.c . The latest version only mentions wifi4, no futher evaluation to wifi5 or wifi6. Nothing we should change from a Dawn pov.

noblemtw, 2022-04-13:
> DAWN doesn't really know whether a client supports 802.11v. It will send transition requests to all clients when it thinks it might be useful. They may or may not understand and act upon it.

Clients that support 802.11v apparently provide the information in the association request according to this [article](https://semfionetworks.com/blog/how-to-check-if-a-client-device-supports-80211v/) I've just stumbled across. I assume dawn would need to get the info via hostapd and I've no idea whether it provides it, but would be a really nice feature to use soft kicking where supported and fall back to a disassoc for legacy clients

My TV seems to be pretty brain dead and is stuck on the wrong AP at the moment, with the new kicking algorithm, dawn does ask it nicely to move, but nothing happens (I assume because it lacks 802.11v). With the old algorithm, no other APs are reported (either because the TV doesn't support 802.11k or I've configured something wrong), so dawn doesn't even attempt to steer it elsewhere...

Other than that, everything is looking good with the latest code, although I've only been running it a few hours, typically after waiting ages for the latest PR to be pushed, it happened while I was on holiday :-)

Ian-Clowes, 2022-04-20:
Some work to improve hostapd integration for DAWN and other potential users would be useful.  I haven't had time to look at it, but may start with a simple "raw" message sharing in addition to hostapd's "formatted" ones so that DAWN can pick what it wants out of the bits.

But you also describe the key challenge that may make that a poor use of developer and runtime cycles.  Some devices just don't share enough info, so no matter how "smart" we make DAWN they can't be managed very well.  The flip side of that being that the simple approach (prob even simpler than DAWN is already) can do a lot of useful control of "willing" clients.

edrikk, 2022-05-07:
Updating that as far as I can tell all is working well.
Maybe good to PR and get this out there for further testing?  A LOT has been fixed here already...

edrikk, 2022-06-04:
@Ian-Clowes U hope you’re ok, and that things are just busy with life.

@PolynomialDivision wondering if the changes can be cherry-picked?

Running without issue for months, and all the unlocked issues in Prod version are resolved.

PolynomialDivision, 2022-06-04:
> @PolynomialDivision wondering if the changes can be cherry-picked?

Yeah of course we can cherry-pick them. You mena just pick latest changes from the master? Would be good to get some ACK from @Ian-Clowes before doing so.

PolynomialDivision, 2022-06-04:
> But you also describe the key challenge that may make that a poor use of developer and runtime cycles. Some devices just don't share enough info, so no matter how "smart" we make DAWN they can't be [...]

I have the impression when I started with DAWN, nothing was working in the area of 802.11v/k. Now after doing all the stuff, client side implementations starting to get better. You also have to think into the future, what may become possible.
Maybe @blocktrron can also share his impression about the 802.11k/v situation.

edrikk, 2022-06-04:
> > @PolynomialDivision wondering if the changes can be cherry-picked?
> 
> Yeah of course we can cherry-pick them. You mean just pick latest changes from the master? Would be good to get some ACK from @Ian-Clowes before doing so.

Yes exactly.  Everything after what in your master (Jan 17 - a7267406).  It has been tested for months, and is working without issues as far as I have been able to tell across four APs.

I even merged your latest commit from your branch when it was first added, and the two have been humming along since.

PolynomialDivision, 2022-06-11:
@edrikk Thanks for testing. I pushed now to OpenWrt Master:
https://github.com/openwrt/packages/commit/0c74ab9a790c81ff57ad1d7741489bf5e1839839

edrikk, 2022-06-13:
Hi @PolynomialDivision 

Just FYI that I'm getting the below spammed into my system log with current master + DAWN.
Had not seen these with the @Ian-Clowes code.  The only difference Is [this commit](https://github.com/berlin-open-wireless-lab/DAWN/commit/850a75c18293fef2586513704142728fa0eba712) which does have a change to dawn_memory_unregister.

> Sun Jun 12 23:11:46 2022 daemon.warn dawn: mem-audit: Releasing memory we hadn't registered (tcpsocket.c:142)...
Sun Jun 12 23:11:46 2022 daemon.warn dawn: mem-audit: attempted to register memory already registered (M@tcpsocket.c:114)...
Sun Jun 12 23:11:46 2022 daemon.warn dawn: mem-audit: Releasing memory we hadn't registered (tcpsocket.c:142)...
Sun Jun 12 23:11:46 2022 daemon.warn dawn: mem-audit: attempted to register memory already registered (M@tcpsocket.c:114)...
Sun Jun 12 23:11:46 2022 daemon.warn dawn: mem-audit: Releasing memory we hadn't registered (tcpsocket.c:142)...

PolynomialDivision, 2022-06-13:
@edrikk You are right, this commit introduces a bug. The ret is a newly allocated pointer, and we don't want to "unregister". @neheb wanted to fix gcc12 compilation. I guess there was a bug, that ptr was already freed and was used afterwards to unregister from memory allocation.

PolynomialDivision, 2022-06-13:
> that's the gcc warning, yeah.

Any recommendation how to fix it in a good way?

neheb, 2022-06-13:
get rid of

```
       if (ret != NULL)
          dawn_memory_unregister(DAWN_REALLOC, file, line, ret);
```

PolynomialDivision, 2022-06-13:
@edrikk, @neheb Shouldn't this fix the issue?
https://github.com/berlin-open-wireless-lab/DAWN/pull/175/files

edrikk, 2022-06-14:
Perfect!  Did a compile and indeed the messages are gone.  Well done!

[8 more replies]

closed 2022-06-16 by PolynomialDivision