berlin-open-wireless-lab/DAWN #164: Question: How to remove the dependence from UMDNS ?

lars18th, 2022-03-08:
Hi,

I'm using DAWN in openwrt succesfully with UDP broadcast connections over the broadcast addess of my IPv4 LAN. So I have configured:
```
    option broadcast_ip         '192.168.1.255'
    option broadcast_port       '1025'
    option tcp_port             '1026'
    option network_option       '0' # 0 udp broadcast, 1 multicast, 2 tcp
    option shared_key           'MYsharedDkey'
```

In this scenario I feel the mDNS service it's irrelevant, as the communication is stablished between the APs using the LAN wired network. So the question is what I need to do for removing the dependency with the UMDNS package.

Any help please?
Thank you.

Ian-Clowes, 2022-03-27:
There's no way to leave it out of the build at the moment, but it wouldn't be too hard to add a #define to remove the no longer needed parts of the code.

It wouldn't change things like size of the binary much though.  ubus is used to "abstract" the uMDNS behaviour.

Did you have some other aim?

lars18th, 2022-03-28:
Hi @Ian-Clowes ,

Thank you for the answer. My concern is not about the size of the binary. The problem is that the mDNS generates some troubles even you don't need it. For example, the start script assumes that this process is installed and running. But I don't want to waste memory and CPU for this irrelevant and not necessary service. Futhermore, sometimes the DAWN process crashes and I feel is related to a missing mDNS functions. So, my suggestion is to provide the option to work without it. Some users (like me) want to use DAWN on two or three AP in the home to enable a mesh wifi. This works with minimal configuration and without the pain of the mDNS. I understand that in a wide-radio-LAN the mDNS service is mandatory. But the current DAWN package could be used in small environments as well, and for this reason the dependency with the mDNS could be (optionally) removed.

I hope you could want to work on remove this dependency.
Thank you for you support.
Regards.

Ian-Clowes, 2022-03-28:
I think there are two things to seperate on uMDNS.

As far as I know uMDNS is not installed by DAWN.  If you execute `ubus call umdns browse` on an OpenWrt device you'll see a lot of services using it., So DAWN not using it will not generally chnage its presence.  Are you saying that you've emilinated all other uses of it?  If not then it will not be "irrelevant and unnecessary" to the overall system.

I suspect the startup script could inspect the networking parameters before making use of uMNDS.  I've never looked at that, but maybe you can see how feasible that would be?  I think the DAWN code ignores uMDNS related behaviours if not using TCP, and maybe that is another thing you can check.  Do you see entries for DAWN in that `browse` list?

Finally, and perhaps more significant, your crashes may be related to not using TCP.  There was a process hang (so not formally a crash) in some network handling that was fixed in the TCP code, but may not have been in other network mechanisms.  If you turn the loglevel up to 3 and use `logread -f`|grep dawn` to keep an eye on things then if / when the output stops the last few function entry debug lines should help confirm what is happening.

lars18th, 2022-03-28:
Hi @Ian-Clowes ,

1. Related to the uMDNS, I don't use at all on my openwrt APs: `root@OWRT:~# ubus call umdns browse` --> `Command failed: Not found`. And you're right: the dawn package doesn't install the umdns package. However, the dependency exists. And for this reason I ask if you could support to run without it.

2. Time ago I've detected some trouble in the start script of the uMDNS because the lack of the `seccomp` tool. Without the seccomp (that is not necessary to run the umdns package as well, but the configuration assumes it when starting the umdns daemon) the DAWN can't be started. And all because the dependency of dawn --> umdns --> seccomp. IMHO I feel is preferable to not enforce these dependencies. If they're really needed, then provide a check for the users that require them. But in any other case, please enable the option to run without them. You agree?

3. I'm not aware that the hang is because the UDP communication. I prefer to use it for this reason: inside a LAN the most simple and fast _broadcast_ communication model is the IPv4 broadcast domain. However, with such communication model only datagrams can be exchanged. Therefore, I prefer to use UDP over TCP, as the second implies to stablish a TCP socket between all nodes ((n-1)! links are a lot of sockets). And this is be far a lot more complex than simple broadcast messages.

So, regarding the "hanging" of the dawn process: you think that the UDP control plane could be enhanced? You'll work on fix this bug? It could help if I enable the loglevel and I share the info? If the answers are YES, then I'll collaborate with this information.

Regards.

Ian-Clowes, 2022-03-28:
Using your numbers...

1. OK, that does support the broader goal to reduce the DAWN dependency.  I didn't want you working to false assumptions about DAWN being the reason it was installed though.
2. Untangling OpenWrt optional startup and install dependencies is probably beyond me.  I agree it could be useful to have an alternate install mechanism for not requiring uMDNS, but don't think I can dreate it.  Getting to a point where a forced install of the DAWN pakage via opkg telling it to ignore dependencies would be workable would be as good as I'd expect in the short term, although I probably won't have any time to test that.
3. So the fix may have had wider impact than I understood if UDP appears to work reliably.  I agree UDP should work and wasn't trying to steer you to TCP.  I just wanted you to be aware it is a less tested path.

By all means turn up the logging and if you see a halt or "tight loop" behaviour post the relevant looking messages so we can figure out possile cause / resolution.

lars18th, 2022-03-28:
Hi @Ian-Clowes ,

So I'll do more tests with my UDP broadcast configuration adding more LOGs. At time, I'm rebooting the AP's every day, but sometimes I need to restart the dawn service manually because some radios are not correctly working.

And regarding the dependency with the uMDNS, I want to be clear: The dawn openwrt package could be installed without it. The problem is that internally you're assuming that the mdns service is installed and up (running). So perhaps you need only to assume that it could not be the case, and with simple workaround or switches you can solve the dependency.

Thank you for your support.

PolynomialDivision, 2022-03-28:
Without reading the whole thread, we can just modify the [Makefile](https://github.com/openwrt/packages/blob/master/net/dawn/Makefile) to add a package option/selection to seperate in umdns enabled dawn.