berlin-open-wireless-lab/DAWN #166: 802.11AX HE MCS in scoring

andyleadbetter, 2022-04-04:
I have 2 802.11AX routers with a single SSID configured to broadcast a single SSID in dual 2.4/5G wifi on channels   100/136 for 5G and 1/6 for 2.4G

A wifi 6 station connects using 802.11AX protocols, and the wifi dashboard says it has  the following properties on it's link
680.6 Mbit/s, 80 MHz, HE-MCS 7, HE-NSS 2, HE-GI 1
816.7 Mbit/s, 80 MHz, HE-MCS 8, HE-NSS 2, HE-GI 1

30:AB:6A:78:83:69 | E8:9F:80:D5:D4:C5 | 5.500 GHz Channel: 100 | True | True | -86 | -1 | -1 | 10.59 % | 10 | 193
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  | E8:9F:80:5D:2F:18 | 5.680 GHz Channel: 136 | False | False | -51 | 120 | 90 | 4.31 % | 2 | 193
  | E8:9F:80:5D:2F:17 | 2.437 GHz Channel: 6 | False | False | -50 | -1 | -1 | 10.20 % | 7 | 150
  | E8:9F:80:D5:D4:C4 | 2.412 GHz Channel: 1 | True | False | -80 | -1 | -1 | 4.71 % | 5 | 60

The station should be favouring massively channel 136 based on RSSI, but the VHT false seems to skew the numbers.  

What is the best config for a dual band network with 2 backhaul wired connected routers? 

```
config network
    option broadcast_ip         '10.0.0.255'
    option broadcast_port       '1025'
    option server_ip            ''
    option tcp_port             '1026'
    option network_option       '2' # 0 udp broadcast, 1 multicast, 2 tcp
    option shared_key           'Niiiiiiiiiiiiick'
    option iv                   'Niiiiiiiiiiiiick'
    option use_symm_enc         '0'
    option collision_domain     '-1'     # enter here aps which are in the same collision domain
    option bandwidth            '-1'     # enter network bandwidth

config ordering
    option sort_order           'cbfs'

config hostapd
    option hostapd_dir          '/var/run/hostapd'

config times
    option update_client         '10'
    option denied_req_threshold  '30'
    option remove_client         '15'
    option remove_probe          '30'
    option remove_ap             '460'
    option update_hostapd        '10'
    option update_tcp_con        '10'
    option update_chan_util      '5'
    option update_beacon_reports '600'

config metric global
    option rssi_weight          '0'
    option rssi_center          '0'
    option initial_score        '0'
    option kicking_threshold    '20'
    option duration             '600'
    option rrm_mode             'apt'
    option ap_weight            '0'
    option ht_support           '50'
    option vht_support          '100'
    option no_ht_support        '-10'
    option no_vht_support       '-20'
    option rssi                 '30'
    option low_rssi             '-50'
    option freq                 '50'
    option chan_util            '0'
    option max_chan_util        '0'
    option rssi_val             '-60'
    option low_rssi_val         '-70'
    option chan_util_val        '0'
    option max_chan_util_val    '0'
    option min_probe_count      '0'
    option bandwidth_threshold  '6'
    option use_station_count    '0'
    option max_station_diff     '0'
    option eval_probe_req       '0'
    option eval_auth_req        '0'
    option eval_assoc_req       '0'
    option kicking              '2'
    option deny_auth_reason     '1'
    option deny_assoc_reason    '17'
    option use_driver_recog     '1'
    option chan_util_avg_period '3'
    option set_hostapd_nr       '1'

config metric 802_11a
    option rssi_weight          '2'
    option rssi_center          '-70'
    option initial_score        '125'

config metric 802_11g
    option rssi_weight          '2'
    option rssi_center          '-70'
    option initial_score        '100'
```

Ian-Clowes, 2022-04-05:
This is an unfortunate side effect of the `/etc/dawn/config` in the installation package being out of date with how things function, and / or you being a long-term DAWN user with a config file that needs a refresh.  I'll work with @PolynomialDivision to get a new baseline config in the next PR.

Many of the options you have under global have evolved to be "per band".  The effect is that they are ignored in the global section, and defaults are used for missing values in the per-band sections.  It is the defaults that are skewing your scores, as both forms of scoring are "competing".

See https://github.com/Ian-Clowes/DAWN/blob/master/CONFIGURE.md for more details, and for help to move the options to the right section so you can see what is happening.

The quick fix is probaly to add these to each of the `802_11a` and `802_11g` sections:
```
    option rssi          '0'
    option low_rssi    '0'
```

turboproc, 2022-04-09:
@Ian-Clowes , might be wrong but raising the question anyway ;-). When supporting WiFi6 would we require an EH configuration next to the current HT and VHT configuration elements ?