berlin-open-wireless-lab/DAWN #167: Some suggestions about logs

McGiverGim, 2022-04-12:
Two things about logs. Usually, TRACE is the lowest level of a log. I think TRACE must be 0 and DEBUG 1.
https://github.com/berlin-open-wireless-lab/DAWN/blob/7a72674063dd34bd21f39c7be8aadb867db67658/src/include/utils.h#L26-L31

Another thing: can you change this log level to DEBUG or TRACE:
https://github.com/berlin-open-wireless-lab/DAWN/blob/7a72674063dd34bd21f39c7be8aadb867db67658/src/utils/dawn_iwinfo.c#L185

This is making my OpenWrt full of "No station connected" messages. 

Thanks!

Ian-Clowes, 2022-04-20:
The levels were originally "invented" independently.  I later realised that hostapd has a similar set, so might standardise on the same at some point.

I think the specific message that you refeerence has already been adjusted in latest code as others commented on it as well.

McGiverGim, 2022-04-20:
> I think the specific message that you refeerence has already been adjusted in latest code as others commented on it as well.

I looked at the "master" version of the code and I see it as "warning". The latest code is at a different branch than master?

Regards!

Ian-Clowes, 2022-04-21:
This will probably get to master once I sort out a PR: https://github.com/Ian-Clowes/DAWN

McGiverGim, 2022-05-11:
@Ian-Clowes 
> I think the specific message that you refeerence has already been adjusted in latest code as others commented on it as well.

The dawn package has been updated in openwrt and the problem with the log message has not been fixed... :(

axxelh, 2022-05-26:
> This will probably get to master once I sort out a PR: https://github.com/Ian-Clowes/DAWN

Just to confirm the original report, the specific message is still present at warning level in the forked repo and master in get_expected_throughput() and get_rssi() in dawn_iwinfo.c. 

In both cases it appears when there are no stations associated with a particular ifname and kicking is enabled. That doesn't seem like a warning condition, as it can happen when all clients move to a single AP. It also appears (as in my case) where there are unused test networks.

In any case it's a lot of logging and seems appropriate for TRACE level.