berlin-open-wireless-lab/DAWN #170: dawn seems to freeze

ff-mgk, 2022-05-14:
Hello, i hope I haven't overlooked an present issue.

I am using 6 dump openwrt-APs, each with 3 uniformly SSIDs.
1 SSID ist open (Freifunk), 2 SSIDs are WPA2-PSK+802.11r.
All 3 SSIDs appear in dawn.
I wonder why, because an open network does not correspond to 802.*, but ok.
I am Using 2.4 + 5 Ghz
--> there are 2*6*3=36 wifi-networks

2 APs are
Xiaomi Redmi Router AX6S aka AX3200
OpenWrt SNAPSHOT r19592-4509b790f0 / LuCI Master git-22.119.37097-531c0ec
dawn 2022-01-17-7a726740-1
luci-app-dawn git-21.282.29102-4a78122

4 APs are
Xiaomi Mi Router 4A Gigabit Edition
OpenWrt 21.02.3 r16554-1d4dea6d4f / LuCI openwrt-21.02 branch git-22.083.69138-0a0ce2a
dawn 2022-01-17-7a726740-1
luci-app-dawn
git-21.299.23946-ae446ce

modified Part of /etc/config/dawn
```
config network
    option broadcast_ip         '172.27.255.255'
    option broadcast_port       '1025'
    option server_ip            ''
    option tcp_port             '1026'
    option network_option       '0' # 0 udp broadcast, 1 multicast, 2 tcp
```
```
root@OpenWrtMi4AGKueche:~# ubus call dawn get_network
Command failed: Request timed out
```
```
Mem: 76412K used, 45808K free, 17044K shrd, 2396K buff, 27392K cached
CPU:  49% usr   0% sys   0% nic  49% idle   0% io   0% irq   0% sirq
Load average: 2.30 2.18 2.18 5/96 9703
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 9055     1 root     R     2984   2%  50% /usr/sbin/dawn
```
```
root@OpenWrtAX3200OG:~# ubus call dawn get_network
Command failed: Request timed out
```
```
Mem: 115636K used, 128516K free, 16476K shrd, 0K buff, 32668K cached
CPU: 100% usr   0% sys   0% nic   0% idle   0% io   0% irq   0% sirq
Load average: 2.40 2.42 2.21 4/101 5380
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 5302     1 root     R     3140   1% 100% /usr/sbin/dawn
```
root@OpenWrtMi4AGKueche:~# logread | grep dawn
```
Fri May 13 23:19:45 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:19:45 2022 daemon.warn dawn[9055]: Failed to update rssi!
Fri May 13 23:19:45 2022 daemon.warn dawn[9055]: Current AP not found in probe array!
Fri May 13 23:19:45 2022 daemon.warn dawn[9055]: No station connected
...
Fri May 13 23:31:39 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:31:39 2022 daemon.warn dawn[9055]: mem-audit: Releasing memory we hadn't registered (msghandler.c:306)...
Fri May 13 23:31:49 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:31:49 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:31:54 2022 daemon.warn dawn[9055]: mem-audit: attempted to register memory already registered (M@msghandler.c:154)...
Fri May 13 23:31:59 2022 daemon.warn dawn[9055]: No station connected
...
Fri May 13 23:43:12 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:22 2022 daemon.warn dawn[9055]: mem-audit: Releasing memory we hadn't registered (datastorage.c:1324)...
Fri May 13 23:43:22 2022 daemon.warn dawn[9055]: mem-audit: Releasing memory we hadn't registered (datastorage.c:1324)...
Fri May 13 23:43:22 2022 daemon.warn dawn[9055]: mem-audit: attempted to register memory already registered (M@msghandler.c:503)...
Fri May 13 23:43:22 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:22 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:32 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:32 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:42 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:42 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:52 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:52 2022 daemon.warn dawn[9055]: No station connected
Fri May 13 23:43:52 2022 daemon.warn dawn[9055]: mem-audit: attempted to register memory already registered (X@ubus.c:855)...
Sat May 14 04:26:33 2022 user.info : luci: accepted login on /admin/dawn/view_network for root from 172.27.163.135
```
hint; the last Line does not repeat, the log stops

Since now i was too slow to catch the beginning...

Regards

kmaras77, 2022-05-14:
I have command in crown:
*/5 * * * *  ubus call dawn get_network || service dawn restart

ff-mgk, 2022-06-07:
*/5 * * * * ubus call dawn get_network || /etc/init.d/dawn restart

is more comfortable for use with cron

PolynomialDivision, 2022-06-16:
Can you update on latest dawn version and test again?