berlin-open-wireless-lab/DAWN #173: log files flooded : daemon.warn dawn[2028]: No station connected

kaikus, 2022-06-09:
even with 

config local
    option loglevel             '0'

the log file is flooded every 10 seconds with 

Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:15 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected
Thu Jun  9 12:14:25 2022 daemon.warn dawn[2028]: No station connected

palfal, 2022-06-12:
same is here

PolynomialDivision, 2022-06-16:
@kaikus You set the debug-level to the lowest. Just set it higher?

kaikus, 2022-06-16:
What does this mean ? Higher level, less entries in log ?

kaikus, 2022-06-18:
So it seems this documentation needs to be updated ?

https://github.com/Ian-Clowes/DAWN/blob/master/CONFIGURE.md

PolynomialDivision, 2022-06-18:
> So it seems this documentation needs to be updated ?
> 
> https://github.com/Ian-Clowes/DAWN/blob/master/CONFIGURE.md

Yep. Thanks for reporting.

PolynomialDivision, 2022-07-04:
Done

closed 2022-07-04 by PolynomialDivision