berlin-open-wireless-lab/DAWN #176: Making DAWN mDNS provider agnostic - feature request

perceival, 2022-06-20:
For services I'm using in my network I need mDNS to work across VLANs. Based on my understanding:
- such functionality is provided by avahi but not by umDNS
- avahi is not working together with umDNS (?)
If above is correct is it possible for DAWN to work with avahi as well as with umDNS?

perceival, 2022-09-11:
Indeed seems umdns does not like to be mixed with avahi. On the other hand samba4 does require avahi support. With DAWN bound with umdns it creates catch 22 scenario.

kirelagin, 2022-10-01:
In my experience, even simple name resolution does not work reliably with umdns, so I had to go back to Avahi on my OpenWrt routers. Hence, yeah, having Avahi support would be great.

lzlrd, 2025-03-18:
+1 on the above, until Umdns introduces reflection then many users are stuck with Avahi.

RobotRoss, 2025-06-05:
+1 to the above. ".local" name resolution does not work with umdns , which I use extensively in my networks.