berlin-open-wireless-lab/DAWN #188: umdns report duplicate ipv4

ptpt52, 2022-07-21:
as reported here https://github.com/openwrt/openwrt/issues/10316

the umdns report blobmsg may contain duplicate 'ipv4' key, this may cause dawn connect to a wrong ip

sample:
```
{
	"_dawn._tcp": {
		"MESH-AGENT_E4671E37EC8B": {
			"ipv4": "192.168.16.1",
			"ipv6": "fda8:42e8:8129::1",
			"ipv6": "fe80::e667:1eff:fe37:ec8b",
			"ipv4": "192.168.15.141",
			"port": 1026
		}
	},
```

dawn may try to connect to 192.168.16.1 but the real target is 192.168.15.141
so it fail to connect to real AP

ptpt52, 2022-07-21:
I am trying to check dup-ipv4 in dawn and then trigger a umdns reload to clear the duplicate 'ipv4' key

see https://github.com/ptpt52/dawn/commit/ec986a04ca1dcddd8b2db5eeed2907b6715225cb

PolynomialDivision, 2022-07-21:
This issue also prevents me from adding ipv6 support to dawn.