berlin-open-wireless-lab/DAWN #193: DAWN Log Errors with July 23 Release:  client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578

edrikk, 2022-07-23:
Have closed issue [190 - [With 0721 release Network Overview showing only own Hostname](https://github.com/berlin-open-wireless-lab/DAWN/issues/190) as the issue was resolved.

Opening new (although potentially related) issue, as I'm noting below log message in all my APs (Xiaomi AX6S running master head) as well as main router+AP (AX3600 running self compiled master head + Robimarko merge).

**AX3600 Example:**

> Sat Jul 23 11:06:44 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:06:44 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:06:45 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:06:45 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:06:47 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 126 ack=1
> Sat Jul 23 11:06:47 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:06:48 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:06:48 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:06:50 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 127 ack=1
> Sat Jul 23 11:06:50 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:06:53 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 128 ack=1
> Sat Jul 23 11:06:53 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:06:54 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:06:54 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:06:55 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:06:55 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:06:56 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 129 ack=1
> Sat Jul 23 11:06:56 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:06:58 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:06:58 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:06:59 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 130 ack=1
> Sat Jul 23 11:06:59 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:07:00 2022 kern.info kernel: [  543.098343] device br-lan entered promiscuous mode
> Sat Jul 23 11:07:02 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 131 ack=1
> Sat Jul 23 11:07:02 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:07:04 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:07:04 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:07:05 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:07:05 2022 daemon.warn dawn: Connection to server closed
> Sat Jul 23 11:07:05 2022 kern.info kernel: [  547.970173] device br-lan left promiscuous mode
> Sat Jul 23 11:07:07 2022 daemon.notice hostapd: wlan2: BEACON-REQ-TX-STATUS 18:48:ca:a1:XX:YY 132 ack=1
> Sat Jul 23 11:07:07 2022 daemon.notice hostapd: wlan2: BEACON-RESP-RX 18:48:ca:a1:XX:YY 0 04
> Sat Jul 23 11:07:08 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 0, total: 33816578
> Sat Jul 23 11:07:08 2022 daemon.warn dawn: Connection to server closed

**One of the AX6S Dumb APs as example:**

> Sat Jul 23 10:57:29 2022 daemon.err dawn: client_to_server_state()=tcpsocket.c@106 eof!, pending: 655729, total: 33816578
> Sat Jul 23 10:57:34 2022 daemon.err dawn: send_tcp()=tcpsocket.c@460 Ustream error(460)!

PolynomialDivision, 2022-07-24:
Can you test latest master again?

edrikk, 2022-07-24:
For the APs I'll have to wait until DAWN package compiles/is ready for attendedsysupgrade.
But for the AX3600 I just compiled from source, and it's looking good.

I notice a couple of messages (not flooding like before), once of which is new.  But this could potentially be because the APs are running the 0722 version still potentially I would guess....

> 
> Sun Jul 24 13:07:11 2022 daemon.warn dawn: Client / BSSID = 88:64:40:XX:YY:ZZ / 88:C3:97:C2:AA:BB: BEACON REQUEST failed
> Sun Jul 24 13:07:11 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 40:f9:46:5a:SS:TT 231 ack=1
> Sun Jul 24 13:07:11 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 40:f9:46:5a:SS:TT 231 04
> Sun Jul 24 13:07:14 2022 daemon.notice hostapd: Beacon request: 88:64:40:XX:YY:ZZ is not connected
> Sun Jul 24 13:07:14 2022 daemon.warn dawn: Client / BSSID = 88:64:40:XX:YY:ZZ / 88:C3:97:C2:CC:DD: BEACON REQUEST failed
> Sun Jul 24 13:07:14 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 40:f9:46:5a:SS:TT 232 ack=1
> Sun Jul 24 13:07:14 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 40:f9:46:5a:SS:TT 232 04
> Sun Jul 24 13:07:17 2022 daemon.notice hostapd: Beacon request: 88:64:40:XX:YY:ZZ is not connected
> Sun Jul 24 13:07:17 2022 daemon.warn dawn: Client / BSSID = 88:64:40:XX:YY:ZZ / 8C:DE:F9:BFEE1:FF: BEACON REQUEST failed
> Sun Jul 24 13:07:17 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 40:f9:46:5a:SS:TT 233 ack=1
> Sun Jul 24 13:07:17 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 40:f9:46:5a:SS:TT 233 04
> Sun Jul 24 13:07:19 2022 daemon.notice hostapd: Beacon request: 88:64:40:XX:YY:ZZ is not connected
> Sun Jul 24 13:07:19 2022 daemon.warn dawn: Client / BSSID = 88:64:40:XX:YY:ZZ / 8C:DE:F9:BF:GG:HH: BEACON REQUEST failed

And a few of these warnings:

> Sun Jul 24 12:59:50 2022 daemon.info hostapd: wlan1: STA 7c:2e:bd:e4:AA:BB WPA: pairwise key handshake completed (RSN)
> Sun Jul 24 12:59:50 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 7c:2e:bd:e4:AA:BB
> Sun Jul 24 12:59:50 2022 daemon.warn dawn: Connection closed
> ...
> ...
> Sun Jul 24 13:04:22 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 40:f9:46:5a:EE:FF 174 04
> Sun Jul 24 13:04:25 2022 daemon.notice hostapd: wlan1: BEACON-REQ-TX-STATUS 40:f9:46:5a:EE:FF 175 ack=1
> Sun Jul 24 13:04:25 2022 daemon.notice hostapd: wlan1: BEACON-RESP-RX 40:f9:46:5a:EE:FF 175 04
> Sun Jul 24 13:04:31 2022 daemon.warn dawn: Connection closed

Will check once the APs get the update to see if anything changes

edrikk, 2022-07-25:
> Those beacon request messages are normal. No idea why hostapd is so talky about them.

yes, but I was more pointing out the DAWN ones.  Left the hostapd ones in for context in case it helps a little.

If the DAWN connection closed one’s are normal, maybe they should be changed to info instead of warning?

edrikk, 2022-07-28:
@PolynomialDivision 
I've updated router + all 3 APs, and after 3 days, I've only noticed 2-3 of the "daemon.warn dawn: Connection closed" warnings.

Not sure whether that's an important event (e.g. warning, something bad happened you should be aware of), or if it's just a message whose error level is higher than it should be...

Let me know if you'd like me to just close this issue and will do so. 
Thanks!

PolynomialDivision, 2022-07-29:
> Not sure whether that's an important event (e.g. warning, something bad happened you should be aware of), or if it's just a message whose error level is higher than it should be...

Depends. Actually the connection should not be closed in normal operation.

Are you seeing also something like this?
https://forum.openwrt.org/t/dawn-a-decentralized-wireless-controller/108768/171

edrikk, 2022-07-30:
I’m not sure..  I just installed statistics on one of the APs.  Will let it run for a while to see…

edrikk, 2022-08-01:
[Not seeing the same in my setup.](https://forum.openwrt.org/uploads/default/optimized/3X/0/e/0eec5c34c2301586ee2c1f2b6e7c715a3a60bf76_2_690x371.jpeg)

closed 2022-08-01 by edrikk

PolynomialDivision, 2022-08-01:
Thanks for testing!