berlin-open-wireless-lab/DAWN #196: MAC Address Ignore List

thecodeprofessor, 2022-08-19:
DAWN is working perfectly with my OpenWrt access points with one exception. It keeps disconnecting my Tuya IOT devices because of little or no data transfer.

Fri Aug 19 19:01:32 2022 daemon.info dawn: Client 38:1F:8D:00:00:00: Kicking due to low active data transfer: RX rate 1.000000 below 6 limit

I have tried setting option bandwidth_threshold to '0' but that does not prevent the kick, it only ignores the bandwidth check.

Is there already a way to create a list of MAC addresses that are never kicked, or have DAWN ignore all clients connected to a specific SSID?

StarSkyZheng, 2022-08-29:
+1
My android TV has a terrible wireless card, and located between two DAWN APs. So TV always been kick between these two APs.
So i hope to add this TV to a list which not managed by DAWN.

Flole998, 2022-10-29:
> Client 38:1F:8D:00:00:00: Kicking due to low active data transfer: RX rate 1.000000 below 6 limit

I am seeing this aswell for some IoT-Clients. However, I haven't figured out how to enable "legacy device kicking" so nothing is happening apparently. I would expect that the same thing that happens when I click "disconnect" in LUCI for a client is also happening if a client didn't react to a kick for x times.

PolynomialDivision, 2022-11-01:
This is now controlled by the disassociation_imminent"` in the `bss_transition_request` method. So a `wnm_disassoc_imminent` is a kick, and `bss_transition_request` a request without kicking the client at the end.

felipecrs, 2024-12-28:
From what I can tell, even though this log message is displayed, the device is not actually being kicked. I can confirm that by pinging the device, and noticing it never stops replying even after the log message.