berlin-open-wireless-lab/DAWN #197: Excluding interfaces

DavidB137, 2022-09-30:
Is it possible to exclude some wireless interfaces (SSIDs) from any Dawn handling? E.g. I have one SSID for people and another for IoT devices, but I only want Dawn to handle kicking for first SSID.

backslashxx, 2022-12-02:
I second this, Im having issues with iot and sometimes guest clients being kicked.

SirLouen, 2023-03-02:
> Is it possible to exclude some wireless interfaces (SSIDs) from any Dawn handling? E.g. I have one SSID for people and another for IoT devices, but I only want Dawn to handle kicking for first SSID.

I have DAWN configure wifi-iface level not wifi-device level.

jokogr, 2023-03-19:
@SirLouen could you share your configuration or point at the parameters you have used? As far as I can understand from [current CONFIGURE.md](https://github.com/berlin-open-wireless-lab/DAWN/blob/9885d77a1942f3ed25f8c5a2d0ad07fcbcc29380/CONFIGURE.md), there are only global and band metrics to tweak, not interface ones.

SirLouen, 2023-03-19:
> @SirLouen could you share your configuration or point at the parameters you have used? As far as I can understand from [current [...]

https://openwrt.org/docs/guide-user/network/wifi/dawn#configure_80211k_and_80211v_on_all_ap-nodes

jokogr, 2023-03-19:
Ah, I see. I have these settings, but apparently the clients I am interested at (Shelly 2.5 devices) support neither 802.11k, nor 802.11v.

SirLouen, 2023-03-19:
> Ah, I see. I have these settings, but apparently the clients I am interested at (Shelly 2.5 devices) support neither 802.11k, nor 802.11v.

I have like 50 shelly connected to my dawn network without issues. Where is your problem?
Also consider using simoultaneously 
https://github.com/berlin-open-wireless-lab/DAWN/issues/210#issuecomment-1452493565

jokogr, 2023-03-19:
I was under the impression that a single Shelly device was getting kicked frequently, but apparently it's the device's fault, I can now see the device steadily connected to a single AP, I just can't ping it. Sorry for the noise.

KosmiK2001, 2023-07-22:
IMHO, it seems to me that the exclusion of certain ssids is necessary. Among the clients, problematic devices are possible, for example, usb-wifi sticks, which can lower the signal strength or, due to some saving modes in the OS, reduce audibility. This results in a kick. If it were possible to exclude ssid's - that would solve such problems.

SirLouen, 2023-07-22:
> IMHO, it seems to me that the exclusion of certain ssids is necessary. Among the clients, problematic devices are possible, for example, usb-wifi sticks, which can lower the signal strength or, due [...]

Generally the philosophy of software must be: do only one thing very well.

There are other modules that do this, why you don't use them instead of requesting a Mesh system to do such?

Just go and install any of such modules. This is the beauty of OpenWrt: you are not limited to DAWN for interoperation of your router capabilities.

DavidB137, 2023-07-24:
> Generally the philosophy of software must be: do only one thing very well.

SSID exclusion is a basic functionality. It's like saying we don't need `curl` to support DNS name resolution, because we can do it ourselves beforehand (`nslookup`, `dig`, etc.), pass IP address to `curl` and set HTTP headers according to the domain.

SirLouen, 2023-07-24:
> SSID exclusion is a basic functionality. It's like saying we don't need `curl` to support DNS name resolution, because we can do it ourselves beforehand (`nslookup`, `dig`, etc.), pass IP address to `curl` and set HTTP headers according to the domain.

This is arguable, an eternal discussion to decide what is basic and what is not. Personally, I use other better ways of excluding or fixating and don't do much drama. 

In fact, to resolve the DNS it's more than using just a tool, you should be permanently pointing to your internal DNS resolver, which is a way bigger hassle (although it could be laudable if they decided to remove that feature as well)