berlin-open-wireless-lab/DAWN #208: Info: Does now openwrt support 802.11 r/k/v natively?

imwhocodes, 2023-01-12:
Hi
I'm running openwrt snapshot and I noticed a new tab with new configuration about 802.11 r/k/v

It not completely clear what they do / how they integrate with other AP (**I'm** _supposing_ they work only for roaming from 2.4Gh to 5Gh on the same AP)

Is there any idea on how they should be configured to cooperate / integrate with DAWN?

`OpenWrt SNAPSHOT r21772-0a35d3f992 / LuCI Master git-22.361.69865-deed682`

[image]
[image]

Thanks,
Luca

rany2, 2023-05-13:
It does not support it without a third party program. Setting these only make it possible for an external program like DAWN to provide this information (or in the case of BSS Transition, take an action), but OpenWRT itself does not provide any functionality that would make these options useful.

PolynomialDivision, 2023-05-14:
Indeed. There are some things you can do statically to make use of 802.11v, e.g. configuring neighbor reports that allows clients to skip scanning and directly connect to near aps.