berlin-open-wireless-lab/DAWN #210: Kicking mechanism 2, rssi_center < -70 have bad results

Jonsonmon, 2023-03-02:
My client(phone) is currently with -80 signal .
When configuring Dawn with value higher than -70 
(-71, -72 ...)  in rssi_center value I can see BSS_TM requests..
When im using rssi_center of -65 for example, There is no BSS_TM requests anymore...
Remember my device is on -80 signal

SirLouen, 2023-03-02:
Makes sense. RSSI center is weighted so technically -80dB could be within the weighted values. 
Why don't you use rssi_val and low_rssi_val to exclude that -80dB straightly?

Jonsonmon, 2023-03-02:
Using  rasi_val and low_rasi_val is for Kicking mechanism number 1.
Im talking about Kicking mechanism number 2 .
Those values are not in use with `kicking='2'`

SirLouen, 2023-03-02:
> Using rasi_val and low_rasi_val is for Kicking mechanism number 1. Im talking about Kicking mechanism number 2 . Those values are not in use with `kicking='2'`

Sure, but it makes sense how its working. What are you expecting with a weighted value instead of an absolute value? 
If you set this to -72dB, your -80dB won't be probably kicked because is within the weight average.

Do you expect that anything beyond -72dB is kicked? Then set an absolute kick value with a kicking mechanism 1 and you will be ready to go.

Jonsonmon, 2023-03-02:
You have a mistake.
I'm USING absolute rssi ! It's kicking number 2
Kicking number 1 is score comparison.

SirLouen, 2023-03-02:
> You have a mistake. I'm USING absolute rssi ! It's kicking number 2 Kicking number 1 is score comparison.

Maybe I'm getting you wrong. 

This is what I've read:

> When configuring Dawn with value higher than -70 (-71, -72 ...) in rssi_center value I can see BSS_TM requests..

So I think your problem is with `kicking_threshold` try to set it to 0 and check again the BSS_transition requests

Jonsonmon, 2023-03-02:
kicking_threshold is already set to 0

SirLouen, 2023-03-02:
> kicking_threshold is already set to 0

From my understanding those BSS_transition appear when the client is asked to move. What is weird for me is the opposite, the fact that a connected device with center to -65dB is kicked without a BSS_transition trace. This is what I would like to investigate further.

Apart from this, personally I would not use DAWN with kicking number 2. Doesn't make much sense to me, because you have poor control over the kicking method. According to the docs, they think that a hostapd hard kick may interfere in legacy devices, but in my case, it has been proven to be the opposite: legacy devices tend to keep reconnecting to the same node without a hard kick.

So basically depending on the type of network I would not chose to use DAWN instead, the legacy LEDE's [modhostapd](https://github.com/mk248/lede-modhostapd). There is a very simple to set and use script based on such that I think it will work better [for you in this very specific scenario](https://github.com/barbieri/barbieri-playground/tree/master/openwrt/wifi-disconnect-low-signal).

You can set the AP kick to hard, very hard and soft, depending on the algorithm you use within the lua code.

Jonsonmon, 2023-03-03:
I know that script, i'm not looking for simply disconnect clients .
I'm looking for roaming, Sending Beacon and using bss_transition_request .