berlin-open-wireless-lab/DAWN #218: [multicast mode] Allow binding to a specific IP instead of wildcard

rany2, 2023-05-13:
I thought that setting `server_ip` would fix it, but it doesn't seem to check that value in multicast mode.

PolynomialDivision, 2023-05-14:
What do you want to achieve with this? You want a node receiving everything while the other nodes receive the traffic via multicast?

rany2, 2023-05-14:
@PolynomialDivision The reason is that I have an encrypted L2 network for nodes to communicate securely over, it's used for RADIUS, NTP, etc and I don't want to have umdns multicast spam everyone else on my network. So putting it on the encrypted network makes everything a lot cleaner and more secure.

However, it wouldn't make much sense to do this if DAWN just listens on all addresses and some of these devices have no firewall due to being dumb APs.

rany2, 2023-05-14:
Also I should add that the first thing I tried was the UDP broadcast, but then I realized the messages are corrupted due to the large amount of stations (#217); which caused me to want to turn back to the most common and tested mode but not being to set an IP to bind to was a deal breaker for me.

commit f8cdae3f (2025-03-03): tcpsocket: add option to bind to specific ip

Fixes #218
closed 2025-03-03 by PolynomialDivision