berlin-open-wireless-lab/DAWN #230: infrequent SIGSEGV on OpenWrt 22.03.5, r20134-5f15225c1e, mipsel_24kc

ifacify, 2023-09-11:
I have several mipsel_24kc mt7621 openwrt routers running dawn in the network, along with a couple of aarch64 mt7622b routers. The mt7621 builds of dawn seems to have some sort of infrequent memory corruption problem while the aarch64 builds doesn't.
```
192.168.1.2 <6>Sep  3 21:36:41 ap-cc5d kernel: [2614446.922073] br-lan: port 5(wlan1) entered disabled state
192.168.1.2 <6>Sep  3 21:36:41 ap-cc5d kernel: [2614446.997885] device wlan0 left promiscuous mode
192.168.1.2 <6>Sep  3 21:36:41 ap-cc5d kernel: [2614447.002603] br-lan: port 4(wlan0) entered disabled state
192.168.1.2 <6>Sep  3 21:36:42 ap-cc5d kernel: [2614447.094972] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000018
192.168.1.2 <6>Sep  3 21:36:42 ap-cc5d kernel: [2614447.103507] epc = 0040c403 in dawn[400000+13000]
192.168.1.2 <6>Sep  3 21:36:42 ap-cc5d kernel: [2614447.108434] ra  = 0040c401 in dawn[400000+13000]
192.168.1.1 <27>Sep  3 21:36:42 ap-d3b4 dawn: client_notify_state()=tcpsocket.c@76 Closing client-connection, pending: 0, total: 0
--
192.168.1.2 <28>Sep  3 21:36:57 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:36:57 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:36:57 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <6>Sep  3 21:36:57 ap-cc5d kernel: [2614462.603504] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000068
192.168.1.2 <6>Sep  3 21:36:57 ap-cc5d kernel: [2614462.612026] epc = 0040c0fd in dawn[400000+13000]
192.168.1.2 <6>Sep  3 21:36:57 ap-cc5d kernel: [2614462.616862] ra  = 0040c0fd in dawn[400000+13000]
192.168.1.13 <29>Sep  3 21:36:57 ap-5598 hostapd: wlan1: BEACON-REQ-TX-STATUS B8:FE:3C:DF:C4:FC 40 ack=1
--
192.168.1.2 <28>Sep  3 21:37:12 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:37:12 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:37:12 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <6>Sep  3 21:37:12 ap-cc5d kernel: [2614477.763284] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000068
192.168.1.2 <6>Sep  3 21:37:12 ap-cc5d kernel: [2614477.771778] epc = 0040c0fd in dawn[400000+13000]
192.168.1.2 <6>Sep  3 21:37:12 ap-cc5d kernel: [2614477.776623] ra  = 0040c0fd in dawn[400000+13000]
192.168.1.5 <29>Sep  3 21:37:12 ap-8370 hostapd: wlan1: BEACON-REQ-TX-STATUS 5F:5F:5F:87:6C:E7 69 ack=1
--
192.168.1.2 <28>Sep  3 21:37:27 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:37:27 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:37:27 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <6>Sep  3 21:37:27 ap-cc5d kernel: [2614492.977548] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000068
192.168.1.2 <6>Sep  3 21:37:27 ap-cc5d kernel: [2614492.986116] epc = 0040c0fd in dawn[400000+13000]
192.168.1.2 <6>Sep  3 21:37:27 ap-cc5d kernel: [2614492.990958] ra  = 0040c0fd in dawn[400000+13000]
192.168.1.5 <29>Sep  3 21:37:30 ap-8370 hostapd: wlan1: BEACON-REQ-TX-STATUS 5F:5F:5F:87:6C:E7 78 ack=1
--
192.168.1.2 <28>Sep  3 21:37:42 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:37:42 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <28>Sep  3 21:37:42 ap-cc5d dawn: Failed to lookup ID!
192.168.1.2 <6>Sep  3 21:37:43 ap-cc5d kernel: [2614508.140760] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 00000068
192.168.1.2 <6>Sep  3 21:37:43 ap-cc5d kernel: [2614508.149315] epc = 0040c0fd in dawn[400000+13000]
192.168.1.2 <6>Sep  3 21:37:43 ap-cc5d kernel: [2614508.154117] ra  = 0040c0fd in dawn[400000+13000]
192.168.1.10 <27>Sep  3 21:37:45 ap-3178 hostapd: nl80211: kernel reports: key addition failed
--
192.168.1.1 <27>Sep  7 09:37:46 ap-d3b4 hostapd: Failed to set beacon parameters
192.168.1.1 <27>Sep  7 09:37:46 ap-d3b4 hostapd: Failed to set beacon parameters
192.168.1.2 <30>Sep  7 09:37:46 ap-cc5d dawn: Client 8d:26:af:12:b9:c4: Kicking due to low active data transfer: RX rate 6.000000 below 6 limit
192.168.1.2 <6>Sep  7 09:37:46 ap-cc5d kernel: [2916902.619681] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77bd3490
192.168.1.2 <6>Sep  7 09:37:46 ap-cc5d kernel: [2916902.628175] epc = 00403c19 in dawn[400000+13000]
192.168.1.1 <27>Sep  7 09:37:46 ap-d3b4 dawn: client_to_server_state()=tcpsocket.c@103 Closing connection, pending: 0, total: 33816578
192.168.1.2 <6>Sep  7 09:37:46 ap-cc5d kernel: [2916902.632979] ra  = 00403c17 in dawn[400000+13000]
--
192.168.1.17 <30>Sep  9 19:47:35 ap-9aa3 dawn: Client f3:26:7f:db:68:66: Kicking due to low active data transfer: RX rate 6.000000 below 6 limit
192.168.1.2 <30>Sep  9 19:47:35 ap-cc5d dawn: Client a1:ee:b5:3d:59:12: Kicking due to low active data transfer: RX rate 1.000000 below 6 limit
192.168.1.10 <30>Sep  9 19:47:36 ap-3178 dawn: Client 8d:26:af:12:b9:c4: Kicking due to low active data transfer: RX rate 6.000000 below 6 limit
192.168.1.10 <6>Sep  9 19:47:36 ap-3178 kernel: [5513185.610796] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77e11010
192.168.1.10 <6>Sep  9 19:47:36 ap-3178 kernel: [5513185.619391] epc = 00403c19 in dawn[400000+13000]
192.168.1.10 <6>Sep  9 19:47:36 ap-3178 kernel: [5513185.624252] ra  = 00403c17 in dawn[400000+13000]
192.168.1.1 <27>Sep  9 19:47:36 ap-d3b4 dawn: client_notify_state()=tcpsocket.c@76 Closing client-connection, pending: 0, total: 0
--
192.168.1.17 <29>Sep  9 19:48:09 ap-9aa3 hostapd: wlan0: BEACON-REQ-TX-STATUS 1F:23:C8:D0:2C:A2 140 ack=0
192.168.1.17 <29>Sep  9 19:48:09 ap-9aa3 hostapd: wlan0: BEACON-REQ-TX-STATUS 1F:23:C8:D0:2C:A2 141 ack=0
192.168.1.10 <30>Sep  9 19:48:12 ap-3178 dawn: Client 8d:26:af:12:b9:c4: Kicking due to low active data transfer: RX rate 6.000000 below 6 limit
192.168.1.10 <6>Sep  9 19:48:12 ap-3178 kernel: [5513221.322307] do_page_fault(): sending SIGSEGV to dawn for invalid read access from 77be8010
192.168.1.10 <6>Sep  9 19:48:12 ap-3178 kernel: [5513221.330791] epc = 00403c19 in dawn[400000+13000]
192.168.1.1 <27>Sep  9 19:48:12 ap-d3b4 dawn: client_to_server_state()=tcpsocket.c@103 Closing connection, pending: 0, total: 33816578
192.168.1.1 <28>Sep  9 19:48:12 ap-d3b4 dawn: Connection to server closed
```
All instances of dawn are using almost default config, with just shared_key and iv changed.

PolynomialDivision, 2023-09-11:
Can you try latest main dawn build? Another person disabled lto, because of strange segfaults. Maybe it also helps you.

ifacify, 2023-09-11:
I'll try it first on some other non production devices, If it does not segfault I may consider packaging dawn with lto disabled on those devices. Will need a few days to see if the segfault still happens. Thank you.

closed 2023-12-20 by ifacify